dresden-elektronik/deconz-rest-plugin #4730: ZLL.DB has grown (too) huge

mk1blacklimited, 2021-04-08:
## Describe the question or issue you are having
My zll.db has grown huge, making adding/removing devices extremely slow:

-rw-r--r-- 1 pi pi 802650112 Apr  8 13:00 zll.db

What can I do about that?

## Screenshots

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.09.03
 - Device: RaspBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

senilio, 2021-04-08:
Can you check how many devices is listed in your db?
```
cd ~/.local/share/dresden-elektronik/deCONZ
sqlite3 zll.db "select * from devices;" | wc -l
```

mk1blacklimited, 2021-04-08:
Well, I've restored a backup I did back in january, so this version shows 117 devices. However, just after the restore the size of the zll.db was 326656. Now, a few hours later, it's grown to 7154688. Is that normal?

I've backed up the 802 megabyte zll.db as well so I can put it back if need be.

senilio, 2021-04-08:
No this is not normal.

While I'm not going to score points with the SQL elite for proposing this, this elaborate oneliner will hopefully give us an idea if there's a specific table containing all this (bogus) data:
```
for i in $(sqlite3 zll.db ".tables"); do echo -en "$i: "; sqlite3 zll.db "SELECT SUM("pgsize") FROM "dbstat" WHERE name=\"$i\";"; done
```

This is what my output looks like:
```
auth: 24576
gateways: 4096
scenes: 4096
userparameter: 4096
config2: 4096
groups: 4096
schedules: 4096
zbconf: 4096
device_descriptors: 12288
nodes: 32768
sensors: 24576
zcl_values: 4096
device_gui: 4096
resourcelinks: 4096
source_route_hops: 4096
devices: 4096
rules: 4096
source_routes: 4096
```

Should you find a suspect here, you can look at the content by running something like:
```sqlite3 zll.db "select * from your_table limit 500;```

This will hopefully help us pinpoint what is going on.

mk1blacklimited, 2021-04-09:
Thanks for helping :) Today the size has grown to 19715072. Output from the sql-statement:

```
auth: 52224
gateways: 1024
scenes: 1024
userparameter: 1024
config2: 4096
groups: 6144
schedules: 1024
zbconf: 17408
device_descriptors: 17408
nodes: 19334144
sensors: 49152
zcl_values: 41984
device_gui: 11264
resourcelinks: 1024
source_route_hops: 1024
devices: 7168
rules: 5120
source_routes: 1024

```
nodes looks suspicious?

Not sure what to look for, but here is the start of the output:

`sqlite3 zll.db "select * from nodes limit 500;"
84:18:26:00:00:ca:54:24-03|5|deleted|Høyre oppe||3|Classic A60 TW|OSRAM|V1.03.07|
84:18:26:00:00:ca:5f:4b-03|4|deleted|Venstre oppe||3|Classic A60 TW|OSRAM|V1.05.09|
00:12:4b:00:1b:6a:eb:c1-0b|33|deleted|Light 33||11||Unknown||{"attr/modelid":null,"attr/name":"Light 33","attr/swversion":null,"attr/type":"Extended color light","attr/uniqueid":"00:12:4b:00:1b:6a:eb:c1-0b","config/colorcapabilities":0,"config/ctmax":0,"config/ctmin":0,"state/alert":null,"state/bri":0,"state/colormode":"hs","state/ct":0,"state/hue":0,"state/on":false,"state/reachable":false,"state/sat":0,"state/x":0,"state/y":0}
00:12:4b:00:1b:6a:eb:c1-0c|32|deleted|Light 32||12|GL-S-003Z|GLEDOPTO|1.0.2|{"attr/modelid":"GL-S-003Z","attr/name":"Light 32","attr/swversion":"1.0.2","attr/type":"Color dimmable light","attr/uniqueid":"00:12:4b:00:1b:6a:eb:c1-0c","config/colorcapabilities":31,"state/alert":null,"state/bri":254,"state/colormode":"hs","state/ct":322,"state/hue":0,"state/on":true,"state/reachable":false,"state/sat":0,"state/x":20480,"state/y":21845}
00:12:4b:00:1b:6a:e7:65-0b|37|deleted|Light 37||11||Unknown||{"attr/modelid":null,"attr/name":"Light 37","attr/swversion":null,"attr/type":"Extended color light","attr/uniqueid":"00:12:4b:00:1b:6a:e7:65-0b","config/colorcapabilities":0,"config/ctmax":0,"config/ctmin":0,"state/alert":null,"state/bri":0,"state/colormode":"hs","state/ct":0,"state/hue":0,"state/on":false,"state/reachable":false,"state/sat":0,"state/x":0,"state/y":0}
00:12:4b:00:1b:6a:e7:65-0c|36|deleted|Light 36||12|GL-S-003Z|GLEDOPTO|1.0.3|{"attr/modelid":"GL-S-003Z","attr/name":"Light 36","attr/swversion":"1.0.3","attr/type":"Color dimmable light","attr/uniqueid":"00:12:4b:00:1b:6a:e7:65-0c","state/alert":null,"state/bri":254,"state/colormode":"xy","state/hue":0,"state/on":true,"state/reachable":false,"state/sat":0,"state/x":0,"state/y":0}
00:12:4b:00:1b:6a:e7:98-0b|39|deleted|Light 39||11||Unknown||{"attr/modelid":null,"attr/name":"Light 39","attr/swversion":null,"attr/type":"Extended color light","attr/uniqueid":"00:12:4b:00:1b:6a:e7:98-0b","config/colorcapabilities":0,"config/ctmax":0,"config/ctmin":0,"state/alert":null,"state/bri":0,"state/colormode":"hs","state/ct":0,"state/hue":0,"state/on":false,"state/reachable":false,"state/sat":0,"state/x":0,"state/y":0}
00:12:4b:00:1b:6a:e7:98-0c|38|deleted|Light 38||12|GL-S-003Z|GLEDOPTO|1.0.3|{"attr/modelid":"GL-S-003Z","attr/name":"Light 38","attr/swversion":"1.0.3","attr/type":"Color dimmable light","attr/uniqueid":"00:12:4b:00:1b:6a:e7:98-0c","state/alert":null,"state/bri":255,"state/colormode":"xy","state/hue":33792,"state/on":false,"state/reachable":false,"state/sat":34,"state/x":65279,"state/y":65279}
84:18:26:00:00:c9:d3:43-03|9|deleted|Venstre under lerret||3|Classic A60 RGBW|OSRAM|V1.05.10|{"attr/manufacturername":"OSRAM","attr/modelid":"Classic A60 RGBW","attr/name":"Venstre under lerret","attr/swversion":"V1.05.10","attr/type":"Extended color light","attr/uniqueid":"84:18:26:00:00:c9:d3:43-03","config/colorcapabilities":31,"config/ctmax":500,"config/ctmin":151,"state/alert":null,"state/bri":254,"state/colormode":"ct","state/ct":370,"state/hue":0,"state/on":true,"state/reachable":false,"state/sat":0,"state/x":24939,"state/y":24701}
90:fd:9f:ff:fe:33:5a:39-01|19|deleted|Nede høyre||1|TRADFRI bulb E27 W opal 1000lm|IKEA of Sweden|1.2.214|{"attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E27 W opal 1000lm","attr/name":"Nede høyre","attr/swversion":"1.2.214","attr/type":"Dimmable light","attr/uniqueid":"90:fd:9f:ff:fe:33:5a:39-01","state/alert":null,"state/bri":20,"state/on":true,"state/reachable":false}
84:18:26:00:00:c9:fc:98-03|8|deleted|Nede høyre||3|Classic A60 RGBW|OSRAM|V1.05.10|{"attr/manufacturername":"OSRAM","attr/modelid":"Classic A60 RGBW","attr/name":"Nede høyre","attr/swversion":"V1.05.10","attr/type":"Extended color light","attr/uniqueid":"84:18:26:00:00:c9:fc:98-03","config/colorcapabilities":31,"config/ctmax":500,"config/ctmin":151,"state/alert":null,"state/bri":20,"state/colormode":"ct","state/ct":370,"state/hue":0,"state/on":true,"state/reachable":false,"state/sat":0,"state/x":24939,"state/y":24701}
00:0b:3c:ff:fe:f4:2e:37-01|10|deleted|Nede høyre||1|TRADFRI bulb E27 W opal 1000lm|IKEA of Sweden|1.2.214|{"attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E27 W opal 1000lm","attr/name":"Nede høyre","attr/swversion":"1.2.214","attr/type":"Dimmable light","attr/uniqueid":"00:0b:3c:ff:fe:f4:2e:37-01","state/alert":null,"state/bri":254,"state/on":false,"state/reachable":false}
84:18:26:00:00:c9:d2:ee-03|27|deleted|Lights||3|Classic A60 RGBW|OSRAM|V1.05.10|{"attr/manufacturername":"OSRAM","attr/modelid":"Classic A60 RGBW","attr/name":"Lights","attr/swversion":"V1.05.10","attr/type":"Extended color light","attr/uniqueid":"84:18:26:00:00:c9:d2:ee-03","config/colorcapabilities":31,"config/ctmax":500,"config/ctmin":151,"state/alert":null,"state/bri":254,"state/colormode":"ct","state/ct":3 [... cut ...] e/hue":0,"state/on":true,"state/reachable":false,"state/sat":0,"state/x":30015,"state/y":26870}
90:fd:9f:ff:fe:00:fb:4f-01|18|deleted|Nede venstre||1|TRADFRI bulb E27 W opal 1000lm|IKEA of Sweden|1.2.214|{"attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E27 W opal 1000lm","attr/name":"Nede venstre","attr/swversion":"1.2.214","attr/type":"Dimmable light","attr/uniqueid":"90:fd:9f:ff:fe:00:fb:4f-01","state/alert":null,"state/bri":254,"state/on":false,"state/reachable":false}
`

senilio, 2021-04-09:
To see what keeps getting appended to the table, try this:
```
sqlite3 zll.db "SELECT * FROM nodes ORDER BY id DESC LIMIT 100;"
```

mk1blacklimited, 2021-04-09:
Right, this seems to be repeating:

`d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T04:44Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}
d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T04:33Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}
d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T04:25Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}
d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T04:13Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}
d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T04:03Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}
d0:cf:5e:ff:fe:7b:5e:4d-01|84|normal|Lerret høyre/Thor|65520,14,5|1|TRADFRI bulb E14 W op/ch 400lm|IKEA of Sweden|2.3.068|{"attr/id":"84","attr/lastannounced":"2021-01-11T12:18:13Z","attr/lastseen":"2021-04-09T03:59Z","attr/manufacturername":"IKEA of Sweden","attr/modelid":"TRADFRI bulb E14 W op/ch 400lm","attr/name":"Lerret høyre/Thor","attr/swversion":"2.3.068","attr/type":"Dimmable light","attr/uniqueid":"d0:cf:5e:ff:fe:7b:5e:4d-01","state/alert":null,"state/bri":25,"state/on":true,"state/reachable":true}`

So - delete it and re-add? Or is it a device I should throw away?

senilio, 2021-04-09:
I don't know if anything is wrong with the bulb tbh.

I would start with what you propose, deleting this particular bulb via Phoscon, then check if DB size have improved. If it's better, re pair this bulb.

If the DB is still huge even after deleting this bulb via Phoscon, we'll need to delete all duplicate nodes with id 84 from the nodes table by running: ```sqlite3 zll.db "DELETE FROM nodes where id=84;"```
Then restart deCONZ.

I don't know if this is an issue that have been fixed in newer versions of deCONZ, but you might consider upgrading to tha latest stable version once you have this problem under control.

senilio, 2021-04-09:
Let's also check if id 84 is the only one with duplicates, or if there are others:
```
sqlite3 zll.db "SELECT id FROM nodes;"|sort -n|uniq -c
```

mk1blacklimited, 2021-04-09:
Deleted the bulb, but the database keeps increasing. Output from you last sql:

`    247 1
    692 2
     95 3
      2 4
      1 5
    210 6
    211 7
      3 8
      2 9
      1 10
    736 11
    178 12
    156 13
      6 14
      8 15
    166 16
      1 17
      2 18
      1 19
    203 20
    805 21
    752 22
    574 23
    476 24
    933 25
    812 26
      2 27
    245 28
      3 29
    112 30
    126 31
      2 32
      1 33
    130 34
    252 35
      1 36
      1 37
      2 38
      3 39
    145 40
    646 41
    598 42
    538 43
    541 44
    406 45
    765 46
    639 47
    720 48
    627 49
      1 50
    745 51
    172 52
    125 53
    147 54
    441 55
      2 56
    138 57
    157 58
    869 59
      1 60
    499 61
    467 62
    239 63
    628 64
    601 65
    211 66
    133 68
    133 69
      2 70
     19 71
    749 72
    779 73
    727 74
      4 75
      2 76
      3 77
     21 78
     26 79
    656 80
      1 81
      3 82
      1 83
    120 84
      1 86
      2 87
      1 90
      1 93
      1 94
      1 98
      1 101
      2 103
      1 107
      1 109
      1 110
      2 111
      1 112
      1 113
      1 114
      1 115
      1 116
      1 117
      2 118
      2 121
      2 125
      1 128
      1 130
      1 132
      1 135
      1 136
      1 137
      1 138
      1 139
      1 140
      2 147
      1 150
      1 154
      1 159
      1 161
      1 162
      1 163
      1 164
      1 165
      1 166
      1 169
      1 171
      1 172
      1 173
      1 176`

Mimiix, 2021-04-09:
@SwoopX This issue reminds me of tht issue with the fake nodes. I forgot what issue it was.

senilio, 2021-04-09:
😓

Doesn't seem like we have a pretty solution for this... this is the closest I can find:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4542#issuecomment-793870688

mk1blacklimited, 2021-04-09:
OK, so the datbase is FUBAR then, great...:) 

I don't quite understand when I put back a backup from january, with a small database of 326656 bytes, it still includes 'something' that makes the database grow exponentially?

What would be the sensible way forward? Stopping deconz, deleting the database alltogether, restart deconz and re-add everything?

mk1blacklimited, 2021-04-09:
Oh, and I do not have 176 devices, more like 90.

mk1blacklimited, 2021-04-09:
Had another backup 4 days before I put back now. File size hasn't grown much, and the sql output looks a lot better:

sqlite3 zll.db "SELECT id FROM nodes;"|sort -n|uniq -c
      1 Motion Sensor trapp oppe L
      1 1
      1 2
      1 3
      1 4
      1 5
      1 7
      1 8
      1 9
      1 10
      1 12
      1 13
      1 14
      1 15
      1 16
      1 17
      1 18
      1 19
      1 20
      1 21
      1 22
      1 23
      1 24
      1 25
      1 26
      1 27
      1 28
      1 29
      1 30
      1 31
      1 32
      1 33
      1 34
      1 35
      1 36
      1 37
      1 38
      1 39
      1 40
      1 41
      1 42
      1 43
      1 44
      1 45
      1 46
      1 47
      1 48
      1 49
      1 50
      1 51
      1 53
      1 54
      1 55
      1 56
      1 57
      1 59
      1 60
      1 61
      1 62
      1 63
      1 64
      1 65
      1 66
      1 68
      1 69
      1 70
      1 71
      1 72
      1 73
      1 74
      1 75
      1 76
      1 77
      1 78
      1 79
      1 80
      1 81
      1 82
      1 84

mk1blacklimited, 2021-04-09:
Table output:

`auth: 52224
gateways: 1024
scenes: 1024
userparameter: 1024
config2: 4096
groups: 6144
schedules: 1024
zbconf: 17408
device_descriptors: 16384
nodes: 71680
sensors: 49152
zcl_values: 41984
device_gui: 6144
resourcelinks: 1024
source_route_hops: 1024
devices: 7168
rules: 5120
source_routes: 1024`

mk1blacklimited, 2021-04-09:
Bugger...:

pi@phoscon:~/.local/share/dresden-elektronik/deCONZ $ ls -l zll.db 
-rw-r--r-- 1 pi pi 324608 Apr  9 11:07 zll.db
pi@phoscon:~/.local/share/dresden-elektronik/deCONZ $ ls -l zll.db 
-rw-r--r-- 1 pi pi 1093632 Apr  9 11:57 zll.db

mk1blacklimited, 2021-04-09:
Table output:

auth: 52224
gateways: 1024
scenes: 1024
userparameter: 1024
config2: 4096
groups: 6144
schedules: 1024
zbconf: 17408
device_descriptors: 17408
nodes: 775168
sensors: 106496
zcl_values: 41984
device_gui: 5120
resourcelinks: 1024
source_route_hops: 1024
devices: 7168
rules: 5120
source_routes: 1024

mk1blacklimited, 2021-04-09:
Count:

pi@phoscon:~/.local/share/dresden-elektronik/deCONZ $ sqlite3 zll.db "SELECT id FROM nodes;"|sort -n|uniq -c
      1 Fjernkontroll barnerom
      1 Motion Sensor trapp oppe L
      1 Multisensor VP
      7 1
     34 2
      3 3
      1 4
      1 5
      8 7
      1 8
      1 9
      1 10
      7 12
      4 13
      2 14
      1 15
      6 16
      2 17
      1 18
      1 19
      5 20
     28 21
     21 22
     24 23
     27 24
     39 25
     33 26
      1 27
     33 28
      1 29
      3 30
      2 31
      1 32
      1 33
      2 34
      8 35
      1 36
      1 37
      1 38
      1 39
      3 40
     23 41
     23 42
     13 43
     19 44
     13 45
     29 46
     22 47
     20 48
     23 49
      2 50
     31 51
      3 53
      8 54
     21 55
      1 56
      6 57
     30 59
      1 60
     28 61
     22 62
     13 63
     23 64
     22 65
      9 66
      8 68
      4 69
      1 70
      2 71
     20 72
     29 73
     23 74
      1 75
      1 76
      1 77
      2 78
      2 79
     17 80
      1 81
      1 82
      2 84

oywino, 2021-04-09:
Have you tried upgrading to 2.10.04 ?

mk1blacklimited, 2021-04-09:
Yes, running that since restirung backup #2 :)

mk1blacklimited, 2021-04-09:
Now running a backup I made dec. 1 2020. Still holding file size after a couple of hours.

senilio, 2021-04-11:
> Now running a backup I made dec. 1 2020. Still holding file size after a couple of hours.

Hey, are things still holding up after the upgrade?

SwoopX, 2021-04-11:
> @SwoopX This issue reminds me of tht issue with the fake nodes. I forgot what issue it was.
> 

@Mimiix ah, not quite. That one was a mystery indeed.

I recall we had the same issue like a month ago here. Think the user manually deleted any duplicates from the DB manually :/

mk1blacklimited, 2021-04-11:
> > Now running a backup I made dec. 1 2020. Still holding file size after a couple of hours.
> 
> Hey, are things still holding up after the upgrade?

Yep. Must be something really funky stuff in those newer backups I guess. :)

This might all be caused by a faulty SD-card. I had to replace it and restore a backup back in february (I think). All went well after the restore, but I never checked the database back then (why would I :) ).