dresden-elektronik/deconz-rest-plugin #4731: legrand - Valena Next with Netatmo - connected wall plug

kvanisterdael, 2021-04-08:
## Device

- Product name: Valena Next with Netatmo - connected wall plug
- Manufacturer: Legrand
- Model identifier: Connected outlet
- Device type : Please remove all unrelated device types. 
  - Other: Outlet (plug)

## Screenshots
[image]

[image]

### Basic
[image]

### Identify
[image]

### On/Off
[image]
[image]

Smanar, 2021-04-08:
This device work natively, bad included for you, the name is grayed in deconz.

kvanisterdael, 2021-04-22:
I tried another time and I can confirm it works as expected.

closed 2021-04-22 by kvanisterdael