dresden-elektronik/deconz-rest-plugin #4733: legrand - Valena Next with Netatmo - Wireless shutter switch

kvanisterdael, 2021-04-08:
## Device

- Product name: Valena Next with Netatmo - Wireless shutter switch (ref. 741808)
- Manufacturer: Legrand
- Model identifier: Shutters central remote switch
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots
[image]

[image]

### Basic
[image]

### Identify
[image]

### Remark
Just like the lighting switch, the switch has 2 buttons (top and bottom) that ideally could generate a different buttonevent.

Smanar, 2021-04-08:
Ok so for the switch it s normal, but not for this one, I realy think you need to have different command for this one.

Can you enable log, deconz/help/log with "info" and "info_l2", press shortly "up" and "down", and show the logs pls ?

kvanisterdael, 2021-04-22:
For clarity: the button doesn't do anything for the moment, so no button events on the Websockets API...

Here are the requested logs; I have no idea what you are looking for, so I hope it's in there:

`15:07:10:581 enqueue event config/localtime for /config/
15:07:10:588 poll node 84:2e:14:ff:fe:7a:89:cb-01
15:07:10:594 Poll light node Ikea RGB
15:07:10:608 rule event /config/localtime: 15:07:09.581 -> 15:07:10.581 (1000ms)
15:07:10:637 Poll APS request to 0x842E14FFFE7A89CB cluster: 0x0006 dropped, values are fresh enough
15:07:11:200 Beacon src: 0xDB8E ch: 11 updateId: 1
15:07:11:240 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:11:251 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:11:300 Beacon src: 0xAC65 ch: 11 updateId: 1
15:07:11:497 Beacon src: 0xDB8E ch: 11 updateId: 1
15:07:11:538 Beacon src: 0xAC65 ch: 11 updateId: 1
15:07:11:562 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:11:581 enqueue event config/localtime for /config/
15:07:11:596 rule event /config/localtime: 15:07:10.581 -> 15:07:11.581 (1000ms)
15:07:11:612 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:11:754 poll node 00:17:88:01:08:e8:1a:0e-0b
15:07:11:761 Poll light node Eetkamer opladers
15:07:11:799 Beacon src: 0xDB8E ch: 11 updateId: 1
15:07:11:811 read attributes of 0x0017880108E81A0E cluster: 0x0006: [ 
15:07:11:823 0x0000 
15:07:11:830 ]
15:07:11:837 add task 3857 type 19 to 0x0017880108E81A0E cluster 0x0006 req.id 93
15:07:11:844 Poll APS request 93 to 0x0017880108E81A0E cluster: 0x0006
15:07:11:851 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:11:886 Beacon src: 0xAC65 ch: 11 updateId: 1
15:07:11:931 Poll APS confirm 93 status: 0x00
15:07:11:938 Erase task req-id: 93, type: 19 zcl seqno: 182 send time 0, profileId: 0x0104, clusterId: 0x0006
15:07:11:965 Node data 0x0017880108e81a0e profileId: 0x0104, clusterId: 0x0006
15:07:11:974 0x0017880108E81A0E: update ZCL value 0x0B/0x0006/0x0000 after 0 s
15:07:11:982 enqueue event event/checkgroupanyon for /groups/65520
15:07:12:150 Beacon src: 0xDB8E ch: 11 updateId: 1
15:07:12:195 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:12:240 Beacon src: 0xAC65 ch: 11 updateId: 1
15:07:12:247 Beacon src: 0x4DF9 ch: 11 updateId: 1
15:07:12:259 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:12:443 Beacon src: 0xDD8D ch: 11 updateId: 1
15:07:12:469 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:12:510 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:12:581 enqueue event config/localtime for /config/
15:07:12:629 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
15:07:12:636 COM: /dev/ttyS0 :  (0x0000/0x0000)
15:07:12:643 dev /dev/ttyS0 (/dev/ttyS0)
15:07:12:650 rule event /config/localtime: 15:07:11.581 -> 15:07:12.581 (1000ms)
15:07:12:663 GW firmware version: 0x26390500
15:07:12:676 GW firmware version is up to date: 0x26390500
15:07:12:750 poll node 60:a4:23:ff:fe:88:13:8d-01
15:07:12:756 Poll light node Lavalamp Warre
15:07:12:770 Beacon src: 0xDB8E ch: 11 updateId: 1
15:07:12:795 Beacon src: 0xAC65 ch: 11 updateId: 1
15:07:12:803 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:12:810 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:12:826 Poll APS request to 0x60A423FFFE88138D cluster: 0x0006 dropped, values are fresh enough
15:07:12:837 Beacon src: 0xDD8D ch: 11 updateId: 1
15:07:13:581 enqueue event config/localtime for /config/
15:07:13:598 rule event /config/localtime: 15:07:12.581 -> 15:07:13.581 (1000ms)
15:07:13:732 poll node 60:a4:23:ff:fe:88:15:11-01
15:07:13:739 Poll light node Verwarming badkamer ouders
15:07:13:791 Poll APS request to 0x60A423FFFE881511 cluster: 0x0006 dropped, values are fresh enough
15:07:14:580 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
15:07:14:587 enqueue event config/localtime for /config/
15:07:14:604 rule event /config/localtime: 15:07:13.581 -> 15:07:14.587 (1006ms)
15:07:14:668 poll node 60:a4:23:ff:fe:88:b9:41-01
15:07:14:677 Poll light node Verwarming badkamer kinderen
15:07:14:721 Poll APS request to 0x60A423FFFE88B941 cluster: 0x0006 dropped, values are fresh enough
15:07:15:581 enqueue event config/localtime for /config/
15:07:15:607 poll node 60:a4:23:ff:fe:8c:cc:51-01
[... 41 lines cut ...]
15:07:18:831 0x0000 
15:07:18:837 ]
15:07:18:844 add task 3886 type 19 to 0x5C0272FFFE694793 cluster 0x0006 req.id 138
15:07:18:851 Poll APS request 138 to 0x5C0272FFFE694793 cluster: 0x0006
15:07:18:925 Poll APS confirm 138 status: 0x00
15:07:18:933 Erase task req-id: 138, type: 19 zcl seqno: 183 send time 0, profileId: 0x0104, clusterId: 0x0006
15:07:18:979 Node data 0x5c0272fffe694793 profileId: 0x0104, clusterId: 0x0006
15:07:18:987 0x5C0272FFFE694793: update ZCL value 0x01/0x0006/0x0000 after 0 s
15:07:18:994 enqueue event event/checkgroupanyon for /groups/65520
15:07:19:002 Beacon src: 0xB22B ch: 11 updateId: 1
15:07:19:032 Beacon src: 0xB5C4 ch: 11 updateId: 1
15:07:19:581 enqueue event config/localtime for /config/
15:07:19:591 rule event /config/localtime: 15:07:18.581 -> 15:07:19.581 (1000ms)
15:07:19:743 poll node 00:17:88:01:08:e8:1a:0e-0b
15:07:19:750 Poll light node Eetkamer opladers
15:07:19:791 read attributes of 0x0017880108E81A0E cluster: 0x0006: [ 
15:07:19:798 0x0000 
15:07:19:804 ]
15:07:19:810 add task 3891 type 19 to 0x0017880108E81A0E cluster 0x0006 req.id 147
15:07:19:820 Poll APS request 147 to 0x0017880108E81A0E cluster: 0x0006
15:07:19:930 Poll APS confirm 147 status: 0x00
15:07:19:937 Erase task req-id: 147, type: 19 zcl seqno: 184 send time 0, profileId: 0x0104, clusterId: 0x0006
15:07:19:964 Node data 0x0017880108e81a0e profileId: 0x0104, clusterId: 0x0006
15:07:19:972 0x0017880108E81A0E: update ZCL value 0x0B/0x0006/0x0000 after 0 s
15:07:19:979 enqueue event event/checkgroupanyon for /groups/65520`

Smanar, 2021-04-23:
Effectively, there is realy nothing about this device ?

And you can see it in the API ? If yes can you share the device JSON, else if not, can you retry an inclusion or better as you can see it in deconz.
- set phoscon in permit join (add new sensor)
- read again basic attribute on deconz, cluster 0x0000, button "read" (probably need to wake up the device in same time)

kvanisterdael, 2021-04-23:
I paired the device again and here are the logs when I push up and then down:

```
10:12:30:081 enqueue event config/localtime for /config/
10:12:30:092 rule event /config/localtime: 10:12:29.081 -> 10:12:30.081 (1000ms)
10:12:30:260 poll node 60:a4:23:ff:fe:88:13:8d-01
10:12:30:268 Poll light node Lavalamp Warre
10:12:30:313 Poll APS request to 0x60A423FFFE88138D cluster: 0x0006 dropped, values are fresh enough
10:12:31:081 enqueue event config/localtime for /config/
10:12:31:092 poll node 60:a4:23:ff:fe:88:15:11-01
10:12:31:101 Poll light node Verwarming badkamer ouders
10:12:31:108 Idle timer triggered
10:12:31:115 Force binding of attribute reporting for node Remote Switch
10:12:31:129 rule event /config/localtime: 10:12:30.081 -> 10:12:31.081 (1000ms)
10:12:31:227 Poll APS request to 0x60A423FFFE881511 cluster: 0x0006 dropped, values are fresh enough
10:12:31:654 enqueue event event/validgroup for /sensors/28
10:12:31:661 [INFO] - Button 1002 - Shutters central remote switch, broadcast to: 0x0007, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), action: Move up (with on/off), payload: None, zclSeq: 2
10:12:31:668 enqueue event state/buttonevent for /sensors/28
10:12:31:675 enqueue event state/lastupdated for /sensors/28
10:12:31:683 Force binding of attribute reporting for sensor Remote Switch
10:12:31:691 queue binding task for 0x0004740000A9AFF4, cluster 0x0102
10:12:31:717 Websocket 192.168.0.14:58632 send message: {"e":"changed","id":"28","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-04-23T08:12:31.675"},"t":"event","uniqueid":"00:04:74:00:00:a9:af:f4-01-0102"} (ret = 165)
10:12:31:726 Websocket 192.168.0.3:49885 send message: {"e":"changed","id":"28","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-04-23T08:12:31.675"},"t":"event","uniqueid":"00:04:74:00:00:a9:af:f4-01-0102"} (ret = 165)
10:12:31:747 discard sensor state push for 28: state/lastupdated (already pushed)
10:12:32:081 enqueue event config/localtime for /config/
10:12:32:096 rule event /config/localtime: 10:12:31.081 -> 10:12:32.081 (1000ms)
10:12:32:188 poll node 60:a4:23:ff:fe:88:b9:41-01
10:12:32:196 Poll light node Verwarming badkamer kinderen
10:12:32:237 Poll APS request to 0x60A423FFFE88B941 cluster: 0x0006 dropped, values are fresh enough
10:12:32:604 Bind response success for 0x0004740000a9aff4 ep: 0x01 cluster: 0x0102
10:12:33:081 enqueue event config/localtime for /config/
10:12:33:088 poll node 5c:02:72:ff:fe:69:47:93-01
10:12:33:095 Poll light node Niko opbouw
10:12:33:104 Idle timer triggered
10:12:33:112 Force read attributes for ZHASwitch SensorNode Knop Jules
10:12:33:119 don't create binding for attribute reporting of sensor Knop Jules
10:12:33:125 Force binding of attribute reporting for node Knop Jules
10:12:33:139 rule event /config/localtime: 10:12:32.081 -> 10:12:33.081 (1000ms)
10:12:33:210 read attributes of 0x5C0272FFFE694793 cluster: 0x0006: [ 
10:12:33:224 0x0000 
10:12:33:237 ]
10:12:33:246 add task 282311 type 19 to 0x5C0272FFFE694793 cluster 0x0006 req.id 26
10:12:33:256 Poll APS request 26 to 0x5C0272FFFE694793 cluster: 0x0006
10:12:33:328 Poll APS confirm 26 status: 0x00
10:12:33:336 Erase task req-id: 26, type: 19 zcl seqno: 84 send time 0, profileId: 0x0104, clusterId: 0x0006
10:12:33:385 Node data 0x5c0272fffe694793 profileId: 0x0104, clusterId: 0x0006
10:12:33:395 0x5C0272FFFE694793: update ZCL value 0x01/0x0006/0x0000 after 0 s
10:12:33:404 enqueue event event/checkgroupanyon for /groups/65520
10:12:33:490 Node data 0x60a423fffe881173 profileId: 0x0104, clusterId: 0x0008
10:12:33:503 0x60A423FFFE881173: update ZCL value 0x01/0x0008/0x0000 after 0 s
10:12:33:512 ZCL attribute report 0x60A423FFFE881173 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
10:12:33:520 	payload: 000020fe
10:12:33:577 enqueue event event/validgroup for /sensors/28
10:12:33:585 [INFO] - Button 2002 - Shutters central remote switch, broadcast to: 0x0007, endpoint: 0x01, cluster: WINDOW_COVERING (0x0102), action: Move down (with on/off), payload: None, zclSeq: 3
10:12:33:593 enqueue event state/buttonevent for /sensors/28
10:12:33:600 enqueue event state/lastupdated for /sensors/28
10:12:33:608 Force binding of attribute reporting for sensor Remote Switch
10:12:33:615 queue binding task for 0x0004740000A9AFF4, cluster 0x0102
10:12:33:645 Websocket 192.168.0.14:58632 send message: {"e":"changed","id":"28","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2021-04-23T08:12:33.600"},"t":"event","uniqueid":"00:04:74:00:00:a9:af:f4-01-0102"} (ret = 165)
10:12:33:653 Websocket 192.168.0.3:49885 send message: {"e":"changed","id":"28","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2021-04-23T08:12:33.600"},"t":"event","uniqueid":"00:04:74:00:00:a9:af:f4-01-0102"} (ret = 165)
10:12:33:669 discard sensor state push for 28: state/lastupdated (already pushed)
10:12:34:081 enqueue event config/localtime for /config/
10:12:34:096 rule event /config/localtime: 10:12:33.081 -> 10:12:34.080 (999ms)
10:12:34:155 poll node 60:a4:23:ff:fe:8c:cc:51-01
10:12:34:163 Poll light node Kerstmis bol
10:12:34:209 Poll APS request to 0x60A423FFFE8CCC51 cluster: 0x0006 dropped, values are fresh enough
10:12:34:581 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
10:12:35:081 enqueue event config/localtime for /config/
10:12:35:088 Wait 1s till query finished
10:12:35:095 poll node 60:a4:23:ff:fe:88:11:73-01
10:12:35:101 Poll light node Bureau touch scherm
10:12:35:115 rule event /config/localtime: 10:12:34.080 -> 10:12:35.081 (1001ms)
10:12:35:144 Poll APS request to 0x60A423FFFE881173 cluster: 0x0006 dropped, values are fresh enough
10:12:35:194 Bind response success for 0x0004740000a9aff4 ep: 0x01 cluster: 0x0102
```

kvanisterdael, 2021-04-23:
If I check the Websockets now, I can see:
- 1002: up short push
- 2002: down short push
- 3002: up long push release
- 3003: down long push release  

Double click has no output

Smanar, 2021-04-23:
I think it s normal, I don't see what we can have on a double click on a shutter command.
But if a command exist you will see it in log

>[INFO] - Button ....................................

closed 2021-05-22 by github-actions[bot]