dresden-elektronik/deconz-rest-plugin #4737: HY367RT Tuya Thermostat

katekyonni, 2021-04-09:
## Device

- Product name: HY367RT
- Manufacturer: _TZE200_cwnjrr72
- Model identifier: TS0601
- Device type :
  - Thermostat

## Screenshots

[image]

[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

Here is a listing of the device on Aliexpress: https://www.aliexpress.com/item/1005002225828787.html?spm=a2g0s.9042311.0.0.2e5c4c4dyb1U2D

I found this issue on the zigbee2mqtt Github where the device was added: https://github.com/Koenkk/zigbee2mqtt/issues/6873

I hope this is all you need, if not let me know.
Thank you!

Mimiix, 2021-04-09:
Please use search next time. This is a duplicate.

Mimiix, 2021-04-09:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4675

closed 2021-04-09 by Mimiix