dresden-elektronik/deconz-rest-plugin #4739: RaspbeeII Pinout confirmation

adambrdn, 2021-04-09:
## Describe the question or issue you are having

The Raspbee device is listed on pinout.xyz but I was looking for either :

- Confirmation that the pinout is the same for Raspbee II devices. 

- Details of what the pinout actually is. 

If it is different, I am happy to contribute to pinout.xyz and update, but wanted to confirm here first. I have searched and the usermanual for the Raspbee device states what the pins are, but there's no such document for the Raspbee II device (including in the FCC filings). 

Grateful for any response.

Thanks

## Screenshots

Not applicable

## Environment

Not applicable

## deCONZ Logs

Not applicable

## Additional context

Not applicable

ebaauw, 2021-04-09:
I haven't seen any doc either, but I understand it uses the same pins as the RaspBee (GPIO 14 and 15 for the serial port; GPIO 17 for reset; don't know what GPIO 18 is used for), _and_ GPIO 2 and GPIO 3 for I2C to the real-time clock.

adambrdn, 2021-04-10:
Excellent. Thanks for the info - can these be pinouts confirmed via the codebase?

ebaauw, 2021-04-10:
Not via the REST API code, as that doesn’t communicate directly with the RaspBee.  The deCONZ core and the device firmware are not open source, so I really wouldn’t know.

The I2C GPIOs can be confirmed indirectly through the install instructions for the real-time clock function; the serial port GPIOs indirectly through the serial interface documentation.  I’m not even sure if deCONZ uses the other two GPIOs, or if they’re needed only by the firmware flasher (which isn’t open source either).  But I can tell from experience you don’t want to mess with GPIO 17 while deCONZ is running - managed to render the RaspBee unresponsive, requiring a power cycle.

closed 2021-05-09 by github-actions[bot]