dresden-elektronik/deconz-rest-plugin #4747: Philips Hue Go v2: nonstop connected as zigbee router, battery only lasts 2 to 3 days

814d3, 2021-04-10:
## Describe the question or issue you are having

Hi everyone, I'm new to Github, hopefully I created this issue/question correct.
I'm using deconz for several months now and apart of some drops of my FLS-PP everything seems fine except my Philips Hue Go (new version (with BT)).
I connected it via Phoscon without problems and want to use it as a night light, triggered by a motion sensor. Basically this works. But as the manual mentions, the lamp will go in standby after 6? hours and will not be reachable via zigbee/BT. So I setup a scene which will be activated each 5 hours for 1 minute to keep the lamp reachable. So it acts as a normal zigbee router. Maybe this is the problem, that the battery lasts only 2 to 3 days - active (glowing) for about only 20 minutes.
Is there a way to maybe reduce the router activity of the lamp to save energy? The lamp is only 3 months old, why I exclude a faulty battery.
Please ask if you need more information for investigation.

Thanks in advance.

## Screenshots

## Environment

 - Host system: Raspberry Pi3b
 - Running method: Raspbian, deCONZ via apt-source, currently trying HomeAssistant in Docker (no deCONZ Plugin)
 - Firmware version: 26660700
 - deCONZ version: 2.10.04
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: USB-Flashdrive, USB-Headset

## deCONZ Logs

## Additional context

ebaauw, 2021-04-10:
> Is there a way to maybe reduce the router activity of the lamp to save energy?

No, the Zigbee standard requires routers to have their radio powered at all times.

I have little experience with battery-powered routers; I used to have a Heiman Siren with a backup-battery, but it's broken.  I used to continue to run as router when disconnected from the mains, until the backup battery depleted, at which point the device simply shut down.  I suppose the same would hold for the Hue Go.

So either control it by Bluetooth while battery powered; or control it through Zigbee/deCONZ while keeping it connected to mains/the charger.  You might use a smart plug and some automations to provide mains power to the Go, only when the battery runs low.

814d3, 2021-04-11:
@ebaauw thanks for your comment. Unfortunately there is no power plug in this area, why I chose the Hue Go. My hope was, that that device is more energy efficient. I'm not familiar with the zigbee protocols, but could the "Green Power" feature help in any way? At least, could the lamp somehow send it's battery state?

[image]

Greetings

ebaauw, 2021-04-11:
The green power endpoint is to support ZGP devices, nothing to do with the Go itself.

I don’t see a _Power Configuration_ cluster, so I don’t think it supports reporting the battery state.  Does it show in the Hue app, or in the Hue Bluetooth app?  If so, It might report the battery state through the _Power Descriptor_ or the 0xFC01 cluster, but we would probably need to sniff the communication between the Go and the Hue bridge to know how to support that from deCONZ.

814d3, 2021-04-12:
I installed the Hue Bluetooth App und paired with the lamp, but there are no information related to the battery state. (It reported, that the firmware is up to date) So it seems to be hopeless. 
At least I could change the Li-Ion Batteries to some with a little bit higher capacity, but this will only extend the reachability by 1/2 to 1 day.

closed 2021-05-11 by github-actions[bot]

sambokai, 2022-05-10:
Any updates on this? Specifically the battery state reporting via zigbee

Mimiix, 2022-05-10:
@sambokai Please refer to this comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4747#issuecomment-817343261

Unless there has been an firmware update changing this, nothing has changed.