dresden-elektronik/deconz-rest-plugin #4750: Device request: TUYA Zigbee Smart Led Controller _TZ3000_qqjaziws

vanarebane, 2021-04-11:
## Device

- Product name: TUYA Zigbee Smart Led Controller for RGBCW Strip 
- Manufacturer: _TZ3000_qqjaziws
- Model identifier: TS0505B
- Device type :
  - Light

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control
[image]

### Color Control
[image]

### ZLL
[image]

### OTAU
[image]

### Time
(Empty)

Smanar, 2021-04-11:
Hello, this device don't work natively ?

vanarebane, 2021-04-11:
> Hello, this device don't work natively ?

Hi, This device does not show up in Phoscon when pairing, but device appears in deCONZ. Does not show up in Home Assistant (trough Rest API I guess)

vanarebane, 2021-04-11:
Oh, it appeared as Extended Light after re-pairing another switch

closed 2021-05-10 by github-actions[bot]