dresden-elektronik/deconz-rest-plugin #4751: Device name

vanarebane, 2021-04-11:
## Device

- Product name: Tuya ZigBee 3 Button Wireless 9 Scene Switch Push Button
- Manufacturer: _TZ3000_rrjr1q0u
- Model identifier: TS0043
- Device type : 
  - Switch

[image]

## Screenshots

[image]

## 1 Switch
### Basic
[image]

### Time
[image]

### Power Configuration
[image]

### On/Off
[image]

### OTAU
[image]

## 2 Switch
### Power Configuration
[image]

### On/Off
[image]

### Basic
[image]

## 3 Switch
### Power Configuration
[image]

### On/Off
[image]

Smanar, 2021-04-11:
Hello do you have a linux machine to test code ? Else I think it will work, nothing special (except a typo in the "random" name.

vanarebane, 2021-04-12:
Hi, I don't have Linux machine. I could find time to run a virtual Linux. Do you have any instructions for step-by-step on how to test?

Smanar, 2021-04-12:
Sure.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_33 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then re-include the device.

closed 2021-04-25 by manup