dresden-elektronik/deconz-rest-plugin #4757: SONOFF SNZB-04 Re-Pairing?

majorsl, 2021-04-11:
## Describe the question or issue you are having

I am unsure if this is a bug report, me not doing something right, or just a quirk of the device. I have several of door/window sensors. When the battery dies, I'd expect to replace the battery and just have it show up again in Phoscon. However, almost always I end up have to re-pair or delete and re-add the sensor.

Does anyone have these and is this common, or is there something I'm missing after a battery replacement?

I have about an equal number of Centralite door/window sensors and just replacing a dead battery is good enough to get them going again without having to fiddle with pairing processes.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.09.03
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mschul, 2021-04-13:
I am facing the same problem as you. Also with the Sonoff Buttons SNZB-01.

Mimiix, 2021-04-13:
Would be intresting to see some logs on when this happens

SwoopX, 2021-04-13:
I doubt this is deconz related. Especially cheap devices do not seem to store the network configuration persistently in memory or seem to fail on the rejoin.

majorsl, 2021-04-13:
Thanks all for the suggestions so far. I think it might be more related to the device specifically, although if I go in with VNC I can see it (or a representation of it) connected to the controller or the nearest node. It does seem to be "talking".

I am moving away from the SONOFF sensors in favor or Aqara which I have much better luck with.  I think the SONOFF ones are best used for Windows that aren't open/closed often so you get the longest life out of the battery once paired.  If I had it to do again, I wouldn't have purchased them at all.

closed 2021-05-12 by github-actions[bot]

realflash, 2022-01-09:
Could someone comment on which logs you would want to see? Sometimes they seem to be OK, sometimes they require re-pairing.