dresden-elektronik/deconz-rest-plugin #4759: tint LED-Globeform white+color

cagapisto, 2021-04-11:
## Device

- Product name: tint LED-Globeform white+color
- Manufacturer: MLI
- Model identifier: tint-ExtendedColor
- Device type:
  - Light

https://www.mueller-licht.de/produktinformationen/artikel/?q=404036

## Screenshots

###Node Info
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Color Control
[image]
[image]
[image]
[image]

### Other clusters that are not mentioned above
[image]
[image]

cagapisto, 2021-05-10:
The device request is still existing, would be great if the issue could be reopened.
If there's anything I can do to support in dealing with the request, I'll gladly help (if it's in my skill set).

Mimiix, 2021-05-14:
Whats not working on it @cagapisto ?

cagapisto, 2021-05-19:
I'm sorry, I think I've misread the thread for 'supported devices' in a way as "most lights work with the generic implementation, but 'official' supported devices (those listed in the thread) have a better implementation (regarding color accuracy or such) than the out-of-the-box ones".

If this isn't the case, can I maybe ask why there are some devices on the list of supported devices (e.g. the Müller Licht tint devices), even if they are supported through the generic implementation while others are not listed?
Maybe another stupid question, but smart home devices really aren't one of my strengths: Is there, despite normal usage, a set of tests I can perform with the bulb to see if really everything works correctly?

The only things I found that are not quite working with the bulb are:
- Starting scenes that I've saved
- Schedules for switching the light on or off
I'm mostly using the Hue Essentials App to interact with the bulb, so the problem might be on the side of the App and not with deconz.

closed 2021-06-18 by github-actions[bot]