dresden-elektronik/deconz-rest-plugin #4760: Xiaomi Multisensor still not visible in Deconz (shows up in Phoscon and HA)

fribse, 2021-04-12:
## Describe the bug

I have two Xiaomi Multipurpose sensor's added to the environment. At first they were added some time ago, and showed up nicely in Phoscon. Then I had to readd one to the zigbee, and then it doesn't show up in Deconz, though it does show in Phoscon. It still showed the old one with the right name and all.
With the later version, it now doesn't show the 'old' multisensor either.

## Steps to reproduce the behavior

Get a Xiaomi Multisensor
Add it as zigbee device
It shows up in Phoscon and Home Assistant
It does NOT show in Deconz.

## Expected behavior

It should show up in Phoscon so it can be renamed and managed there.

## Screenshots

[image]
[image]
[image]
[image]
[image]

## Environment

 - Host system: NUC with PVE
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26670700)
 - deCONZ version: (2.10.04)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, AEOTEC Z-wave stick

## deCONZ Logs

## Additional context

Smanar, 2021-04-12:
>It shows up in Phoscon and Home Assistant
>It does NOT show in Deconz.
>It should show up in Phoscon so it can be renamed and managed there.

And your capture show it in deconz ....

There is no mistake ?

fribse, 2021-04-12:
I corrected it, sorry! Running a fever here :-( (not covid)

SwoopX, 2021-04-12:
This is (again) not a Xiaomi device...

fribse, 2021-04-13:
Arh, this is all bad, sorry, shouldn't be doing this when I'm this sick...

closed 2021-04-13 by fribse

Mimiix, 2021-04-13:
Take care!