dresden-elektronik/deconz-rest-plugin #4762: Xiaomi power plug SP-EUC01 not showing all values

VaskoJiri, 2021-04-12:
## Describe the bug

Hi, I have SP-EUC01 plug. In Node-Red i see only power, no voltage or current values.
The problem is int he API. I tried get data through the API and here is result:

"25": { "config": { "on": true, "reachable": true, "temperature": null }, "ep": 1, "etag": "6438984bf195bb5d8113db189b2b29d0", "lastseen": "2021-04-12T20:06Z", "manufacturername": "LUMI", "modelid": "lumi.plug.maeu01", "name": "Consumption 25", "state": { "consumption": 359, "lastupdated": "2021-04-12T20:06:05.734" }, "swversion": "09-10-2019", "type": "ZHAConsumption", "uniqueid": "04:cf:8c:df:3c:89:ed:9d-01-0702" }, "26": { "config": { "on": true, "reachable": true, "temperature": null }, "ep": 1, "etag": "dee8ebd7878775223ea0adda6e46f17b", "lastseen": "2021-04-12T20:06Z", "manufacturername": "LUMI", "modelid": "lumi.plug.maeu01", "name": "Power 26", "state": { "current": null, "lastupdated": "2021-04-12T20:06:06.620", "power": 49, "voltage": null }, "swversion": "09-10-2019", "type": "ZHAPower", "uniqueid": "04:cf:8c:df:3c:89:ed:9d-01-0b04" }

It is the same issue as #3116 I asssume.
Is there still a bug?

## Steps to reproduce the behavior

It is happen all the time.
## Expected behavior

I will see all values.
## Screenshots

## Environment

 - Host system: Raspberry Pi 4b - 4G RAM
 - Running method: Marthoc Docker container 
 - Firmware version: (26680700)
 - deCONZ version: (2.10.04)
 - Device:  ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

## Additional context

SwoopX, 2021-04-12:
The plug doesn't support voltage and current.

VaskoJiri, 2021-04-13:
Thank you, i didn't know that.

closed 2021-04-22 by manup