dresden-elektronik/deconz-rest-plugin #4766: v2.11.0-beta in docker - after mi button click "Segmentation fault (core dumped)"

SAOPP, 2021-04-14:
## Describe the bug

Guys, on the last beta, I noticed a crash in the container after clicking on my button from Xiaomi. Moreover, this is repeated several times in a row until the connection to the stick is restored... judging by the log.

Yesterday I noticed this on another stick, similar behavior was after clicking on the aqara double switch, there I have ubuntu x86 and docker. Later, I encountered similar behavior on my zigbee gateways (rpi + raspbian + docker)

All my three sticks in production is conbee II.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi / PC / NAS
 - Running method: Raspbian / Ubuntu / Marthoc Docker container / Virtual Machine
 - Firmware version: 26580700
 - deCONZ version: 2.11.00 / 14.03.2021
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: сс2531

## deCONZ Logs

09:18:09:028 [INFO] - Button 1000 - lumi.sensor_switch, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Normal press, payload: 00001000, zclSeq: 251
Segmentation fault (core dumped)
## Additional context

I checked right now, and recorded the video of the log after I clicked the button.

https://user-images.githubusercontent.com/5647009/114664548-88791a00-9d04-11eb-8870-677599482890.mp4

Also added screenshoot of my telegram notifier 'bout status of the container, as u can see the container is start/down status some time after clicking to the button.

[image]

SAOPP, 2021-04-14:
Right now back to `2.10.04 / 14.03.2021`, the issue is gone.

Mimiix, 2021-04-14:
Forwarded it to @manup and @SwoopX

SAOPP, 2021-04-14:
I think it would also beI nice if @merdok and @phdelodder drop by here.

phdelodder, 2021-04-14:
This seems to be an issue with deconz it self and not the container

Mimiix, 2021-04-14:
@SAOPP If you felt it was a docker issue, you should've posted this in the docker repository 😅

SAOPP, 2021-04-14:
No guys, it's deconz issue 101 percent.

SAOPP, 2021-04-14:
Earlier I said already in the Discord that my container dropped on some of the latest versions. The behavior was the same, then I thought that this was probably due to lack of memory (although there is plenty of it), and closed the question for myself. But then it happened on another instance, with a completely different architecture.

SAOPP, 2021-04-14:
Well, to summarize, we can consider the fact that having rolled back to the version below, I solved the issue that appeared on yesterday's beta version. 

I am ready to take part in testing, or provide any additional data so that we can find the reason for this behavior.

SwoopX, 2021-04-14:
Hey,

could you try running deconz through a debugger and provoke the crash?

**Headless**
`gdb --args /usr/bin/deCONZ -platform minimal`

**With GUI**
`gdb --args /usr/bin/deCONZ`

You might need to add further deconz parameters as to your requirements. Once this is started, press `r` to let it run and then try to crash deconz. When deconz crashes, it should give you the backtrace, relevant for us to pinpoint the issue. In doubt, let's discuss that in Discord ;)

mcormick, 2021-04-14:
I have the same issue with the 2.11.0-beta in a Docker container on Synology (Conbee2 firmware 26680700), so it seems no isolated case. Had force restarts of the container before I could even login to the VNC environment. I reverted back to 2.10.4 and all is well again.

SAOPP, 2021-04-14:
Well, as we can see, somewhere the "byte falls" in the wrong place. So the problem is exists - all that remains is to catch it.

SAOPP, 2021-04-14:
@mcormick buddy, please share your part of the logs when the container crashes.

mcormick, 2021-04-14:
So, the funny thing is that while trying to get you guys a log, 2.11.0-beta suddenly started to to not crash... I kid you not. Glad I saved a not-so-verbose log to show I got the same segmentation faults as TS. Note; somehow docker log is bottom-up.
[deCONZ.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6313152/deCONZ.txt)
I even reverted to the old 2.10.4 database and then started the beta again and it still works, so far. I enabled DEBUG_APS, DEBUG_ZCL and DEBUG_ZDP in case something happens. Is that what you require?

szulik, 2021-04-15:
Hi, Today I upgraded Docker container (Synology) - I use "latest" tag so the version is 2.09.  And I have exactly the same problem: when I push Aqara Smart Switch deConz crashes & restarts. 

Part of the log attached. I pushed the button at 17:53

2021-04-15 17:53:13,stdout,[marthoc/deconz] Starting deCONZ...

2021-04-15 17:53:12,stdout,Segmentation fault (core dumped)

2021-04-15 17:53:12,stdout,"19:53:12:393 [INFO] - Button 1000 - lumi.sensor_switch, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Normal press, payload: 00001000, zclSeq: 172
"

2021-04-15 17:53:12,stdout,19:53:12:392 APS-DATA.indication from unknown node 0x00158D0004601B4C

2021-04-15 17:53:12,stdout,"19:53:12:392 unknown node 0x00158D0004601B4C (0x6DB5), lqi: 255`

[deconz - error.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6320253/deconz.-.error.zip)

SwoopX, 2021-04-15:
Guys, it's great that you want to support in locating the problem! However, please note that anything else than a backtrace from the debugger is useless for further investigation, as deconz logs do not show the actual code line of the crash.

Please see here what's required https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4766#issuecomment-819403966

Falkvinge, 2021-04-16:
I am seeing this too. The whole thing just dies the moment I press a Xiaomi Smart Switch, and the problem appeared on an upgrade four days ago (when 2.11.0 appeared in the repos).

Environment: VM under Home Assistant, connected to dongle using VMWare. Worked fine up until this upgrade.

This is a real problem for me as much of my home automation depends on pressing these switches.

Falkvinge, 2021-04-16:
I tested downgrading to 2.10.04 (changing the software channel while I was at it), and the problem went away immediately.

Mimiix, 2021-04-16:
> @Falkvinge  This is a real problem for me as much of my home automation depends on pressing these switches.

That's why it's a beta ;) To find things like this. 

**To all responding after this comment: take note of Swoop's comment. i'll remove other comments if they are unrelated/#metoo related**

phdelodder, 2021-04-16:
If you are using 2.11.00 with the ubuntu image and you are experiencing issues, PR marthoc/docker-deconz/pull/387 should fix it for you guys

alinelena, 2021-04-21:
@phdelodder I am using marthdoc/deconz:tatest (2.11.00 and conbee II 2668...) and I see the issue I will try to see if I can get you a backtrace... is reliable like hell... I need to understand which packages need installed on top of what is in the image.

Falkvinge, 2021-04-21:
Nota Bene: I'm seeing the same thing *without Docker*. I'm therefore highly sceptical that changing the Docker image will resolve the root cause.

alinelena, 2021-04-21:
```
Thread 1 "deCONZ" received signal SIGSEGV, Segmentation fault.
0x00007f05485396b0 in deCONZ::Node::nodeDescriptor() const () from /usr/lib/libdeCONZ.so.1
(gdb) bt
#0  0x00007f05485396b0 in deCONZ::Node::nodeDescriptor() const () from /usr/lib/libdeCONZ.so.1
#1  0x00007f0542ff8d98 in DeRestPluginPrivate::checkSensorButtonEvent(Sensor*, deCONZ::ApsDataIndication const&, deCONZ::ZclFrame const&) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#2  0x00007f054303a954 in DeRestPluginPrivate::apsdeDataIndication(deCONZ::ApsDataIndication const&) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#3  0x00007f0543194826 in DeRestPluginPrivate::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#4  0x00007f05473cf7f6 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00000000004dcf95 in ?? ()
#6  0x000000000049194f in ?? ()
#7  0x00000000004dd5c5 in ?? ()
#8  0x00007f05473cf7f6 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00000000004db865 in ?? ()
#10 0x0000000000462bc1 in ?? ()
#11 0x0000000000464403 in ?? ()
#12 0x00000000004dc28d in ?? ()
#13 0x00007f05473d0072 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007f0547edd4c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007f0547ee4970 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007f05473a6489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007f05473a946b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007f05473f8103 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007f0546216f2e in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007f05462171c8 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007f054621725c in g_main_context_iteration () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007f05473f7727 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007f0543dbd081 in ?? () from /usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqminimal.so
#24 0x00007f05473a515b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007f05473ad132 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x000000000043722c in ?? ()
#27 0x00007f0546c5d09b in __libc_start_main (main=0x4361f0, argc=3, argv=0x7ffdf38bdfb8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffdf38bdfa8) at ../csu/libc-start.c:308
#28 0x0000000000437e29 in ?? ()
```

SwoopX, 2021-04-21:
Thanks!!! We're already on it and spotted the place.

alinelena, 2021-04-21:
no problem a pitty I could not get it earlier.

some other comments in general on the container.

it seems to run as use root... not a terrible good idea in my book... 
worst seems to be debian buster but deconz is some random ubuntu 2.11.00-ubuntu-xenial-beta amd64 [installed,local]

SwoopX, 2021-04-21:
> some other comments in general on the container.

We're out on this one I'm afraid 😄 

Could you try a workaround on the troublesome device? Read the node descriptor from it in deconz GUI by right clicking on the device in question -> read node descriptor. Please take note that the device must be awake for this, so waking the device will probably force a crash. Another option would be to rejoin it,

alinelena, 2021-04-21:
ok no issue... shall I start deonz in gdb or normal?

as normal after reading the node from gui seems to work properly... in the sense does what it is supposed to do.

phdelodder, 2021-04-22:
> no problem a pitty I could not get it earlier.
> 
> some other comments in general on the container.
> 
> it seems to run as use root... not a terrible good idea in my book...
> worst seems to be debian buster but deconz is some random ubuntu 2.11.00-ubuntu-xenial-beta amd64 [installed,local]

This is indeed the case. you can create either a PR or and issue in the docker repostitory: https://github.com/marthoc/docker-deconz

Mimiix, 2021-04-22:
This should be fixed in the new beta. Release is planned tommorow!

SwoopX, 2021-05-09:
As we got confirmation that this bug has been fixed, I'm closing this one. Feel free to raise a new issue in case you experience anything different.