dresden-elektronik/deconz-rest-plugin #4767: Sengled Smart Light Bulb E21-N1EA

deviantintegral, 2021-04-14:
## Device

- Product name: Sengled Smart Light Bulb: https://www.amazon.ca/gp/product/B0899LJHVV/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
- Manufacturer: sengled
- Model identifier: E21-N1EA
- Device type :
  - Light

## Screenshots

### Basic

[image]

### Identify

[image]

### Groups

[image]

[image]

### Scenes

[image]
[image]

[image]

### On/Off

[image]

[image]

### Level Control

[image]
[image]

### Color Control

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Simple Metering
[image]

### Diagnostics

[image]

deviantintegral, 2021-04-14:
For anyone with these bulbs; they somewhat work, but the colour control isn't totally accurate for all colour modes. HSV in particular gives you wrong colours.

deviantintegral, 2021-05-07:
_bump_

However, since posting this I switched my home setup over to ZHA. I have another install that is using deconz, but my access to it is sporadic so testing may be tricky.

mxracer95, 2022-11-16:
Re-Requesting support for this bulb. It is recognized as a color dimmable light rather than an extended color light.

Mimiix, 2022-11-16:
Sure reopening

mxracer95, 2022-11-28:
Bumping this request since reopening.

gizmotronic, 2022-12-12:
I just bought a handful of these and have the same problem. [Manually setting the color temperature](/dresden-elektronik/deconz-rest-plugin/issues/4833#issuecomment-894372699) through the API works as expected, even though this is a different model.

mxracer95, 2022-12-12:
I did create a DDF for this using my best understanding on how to do it. It seems to make the bulb work properly in Phoscon but then it doesn't get the RGB options over into Homeseer (via Jowihue plugin) I only get ColorTemperature. I bought a Philips Hue White and Color Ambiance bulb to test and both RGB and CT are read into Homeseer so I still think the issue is with deConz.

Here's my DDF for the Sengled E21-N1EA:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "sengled",
  "modelid": "E21-N1EA",
  "vendor": "sengled",
  "product": "Smart LED E21-N1EA",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/E21-N1EA.json",
  "subdevices": [
    {
      "type": "$TYPE_COLOR_TEMPERATURE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/colorcapabilities",
          "refresh.interval": 84000,
          "default": 16
        },
        {
          "name": "config/ctmax",
          "refresh.interval": 84000,
          "default": 455
        },
        {
          "name": "config/ctmin",
          "refresh.interval": 84000,
          "default": 200
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/colormode",
          "refresh.interval": 300
        },
        {
          "name": "state/ct",
          "refresh.interval": 5
        },
        {
          "name": "state/effect",
          "default": "none"
        },
        {
          "name": "state/hue"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "state/sat",
          "refresh.interval": 5
        },
        {
          "name": "state/x",
          "refresh.interval": 5,
          "read": {
            "at": "0x0003",
            "cl": "0x0300",
            "ep": 0,
            "fn": "zcl"
          }
        },
        {
          "name": "state/y",
          "read": {
            "at": "0x0004",
            "cl": "0x0300",
            "ep": 255,
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0300",
      "report": [
        {
          "at": "0x0007",
          "dt": "0x21",
          "min": 154,
          "max": 500,
          "change": "0x00000001"
        },
        {
          "at": "0x0003",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0004",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0008",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}

gizmotronic, 2022-12-13:
I added this DDF, reset my bulb, re-paired it, and verified that it was properly recognized. It seems to work properly using the Phoscon sliders. However, when exposed to HomeKit via homebridge-deconz, my bulb reverts to RBG whether I set RGB or color temperature values.

Mimiix, 2022-12-13:
@gizmotronic  Good to see some progress! That seems to be a issue in Homebrdige, because Phoscon uses the same API as Homebridge does.

mxracer95, 2022-12-13:
To update, the DDF I posted a few comments up does seem to work. In Homeseer using the JowiHue plugin (v4.0.8.3), the plugin doesn't properly import all the settings so custom settings had to be created in the JowiHueImports.xml file. For anyone looking, I created a post on the Homeseer forums which you can find by searching for the bulb model in HS4, Plugins, JowiHue.

BabaIsYou, 2023-01-10:
If the DDF seems to be fine, may be you could open a Pull Request (PR) to have it included into a next release ?

closed 2023-02-14 by github-actions[bot]