dresden-elektronik/deconz-rest-plugin #4770: Siterwell TRV GS361A-H04 / _TZE200_ zivfvd7h / TS0601

bawe26, 2021-04-15:
## Device

normal TS0601 TRV with just a new manufacturer

- Product name: Siterwell GS361A-H04
- Manufacturer: _TZE200_ zivfvd7h
- Model identifier: TS0601
- Device type : Thermostat

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-04-15:
Could you please add all clusters?

bawe26, 2021-04-15:
[image]

Groups
[image]
[image]

Scenes
[image]
[image]
[image]

Tuya
[image]

OTAU
[image]

Time
[image]

Smanar, 2021-04-15:
Have you tried to include on deconz ? on wich one version ?
Remember for the moment you need 2 inclusions without deletion between both.

I say that because I can see that in code
```
    {"_TYST11_zivfvd7h", "ivfvd7h", "Siterwell", "Tuya_THD GS361A-H04 TRV"},
    {"_TZE200_zivfvd7h", "TS0601", "Siterwell", "Tuya_THD GS361A-H04 TRV"},
```

bawe26, 2021-04-15:
I have seen the mentioned devices within the code, but somehow they are not recognized correctly.

I am using a ConBee II stick on a RPI4 with Firmware 26680700. I am running the marthoc docker package with deconz version 2.11.00 

I have done multiple inclusions with different devices. What changes is just the black bold identifier name. Opening the network via webapp and via deconz has been tried.

Therefore I thought there might be something different so I opened the new request.

bawe26, 2021-04-15:
just tested also deconz version 2.10.04 which shows the same behavior

Smanar, 2021-04-15:
Right, have found something.

Do you have another linux machine to compile and test code ?

bawe26, 2021-04-19:
sorry for getting back that late. I can confirm that the device has been recognized as a thermostat. Everything seem to be working

bawe26, 2021-04-20:
I still found some issue with the HomeAssistant integration. I confirmed that wih api calls, so it seems to be a deCONZ issue.
It seems that some config fields are missing and therefore setting the heatpoint is not working.
Here is the api response:
```
"config": {
"heatsetpoint": 2100,
"offset": null,
"on": true,
"reachable": true,
"schedule": {},
"schedule_on": null
},
"ep": 1,
"etag": "2b4ecc20783f7a10121e4659e73500df",
"lastseen": "2021-04-20T08:48Z",
"manufacturername": "_TZE200_zivfvd7h",
"modelid": "TS0601",
"name": "Thermostat 21",
"state": {
"lastupdated": "2021-04-20T08:48:25.779",
"on": false,
"temperature": 2510
},
"type": "ZHAThermostat",
"uniqueid": "84:71:27:ff:fe:c2:f9:21-01-0201"
```

Changing the temperature on the device itself is working and is reflected immediately in the api. Also the mode is missing (manu, anti freezing and auto), window open, child lock etc. are not there.

Smanar, 2021-04-20:
He, my bad I think,this device have no special feature activated.
From that I have read on other project this device is the same than the "Tuya_THD HY368 TRV" So I have mimic the optionnal features.

Can you make a test with the updated code ?
No need to re-include the device (I think)

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

bawe26, 2021-04-22:
Everything seems to be working. Setting modes and temperature works. Battery state is shown. Inclusion in HA is fine.
Thanks for your support

closed 2021-04-22 by manup

jghaanstra, 2021-11-11:
@bawe26 I paired this device with Deconz version 2.13.1 yesterday but I'm confused about how it shows up in Home Assistant.

- The devices is shown in HA with havc mode off and heat but it can not be changed to heat, it always shows as off
- In HA it shows the modes auto, boost, comfort, complex, eco, holiday, manual while the device only supports manual, auto and freeze (holiday?).
- there is no way to control the open window functionality of the valve (not even through the button on the valve)
- It's not shown as a device in Phoscon.

Could you comment on this in relation to your experiences with this valve thusfar.

Smanar, 2021-11-11:
Phoscon don't support TRV
I don't see the openwindow command in the JSON here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4770#issuecomment-823121459 But i see it on the code, can you show your device JSON ?

Your device is _TZE200_zivfvd7h ?

When you try "heat" it don't change on the TRV but it change on HA ?

jghaanstra, 2021-11-11:
> Phoscon don't support TRV
Kinda figured but thanx for confirming.

The device is indeed `_TZE200_zivfvd7h` / `TS0601`. The open window mode was more a question than an expectation that it was supported. It would be nice to know how to change this without having the Tuya Zigbee bridge and not being able to control it with deCONZ / HA. Here is the JSON from deCONZ. Not sure why mode is `null`.

```
"84": {
        "config": {
            "battery": 100,
            "heatsetpoint": 2200,
            "locked": false,
            "mode": null,
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": false,
            "setvalve": true,
            "windowopen_set": true
        },
        "ep": 1,
        "etag": "164b317a9aad70214edaa1f10d50b388",
        "lastannounced": "2021-11-11T11:11:25Z",
        "lastseen": "2021-11-11T14:19Z",
        "manufacturername": "_TZE200_zivfvd7h",
        "modelid": "TS0601",
        "name": "Radiator keuken",
        "state": {
            "lastupdated": "2021-11-11T13:20:39.776",
            "lowbattery": false,
            "on": false,
            "temperature": 2020,
            "valve": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "84:fd:27:ff:fe:a2:55:0b-01-0201"
    }
```

And this is from HA.

```
{
                "entity_id": "climate.radiator_keuken",
                "config_entry_id": "489b4e5cd30b4eae8b668c063fb61cba",
                "device_id": "7446688cbbf7895b50e2d32c77a4e04e",
                "area_id": null,
                "unique_id": "84:fd:27:ff:fe:a2:55:0b-01-0201",
                "platform": "deconz",
                "name": null,
                "icon": "mdi:radiator",
                "disabled_by": null,
                "capabilities": {
                    "hvac_modes": [
                        "heat",
                        "off"
                    ],
                    "min_temp": 7,
                    "max_temp": 35,
                    "preset_modes": [
                        "auto",
                        "boost",
                        "comfort",
                        "complex",
                        "eco",
                        "holiday",
                        "manual"
                    ]
                },
                "supported_features": 17,
                "device_class": null,
                "unit_of_measurement": null,
                "original_name": "Radiator keuken",
                "original_icon": null,
                "entity_category": null
            },
```

Smanar, 2021-11-11:
I have checked the code again, and I don't see problem

On this device the "windows open" feature is just a on/off setting using the windowopen_set, and on your JSON it seem it s enabled.

For the "mode" issue I need some logs with "info" and "info_l2" when you make the command using HA.

jghaanstra, 2021-11-11:
> On this device the "windows open" feature is just a on/off setting using the windowopen_set, and on your JSON it seem it s enabled.

Missed that, I was able to update the window open mode using the deCONZ REST API. Would still be nice if it was available as entity in HA but I can work with this.

> For the "mode" issue I need some logs with "info" and "info_l2" when you make the command using HA.

This is what I could find which seems related to switching the hvac mode from HA.

```17:08:23:630 Websocket 172.33.0.1:40098 send message: {"config":{"battery":100,"heatsetpoint":2350,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"84","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:a2:55:0b-01-0201"} (ret = 297)
......
......
17:08:25:917 don't create binding for attribute reporting of sensor Radiator keuken
```

Smanar, 2021-11-12:
For the mode command you need to see something like that in log

`14:54:03:691 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x6A, data: 0000`

With 0000 for "auto" 0002 for "off" and 0x0001 for "heat"

jghaanstra, 2021-11-12:
I see something like that when changing the desired temperature from HA.
```
18:34:13:705 Websocket 172.33.0.1:47094 send message: {"attr":{"id":"84","lastannounced":"2021-11-11T11:11:25Z","lastseen":"2021-11-12T18:34Z","manufacturername":"_TZE200_zivfvd7h","modelid":"TS0601","name":"Radiator keuken","productid":"Tuya_THD GS361A-H04 TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"84:fd:27:ff:fe:a2:55:0b-01-0201"},"e":"changed","id":"84","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:a2:55:0b-01-0201"} (ret = 390)
18:34:13:706 Websocket 172.33.0.1:49830 send message: {"attr":{"id":"84","lastannounced":"2021-11-11T11:11:25Z","lastseen":"2021-11-12T18:34Z","manufacturername":"_TZE200_zivfvd7h","modelid":"TS0601","name":"Radiator keuken","productid":"Tuya_THD GS361A-H04 TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"84:fd:27:ff:fe:a2:55:0b-01-0201"},"e":"changed","id":"84","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:a2:55:0b-01-0201"} (ret = 390)
18:34:13:731 Poll APS request to 0x00158D0001DBF6E3 cluster: 0x0006 dropped, values are fresh enough
18:34:13:751 skip create link for 0x9266 (lqi: 62) - 0x16BF (lqi: 0)
18:34:14:176 Erase task req-id: 43, type: 37 zcl seqno: 44 send time 4, profileId: 0x0104, clusterId: 0xEF00
18:34:14:273 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 002c02020004000000e6
18:34:14:274 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 002c02020004000000e6
18:34:14:274 Tuya debug 5 : Status: 0 Transid: 44 Dp: 514 (0x02,0x02) Fn: 0 Data 230
18:34:14:404 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 002c02020004000000e6
18:34:14:404 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 002c02020004000000e6
18:34:14:405 Tuya debug 5 : Status: 0 Transid: 44 Dp: 514 (0x02,0x02) Fn: 0 Data 230
18:34:14:594 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 002c02020004000000e6
18:34:14:594 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 002c02020004000000e6
18:34:14:595 Tuya debug 5 : Status: 0 Transid: 44 Dp: 514 (0x02,0x02) Fn: 0 Data 230
```
There is no such log when changing the mode through.
```
18:34:28:311 Websocket 172.33.0.1:47094 send message: {"config":{"battery":100,"heatsetpoint":2300,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"85","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:0e:70:62-01-0201"} (ret = 297)
18:34:28:312 Websocket 172.33.0.1:49830 send message: {"config":{"battery":100,"heatsetpoint":2300,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"85","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:0e:70:62-01-0201"} (ret = 297)
18:34:28:835 don't create binding for attribute reporting of sensor Radiator keuken
18:34:29:180 poll node 00:15:8d:00:01:db:f6:e3-02-000c
18:34:29:180 Poll ZHAPower sensor node Power 73
18:34:29:443 Poll APS request to 0x00158D0001DBF6E3 cluster: 0x0B04 dropped, cluster doesn't exist
18:34:30:179 poll node 00:15:8d:00:01:db:f6:e3-03-000c
18:34:30:180 Poll ZHAConsumption sensor node Consumption 74
18:34:30:442 Poll APS request to 0x00158D0001DBF6E3 cluster: 0x0702 dropped, cluster doesn't exist
18:34:31:035 skip create link for 0xB5BF (lqi: 158) - 0x16BF (lqi: 0)
18:34:31:179 poll node 00:15:8d:00:01:db:f6:e3-01
18:34:31:179 Poll light node Tablet
18:34:31:231 Poll APS request to 0x00158D0001DBF6E3 cluster: 0x0006 dropped, values are fresh enough
18:34:32:177 poll node 00:0b:57:ff:fe:bb:7d:01-01
18:34:32:178 Poll light node Keukenlamp 2
18:34:32:226 read attributes of 0x000B57FFFEBB7D01 cluster: 0x0006: [ 
18:34:32:227 0x0000 
18:34:32:227 ]
18:34:32:227 add task 49164 type 19 to 0x000B57FFFEBB7D01 cluster 0x0006 req.id 177
18:34:32:228 Poll APS request 177 to 0x000B57FFFEBB7D01 cluster: 0x0006
18:34:32:264 Poll APS confirm 177 status: 0x00
18:34:32:265 Erase task req-id: 177, type: 19 zcl seqno: 56 send time 0, profileId: 0x0104, clusterId: 0x0006
18:34:32:298 Node data 0x000b57fffebb7d01 profileId: 0x0104, clusterId: 0x0006
18:34:32:299 0x000B57FFFEBB7D01: update ZCL value 0x01/0x0006/0x0000 after 0 s
18:34:32:366 Master: read param with arg 0x19
18:34:32:389 Device TTL 4980 s flags: 0x7
18:34:33:280 poll node cc:86:ec:ff:fe:91:2c:ed-01
18:34:33:281 Poll light node Signaalversterker 1
18:34:33:355 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
18:34:33:880 don't create binding for attribute reporting of sensor Radiator keuken
18:34:34:172 poll node 08:6b:d7:ff:fe:03:e5:e2-01
18:34:34:173 Poll light node Plug Midas
18:34:34:222 read attributes of 0x086BD7FFFE03E5E2 cluster: 0x0006: [ 
18:34:34:222 0x0000 
18:34:34:223 ]
18:34:34:223 add task 49173 type 19 to 0x086BD7FFFE03E5E2 cluster 0x0006 req.id 188
18:34:34:223 Poll APS request 188 to 0x086BD7FFFE03E5E2 cluster: 0x0006
18:34:34:265 Poll APS confirm 188 status: 0x00
18:34:34:266 Erase task req-id: 188, type: 19 zcl seqno: 57 send time 0, profileId: 0x0104, clusterId: 0x0006
18:34:34:291 Node data 0x086bd7fffe03e5e2 profileId: 0x0104, clusterId: 0x0006
18:34:34:292 0x086BD7FFFE03E5E2: update ZCL value 0x01/0x0006/0x0000 after 0 s
18:34:35:031 poll node d0:cf:5e:ff:fe:ef:4f:01-01
18:34:35:032 Poll light node Signaalversterker 3
[... cut ...]
18:34:37:855 Poll light node Keukenlamp 3
18:34:37:904 read attributes of 0x000B57FFFE489B9A cluster: 0x0006: [ 
18:34:37:905 0x0000 
18:34:37:905 ]
18:34:37:905 Poll APS request 211 to 0x000B57FFFE489B9A cluster: 0x0006
18:34:37:967 Poll APS confirm 211 status: 0x00
18:34:37:967 Erase task req-id: 211, type: 19 zcl seqno: 59 send time 0, profileId: 0x0104, clusterId: 0x0006
18:34:38:028 Node data 0x000b57fffe489b9a profileId: 0x0104, clusterId: 0x0006
18:34:38:029 0x000B57FFFE489B9A: update ZCL value 0x01/0x0006/0x0000 after 0 s
```

Smanar, 2021-11-12:
You are not sending request at all.
Try with changing the heatpoint, you will logs starting by "Send Tuya request'

You have no error message somewhere (in HA) ? Else can you make a try direclty using the API ? https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/ or enable http log to see the HA request (but if it don't make one, you will see nothing)

jghaanstra, 2021-11-27:
Sorry for taking so long to get back on this.

> You are not sending request at all.

I guess that is the actual issue here. I have tried a couple of times trying to set the mode to "heat" from HA with the INFO and INFO_L2 log open in deCONZ. The only thing registered in the log is this.

```11:29:05:830 Websocket 172.33.0.1:51754 send message: {"config":{"heatsetpoint":2200,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"85","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:0e:70:62-01-0201"} (ret = 283)
```
Every time I try to set the mode this message appears again. When setting the heatpoint I do see the `Send Tuya request`.

```11:31:32:454 Send Tuya request 0x84FD27FFFE0E7062 : Dp_type: 0x02, Dp_identifier 0x02, data: 000000d2
11:31:32:455 add task 4205724 type 37 to 0x84FD27FFFE0E7062 cluster 0xEF00 req.id 150
11:31:32:458 Websocket 172.33.0.1:51754 send message: {"config":{"heatsetpoint":2100,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"85","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:0e:70:62-01-0201"} (ret = 283)
11:31:33:061 Poll APS request to 0x000B57FFFEB7E8E1 cluster: 0x0006 dropped, values are fresh enough
11:31:33:431 don't create binding for attribute reporting of sensor Radiator keuken
11:31:34:231 Poll APS request to 0x000B57FFFE489B9A cluster: 0x0006 dropped, values are fresh enough
11:31:35:292 Erase task req-id: 150, type: 37 zcl seqno: 148 send time 3, profileId: 0x0104, clusterId: 0xEF00
11:31:35:384 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 009402020004000000d2
11:31:35:384 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 009402020004000000d2
11:31:35:384 Tuya debug 5 : Status: 0 Transid: 148 Dp: 514 (0x02,0x02) Fn: 0 Data 210
11:31:35:404 Poll APS request to 0x000B57FFFEBB7C5D cluster: 0x0006 dropped, values are fresh enough
11:31:35:504 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 009402020004000000d2
11:31:35:504 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 009402020004000000d2
11:31:35:504 Tuya debug 5 : Status: 0 Transid: 148 Dp: 514 (0x02,0x02) Fn: 0 Data 210
11:31:35:632 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 009402020004000000d2
11:31:35:633 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 009402020004000000d2
11:31:35:633 Tuya debug 5 : Status: 0 Transid: 148 Dp: 514 (0x02,0x02) Fn: 0 Data 210
11:31:35:767 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x02, Payload 009402020004000000d2
11:31:35:767 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 009402020004000000d2
11:31:35:767 Tuya debug 5 : Status: 0 Transid: 148 Dp: 514 (0x02,0x02) Fn: 0 Data 210
11:31:35:895 Tuya debug Request : Address 0x84FD27FFFE0E7062, Endpoint 0x01, Command 0x01, Payload 017c02020004000000d2
11:31:35:895 Tuya debug 4 : Address 0x84FD27FFFE0E7062 Payload 017c02020004000000d2
11:31:35:895 Tuya debug 5 : Status: 1 Transid: 124 Dp: 514 (0x02,0x02) Fn: 0 Data 210
```

Does this mean it's an issue on HA side?

Smanar, 2021-11-27:
I miss the line just before your logs ^^

>11:31:32:455 add task 4205724 type 37 to 0x84FD27FFFE0E7062 cluster 0xEF00 req.id 150

I need more information about this task

But NVM, according to log it seem you are sending only temperature request.

Can make a try direcly with the api
`curl -H 'Content-Type: application/json' -X PUT -d '{"mode": "heat"}' http://IP:PORT/api/APIKEY/sensors/84/config`

jghaanstra, 2021-11-27:
I used Postman and I can now see a `Send Tuya request` log. When looking at the instance config the mode is indeed updated to `heat`. No changes in HA however, the climate mode status remains `off` and I can not switch it to `heat`.

```
12:52:05:955 add task 4226783 type 19 to 0x000B57FFFEB7E8E1 cluster 0x0300 req.id 58
12:52:05:955 Poll APS request 58 to 0x000B57FFFEB7E8E1 cluster: 0x0300
12:52:06:067 Poll APS confirm 58 status: 0x00
12:52:06:067 Erase task req-id: 58, type: 19 zcl seqno: 99 send time 0, profileId: 0x0104, clusterId: 0x0300
12:52:06:127 Node data 0x000b57fffeb7e8e1 profileId: 0x0104, clusterId: 0x0300
12:52:06:128 0x000B57FFFEB7E8E1: update ZCL value 0x01/0x0300/0x0007 after 0 s
12:52:06:128 0x000B57FFFEB7E8E1: update ZCL value 0x01/0x0300/0x0008 after 0 s
12:52:06:128 0x000B57FFFEB7E8E1: update ZCL value 0x01/0x0300/0x4001 after 0 s
12:52:06:154 sensor 56 (lumi.sensor_motion.aq2): disable presence
12:52:06:157 Websocket 172.33.0.1:51754 send message: {"e":"changed","id":"56","r":"sensors","state":{"lastupdated":"2021-11-27T12:52:06.155","presence":false},"t":"event","uniqueid":"00:15:8d:00:04:4b:11:74-01-0406"} (ret = 163)
12:52:06:178 Send Tuya request 0x84FD27FFFE0E7062 : Dp_type: 0x04, Dp_identifier 0x6A, data: 01
12:52:06:178 add task 4226786 type 37 to 0x84FD27FFFE0E7062 cluster 0xEF00 req.id 62
12:52:06:181 Websocket 172.33.0.1:51754 send message: {"config":{"heatsetpoint":2200,"locked":false,"mode":"heat","offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"85","r":"sensors","t":"event","uniqueid":"84:fd:27:ff:fe:0e:70:62-01-0201"} (ret = 285)
12:52:06:690 poll node 00:0b:57:ff:fe:48:9b:9a-01
12:52:06:690 Poll light node Keukenlamp 3
12:52:06:742 Poll APS request to 0x000B57FFFE489B9A cluster: 0x0006 dropped, values are fresh enough
12:52:07:054 Poll APS request to 0x000B57FFFE489B9A cluster: 0x0008 dropped, values are fresh enough
12:52:07:154 read attributes of 0x000B57FFFE489B9A cluster: 0x0300: [ 
12:52:07:154 0x0007 
12:52:07:155 0x0008 
12:52:07:155 0x4001 
12:52:07:155 ]
12:52:07:155 add task 4226791 type 19 to 0x000B57FFFE489B9A cluster 0x0300 req.id 67
12:52:07:155 Poll APS request 67 to 0x000B57FFFE489B9A cluster: 0x0300
12:52:07:170 Poll APS confirm 67 status: 0x00
12:52:07:171 Erase task req-id: 67, type: 19 zcl seqno: 101 send time 0, profileId: 0x0104, clusterId: 0x0300
12:52:07:231 Node data 0x000b57fffe489b9a profileId: 0x0104, clusterId: 0x0300
12:52:07:232 0x000B57FFFE489B9A: update ZCL value 0x01/0x0300/0x0007 after 0 s
12:52:07:232 0x000B57FFFE489B9A: update ZCL value 0x01/0x0300/0x0008 after 0 s
12:52:07:232 0x000B57FFFE489B9A: update ZCL value 0x01/0x0300/0x4001 after 0 s
12:52:07:254 Set sensor check interval to 1000 milliseconds
12:52:07:799 poll node 00:0b:57:ff:fe:bb:7d:01-01
12:52:07:799 Poll light node Keukenlamp 2
12:52:07:851 Poll APS request to 0x000B57FFFEBB7D01 cluster: 0x0006 dropped, values are fresh enough
12:52:08:154 Poll APS request to 0x000B57FFFEBB7D01 cluster: 0x0008 dropped, values are fresh enough
12:52:08:186 don't create binding for attribute reporting of sensor Radiator keuken
12:52:08:254 read attributes of 0x000B57FFFEBB7D01 cluster: 0x0300: [ 
12:52:08:255 0x0007 
12:52:08:255 0x0008 
12:52:08:255 0x4001 
12:52:08:255 ]
12:52:08:255 add task 4226796 type 19 to 0x000B57FFFEBB7D01 cluster 0x0300 req.id 73
12:52:08:255 Poll APS request 73 to 0x000B57FFFEBB7D01 cluster: 0x0300
12:52:08:270 Poll APS confirm 73 status: 0x00
12:52:08:271 Erase task req-id: 73, type: 19 zcl seqno: 102 send time 0, profileId: 0x0104, clusterId: 0x0300
12:52:08:335 Node data 0x000b57fffebb7d01 profileId: 0x0104, clusterId: 0x0300
12:52:08:335 0x000B57FFFEBB7D01: update ZCL value 0x01/0x0300/0x0007 after 0 s
12:52:08:335 0x000B57FFFEBB7D01: update ZCL value 0x01/0x0300/0x0008 after 0 s
12:52:08:336 0x000B57FFFEBB7D01: update ZCL value 0x01/0x0300/0x4001 after 0 s
12:52:08:364 Master: read param with arg 0x19
12:52:08:388 Device TTL 4801 s flags: 0x7
12:52:08:854 poll node 00:0b:57:ff:fe:bb:7c:5d-01
12:52:08:855 Poll light node Keukenlamp 1
12:52:08:904 Poll APS request to 0x000B57FFFEBB7C5D cluster: 0x0006 dropped, values are fresh enough
12:52:09:220 Poll APS request to 0x000B57FFFEBB7C5D cluster: 0x0008 dropped, values are fresh enough
12:52:09:262 Erase task req-id: 62, type: 37 zcl seqno: 100 send time 3, profileId: 0x0104, clusterId: 0xEF00
12:52:09:326 read attributes of 0x000B57FFFEBB7C5D cluster: 0x0300: [ 
12:52:09:326 0x0007 
12:52:09:327 0x0008 
12:52:09:327 0x4001 
12:52:09:327 ]
```

Smanar, 2021-11-27:
Can you make a capture from HA where you use the "mode" command ?
I think the issue is perhaps from HA, but someone can have an answer here

jghaanstra, 2021-11-27:
Sure thing, here you go.

https://drive.google.com/file/d/1YTsM4RG8A8ozdTwHc3a-O91_U1vtTqOa/view?usp=sharing

Kane610, 2021-11-28:
Can you enable debug logging for the integration? I'd like to see the description read during startup of HA as well as what controls are sent and events which are received

jghaanstra, 2021-11-28:
Ok, I enabled logging and restarted HA. Turns out the mode for the valves that I manually updated through deCONZ API can now be set in HA as well after this reboot. The auto value was added to the mode option as well and changing the mode now sticks. I'm not seeying any visual changes to the valves itself and I have no idea if setting it to off actually closes the valve. I would need to test that later. But the issue is that deCONZ sets the mode to `null` upon pairing which is not handled correclty when trying top update the mode from HA for the first time.

For reference, this is the log for one of the valves for which I have not munually update the mode yet and where the issue is still present.

```
'86': {'config': {'heatsetpoint': 2200,
                               'locked': False,
                               'mode': None,
                               'offset': 0,
                               'on': True,
                               'preset': 'manual',
                               'reachable': True,
                               'schedule': {},
                               'schedule_on': False,
                               'setvalve': True,
                               'windowopen_set': False},
                    'ep': 1,
                    'etag': '30c44ba8c6027d3b2676368e65799bda',
                    'lastannounced': '2021-11-15T08:50:04Z',
                    'lastseen': '2021-11-28T12:31Z',
                    'manufacturername': '_TZE200_zivfvd7h',
                    'modelid': 'TS0601',
                    'name': 'Radiator woonkamer 2',
                    'state': {'lastupdated': '2021-11-28T16:04:17.183',
                              'lowbattery': False,
                              'on': False,
                              'temperature': 2200,
                              'valve': 0},
                    'type': 'ZHAThermostat',
                    'uniqueid': '84:71:27:ff:fe:00:2c:a1-01-0201'},
```

Trying to set the mode:
```
2021-11-28 17:30:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"70","r":"sensors","state":{"lastupdated":"2021-11-28T16:30:32.003","presence":true},"t":"event","uniqueid":"00:15:8d:00:05:82:f4:cd-01-0406"}
2021-11-28 17:30:33 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "http://172.33.0.56:2080/api/8641EBAEF3/sensors/86/config"
2021-11-28 17:30:33 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/sensors/86/config/on': True}}]
2021-11-28 17:30:33 DEBUG (MainThread) [pydeconz.websocket] {"config":{"heatsetpoint":2200,"locked":false,"mode":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":false,"setvalve":true,"windowopen_set":false},"e":"changed","id":"86","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:00:2c:a1-01-0201"}
2021-11-28 17:30:33 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"4","r":"sensors","state":{"lastupdated":"2021-11-28T16:30:33.754","presence":false},"t":"event","uniqueid":"00:15:8d:00:04:14:3f:4b-01-0406"}
```

Smanar, 2021-11-28:
Ha yes, Kane610 have explained me the issue
if you have config/mode = None (and why not Null ?), the HA plugin considere the device don't have "mode" parameter so it use insteaof config/on

So if you use it manualy, it will update the value in deconz API, and HA can re enable this parameter.

Notsosmarthome, 2021-12-07:
I just got some TRV's called "Nedis Zigbee radiator thermostat". Appears to be a version of these mentioned in this thread.

Sucessfully paired two of them to Deconz, but I don't seem to get any functionality from them. I'm using Home Assistant, Hassos on a Rpi3b+. They are not discovered in HA, (or might be, because I have some "unknown device" there).

The manufacturer is: _TZE200_hhrtiq0x
Model is TS0601

I am not an advanced user or posess any greater skill, but is there anything I can do, to get them to show up in HA?

Really grateful for any help here.

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-12-07:
>I am not an advanced user or posess any greater skill, but is there anything I can do, to get them to show up in HA?

It s a new device, so if you want to use on your side, you need to compile yourself the code with the needed modification.
New device integration on deconz is blocked for a time.

Notsosmarthome, 2021-12-07:
@Smanar Thank you for your reply, I appreciate it!

I don't know where to start.. Do you have any tips where I should try and compile the code? On what platform or what do I need? I try to make sense of the posts in this thread, but I have to admit that I don't really get half of it.

How do you mean that new device integration on deconz is blocked?

Smanar, 2021-12-08:
It a  clone of the device _TZE200_kfvq6avy, so you can add it with changing  2/3 lines like here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5497/files

To recompile the code with your changes https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

>How do you mean that new device integration on deconz is blocked?

You will have soon a big change in deconz, the DDF core, and all added devices will be removed to use them, so useless to add new device to remove them in somes months.

Notsosmarthome, 2021-12-08:
Thank you for this info, really appreciated! I will try to make it work after the upcoming DDF core change, but it's a bit overwhelming for a new and inexperinced user.

A bit Off-topic, but I have tried using the rest plugin before to try and fix somethings, but I don't seem to get it to work. Do you know if it is possible to use the rest plugin/api if running Hassos, or is it necessary to run for example Raspbian Buster image, from phoscon.de?

> You will have soon a big change in deconz, the DDF core, and all added devices will be removed to use them, so useless to add new device to remove them in somes months.

Thank you for this info, haven't heard/read anything about it. Am I understanding you correctly,that ALL added devices (lights/ sensors/switches/whatnots) will have to be removed, and then re-added/connected to Deconz after the DDF core change? In that case, won't that be a major pain for several thousand people?

Smanar, 2021-12-09:
> Am I understanding you correctly,that ALL added devices (lights/ sensors/switches/whatnots) will have to be removed, and then re-added/connected to Deconz after the DDF core change? In that case, won't that be a major pain for several thousand people?

All will be invisible (I hope ^^), device will be removed from code and in same time added by DDF system.

>Do you know if it is possible to use the rest plugin/api if running Hassos, or is it necessary to run for example Raspbian Buster image, from phoscon.de?

Ok, so It's my personnal opinion, but HASS use docker system, and no dockers are not easier to use, it s possible yes to recompile the lib to be used on HA, but few user succed, using a clean (and FULL) Raspian installation is easier, you can use the pre-installed image, or a Raspian official one (avoid bullseyes for the moment)

Skare69, 2021-12-12:
I got the same TRV (_TZE200_zivfvd7h / TS0601) and thanks to @jghaanstra its mode can now be set via HA (after setting `{"mode":"auto"}` once via deCONZ REST API. 
Checking the values via REST API all looks good but the device itself seems to still ignore it. The temperature can be set. Changing `windowopen_set` did also work but changing the `mode` seems to simply do nothing. The valve is still open (judging by the heat of the radiator). I can only get it to close by turning the knob on the TRV conuter-clockwise until it shows "OFf". But this does not update the status in deconz at all.

Smanar, 2021-12-12:
Can you compare the request show in deconz log (with "info", debug lines will start by "tuya debug")
- when making the request using HA
- when turning off the device manualy

From the code, you need to have/use 0x02 0xXX (XX is the mode can be 00 01 or 02)

Skare69, 2021-12-12:
Setting the temperature (while it is "off" in HA but not actually off):
```
21:20:17:374 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x02, Dp_identifier 0x02, data: 000000c8
21:20:19:631 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 000c02020004000000c8
21:20:19:632 Tuya debug 5 : Status: 0 Transid: 12 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
21:20:19:770 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 003002020004000000c8
21:20:19:770 Tuya debug 5 : Status: 0 Transid: 48 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
21:20:19:966 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 003002020004000000c8
21:20:19:966 Tuya debug 5 : Status: 0 Transid: 48 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
21:20:20:095 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 003102020004000000c8
21:20:20:095 Tuya debug 5 : Status: 0 Transid: 49 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
```
Clicking on "heat" in HA:
```
21:20:28:732 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x6A, data: 01
```
Clicking on "off" in HA:
```
21:20:39:326 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x6A, data: 02
```

Turning the device off manually:
```
21:22:05:386 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 001c02020004000000c8
21:22:05:387 Tuya debug 5 : Status: 0 Transid: 28 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
21:22:05:457 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 003202020004000000c8
21:22:05:458 Tuya debug 5 : Status: 0 Transid: 50 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743560
21:22:12:511 Device TTL 1677 s flags: 0x7
21:22:14:273 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00330202000400000032
21:22:14:273 Tuya debug 5 : Status: 0 Transid: 51 Dp: 514 (0x02,0x02) Fn: 0 Data 281470681743410
```

Smanar, 2021-12-13:
F*ck
Ok so your device need to use the command 0x02, the one you see when you use it manualy, but in the code it use 0x6A.

Do you have a linux machine to test code modification ?
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_fix_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Skare69, 2021-12-13:
I've set up deCONZ with your branch on an Ubuntu 20.04 virtualbox but the behaviour did not change:

sending: `curl -X PUT http://localhost:8080/api/909DDC09DC/sensors/2/config -d '{"mode":"off"}'`
log:
```
22:42:55:109 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 008002020004000000c8
22:42:55:109 Tuya debug 5 : Status: 0 Transid: 128 Dp: 514 (0x02,0x02) Fn: 0 Data 200
```

sending: `curl -X PUT http://localhost:8080/api/909DDC09DC/sensors/2/config -d '{"mode":"heat"}'`
log:
```
22:44:46:758 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00b602020004000000c8
22:44:46:758 Tuya debug 5 : Status: 0 Transid: 182 Dp: 514 (0x02,0x02) Fn: 0 Data 200
22:44:46:893 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 008102020004000000c8
22:44:46:893 Tuya debug 5 : Status: 0 Transid: 129 Dp: 514 (0x02,0x02) Fn: 0 Data 200
```

Unfortunately the device still does nothing (except for lighting up its screen once).

Not sure how I will be able to properly test this with my home assistant setup though.

Smanar, 2021-12-14:
Ok So I don't think the 0x02 0x02 is the mode command, because the data value, you have 200 or 50, and mode use 0x01 0x02 or 0x03

I have take a llok on z2m code, there are using the 0x02 0x04.

Can you make a try with the new code
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You can send the request, but deconz can't handle return yet (I prefer be sure the request is good before)
On log you will have "Send Tuya request ...." when making the request and I hope good return just after.

Skare69, 2021-12-15:
Looks better now!

Sending: `curl -X PUT http://localhost:8080/api/B62B11D007/sensors/2/config -d '{"mode":"off"}'`
Log:
```
09:34:27:612 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 02
09:34:27:942 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00ba0404000102
09:34:27:942 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00ba0404000102
09:34:27:942 Tuya debug 5 : Status: 0 Transid: 186 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
09:34:28:079 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00ba0404000102
09:34:28:080 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00ba0404000102
09:34:28:080 Tuya debug 5 : Status: 0 Transid: 186 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
09:34:28:193 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 001a0404000102
09:34:28:193 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 001a0404000102
09:34:28:193 Tuya debug 5 : Status: 0 Transid: 26 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
09:34:28:404 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 001b02020004000000e6
09:34:28:404 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 001b02020004000000e6
09:34:28:404 Tuya debug 5 : Status: 0 Transid: 27 Dp: 514 (0x02,0x02) Fn: 0 Data 230
```
Device switches to "manual" mode.

Sending `curl -X PUT http://localhost:8080/api/B62B11D007/sensors/2/config -d '{"mode":"auto"}'`
Log: 
```
09:35:49:457 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 00
```
Device does nothing.

Sending `curl -X PUT http://localhost:8080/api/B62B11D007/sensors/2/config -d '{"mode":"heat"}'`
Log: 
```
09:37:46:595 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 01
09:37:47:450 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00f50404000101
09:37:47:450 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00f50404000101
09:37:47:450 Tuya debug 5 : Status: 0 Transid: 245 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
09:37:47:566 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 001e0404000101
09:37:47:566 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 001e0404000101
09:37:47:566 Tuya debug 5 : Status: 0 Transid: 30 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
09:37:47:764 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 001f02020004000000d2
09:37:47:764 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 001f02020004000000d2
09:37:47:764 Tuya debug 5 : Status: 0 Transid: 31 Dp: 514 (0x02,0x02) Fn: 0 Data 210
```
Device switches to "auto" mode.

Smanar, 2021-12-15:
Ok so I have rewritten all the code, this device use a totally new command for mode, and the 3 mode was  not in the same order.
You can try the new online code (not tested on my side, so you can have error during complilation)

Skare69, 2021-12-17:
Sorry for the delay. I just got a second Conbee 2 to easier debug this stuff. 

Sending `curl -X PUT http://localhost:8080/api/EF7E91BCD7/sensors/2/config -d '{"preset":"manual"}'`

Log: 
```
21:15:24:641 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 02
21:15:28:425 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00110404000102
21:15:28:425 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00110404000102
21:15:28:425 Tuya debug 5 : Status: 0 Transid: 17 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
21:15:28:531 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 00790404000102
21:15:28:531 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00790404000102
21:15:28:531 Tuya debug 5 : Status: 0 Transid: 121 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
21:15:28:745 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 007a02020004000000e1
21:15:28:746 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 007a02020004000000e1
21:15:28:746 Tuya debug 5 : Status: 0 Transid: 122 Dp: 514 (0x02,0x02) Fn: 0 Data 225
```
The device does actually switch to "manual" mode!

Sending `curl -X PUT http://localhost:8080/api/EF7E91BCD7/sensors/2/config -d '{"preset":"auto"}'`
Log:
```
21:16:41:995 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 01
21:16:46:559 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00130404000101
21:16:46:559 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00130404000101
21:16:46:559 Tuya debug 5 : Status: 0 Transid: 19 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
21:16:46:686 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00130404000101
21:16:46:686 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00130404000101
21:16:46:686 Tuya debug 5 : Status: 0 Transid: 19 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
21:16:46:803 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 007e0404000101
21:16:46:803 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 007e0404000101
21:16:46:803 Tuya debug 5 : Status: 0 Transid: 126 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
21:16:47:017 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 007f0202000400000037
21:16:47:017 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 007f0202000400000037
21:16:47:017 Tuya debug 5 : Status: 0 Transid: 127 Dp: 514 (0x02,0x02) Fn: 0 Data 55
```
And the device switches to "auto" mode!

Sending `curl -X PUT http://localhost:8080/api/EF7E91BCD7/sensors/2/config -d '{"preset":"holiday"}'`
Log:
```
21:17:15:700 Send Tuya request 0x84FD27FFFE6DCCB6 : Dp_type: 0x04, Dp_identifier 0x04, data: 00
21:17:17:659 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x02, Payload 00140404000100
21:17:17:659 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00140404000100
21:17:17:659 Tuya debug 5 : Status: 0 Transid: 20 Dp: 1028 (0x04,0x04) Fn: 0 Data 0
21:17:17:764 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 00800404000100
21:17:17:764 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00800404000100
21:17:17:764 Tuya debug 5 : Status: 0 Transid: 128 Dp: 1028 (0x04,0x04) Fn: 0 Data 0
21:17:17:975 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 00810202000400000032
21:17:17:975 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00810202000400000032
21:17:17:975 Tuya debug 5 : Status: 0 Transid: 129 Dp: 514 (0x02,0x02) Fn: 0 Data 50
```
The device switches to "holiday" mode!

Thank you, great work so far!

I just wonder how it could be turned off then. Doing so manually yields:
```
21:18:16:365 Tuya debug Request : Address 0x84FD27FFFE6DCCB6, Endpoint 0x01, Command 0x01, Payload 00820202000400000032
21:18:16:365 Tuya debug 4 : Address 0x84FD27FFFE6DCCB6 Payload 00820202000400000032
21:18:16:365 Tuya debug 5 : Status: 0 Transid: 130 Dp: 514 (0x02,0x02) Fn: 0 Data 50
```

Smanar, 2021-12-18:
So it s was that, we need to use "preset" instead of "mode" ?
We can avoid it during device creation.

>I just wonder how it could be turned off then. Doing so manually yields:

It's a a setheatpoint request with a temperature = 5°C, it's used on lot of tuya thermostat, the "off" is trigger when setting a temperature < 5 °C and the "boost" with a value > 40°C

Notsosmarthome, 2022-06-14:
Glad to hear that you got it to work @Skare69 Still functioning properly?

Any chance that this device now is supported in “regular” deconz without using the Rest API?

Smanar, 2022-06-14:
You mean the GUI ?
Can try with the tuya cluster, If I m right this cluster is usable on the GUI now.

Notsosmarthome, 2022-06-15:
> You mean the GUI ? Can try with the tuya cluster, If I m right this cluster is usable on the GUI now.

Pardon my ignorance dear sir and pardon my lack of info.

What I meant was if this TRV now can be used (I couldnt before, it just connects to deconz but don't show up in HA) and controlled in the HA GUI, after being connected to Deconz through zigbee. And this without having to use the Rest API (out of my skill level).

Would you mind explaining what you meant with "Can try the Tuya cluster.."?

I should also say that I have not updated Deconz to 6.14.1, I am still at 6.11.1 (from dec -21), afraid of braking something if I update. After A LOT of issues, my Conbee 2/Deconz setup has worked flawlessely for several months now. If these particular thermostats would be supported in 6.14.1, I would however try and update.

Thank you

Smanar, 2022-06-15:
>Would you mind explaining what you meant with "Can try the Tuya cluster.."?

Nevermind, was with the deconz GUI nothing to see with what you are asking.

Model _TYST11_zivfvd7h and _TZE200_zivfvd7h are in the code since the deconz version 2.12

You can check the version you have in phoscon (I don't know the HA correspondance)