dresden-elektronik/deconz-rest-plugin #4771: Aqara S1 Smart Scene Panel

aryelevin, 2021-04-15:
## Describe the question or issue you are having

Hello, this issue is just to present a new Aqara device and how it can be integrated on node-red/node.js (or any other server) using deCONZ.
As there is no deCONZ forum, using the github.
This device is a wired switch but have lot of more features.

## Screenshots
[image]

Here's a video to show and explain the device features and capabilities: https://youtu.be/WmuCyrW3AYI

It an be used natively for exemple for:
- 3 gangs switch
- Thermostat/AC control
- Curtains/Blind Rollers/Shutters
- Smart Light Control (Color/CT/Brightness control)
- Scenes
- Weather information as well as time and date

## Environment

## deCONZ Logs

NA

## Additional context

About the integration, this device use a special protocol described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Aqara-S1-Smart-Scene-Panel-Protocol

At the moment, the integration is not possible without the original gateway, probably because of Zigbee stack R22.
I used it with original Aqara M2 Hub to include it on the network, after that, I have deCONZ network with the same MAC/Key/PANID/etc so after it included in the Aqara Hub, I power it down and open deCONZ the I see it on the network.

As proof of concept to be used with node red the device ATM create a light entry in the API which looks like this: 

```
{
  "etag": "86905eda8d6edc3fe6eb954616ddd7e2",
  "hascolor": false,
  "lastannounced": "2021-04-11T10:01:19Z",
  "lastseen": "2021-04-11T10:01Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.switch.n4acn4",
  "name": "On/Off light 1",
  "state": {
    "alert": "none",
    "aqara_s1_auto_update_fw_enabled": false,
    "aqara_s1_font_size": 3,
    "aqara_s1_gestures": 255,
    "aqara_s1_homepage": 3,
    "aqara_s1_language": 1,
    "aqara_s1_lcd_auto_brightness_enabled": false,
    "aqara_s1_lcd_brightness": 50,
    "aqara_s1_panel_communication": "f2ff4115aa710f841b3706410c6169725f636f6e640e020055",
    "aqara_s1_screen_saver_enabled": true,
    "aqara_s1_screen_saver_style": 1,
    "aqara_s1_sound_volume": 0,
    "aqara_s1_standby_enabled": false,
    "aqara_s1_standby_lcd_brightness": 10,
    "aqara_s1_standby_time": 15,
    "aqara_s1_switch1_icon": 0,
    "aqara_s1_switch1_text": "Guests Room",
    "aqara_s1_switch2_icon": 0,
    "aqara_s1_switch2_text": "Guests Room",
    "aqara_s1_switch3_icon": 0,
    "aqara_s1_switch3_text": "Guests Room",
    "aqara_s1_switches_config": 7,
    "aqara_s1_theme": 1,
    "on": false,
    "reachable": true
  },
  "swversion": "08-31-2020",
  "type": "On/Off light",
  "uniqueid": "04:cf:8c:df:3c:88:ce:d8-01"
}
```

But it will change soon.

I'm looking if there is users that want to help to finish the integration ...
Actually, I wish to do it much better so the deep handling will be in the API side together with phoscon, not in the node.js side.

Smanar, 2021-04-15:
About the WIP, the actual entry is just for tests, he know we can't use this kind of JSON, but not sure the better approach to take.

This device is realy a switch with 3 outpputs.
[image]

So the "On/Off light" is ok, but perhaps a new sensor for the rest (it use a Xiaomi manufacture cluster)

aryelevin, 2021-04-15:
Correct, It is a 3 gang switch in addition to all the advanced functions...

Smanar, 2021-04-19:
No one want to make some tries with this device ? ^^

closed 2021-05-18 by github-actions[bot]