dresden-elektronik/deconz-rest-plugin #4774: Climax Remote Keypad

BjornRonnestad, 2021-04-15:
## Device

- Product name: Climax Remote Keypad
- Manufacturer: Climax Technology
- Model identifier: KP-ACE_00.00.03.12TC
- Device type :
  - Other: Remote keypad / Alarm Panel

## Screenshots

### Basic
[image]

### Identify
[image]

### IAS Zone:
[image]
[image]

Node info:
[image]

Manual:
[4514805-Manual-18.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6320337/4514805-Manual-18.pdf)

Would love support for this device :)

Best regards,

Bjørn

Smanar, 2021-04-19:
Ha nice, we are testing a new sensor for keypad https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4758

Do you have a linux machine to compile and test code modification ?

a1ad, 2021-04-23:
there is even NFC tag support! I would be awesome if you could read the NFC tag and use it for different things :-)

[image]

Smanar, 2021-04-23:
I think the rfid reader will procoke same notification than when yiu use a pin code, but yes will be interesting to test this device

froogl, 2021-05-16:
Bump. Please don't stale this issue.

Smanar, 2021-05-16:
There is no one with this device to make tries ?

BjornRonnestad, 2021-05-19:
Hi, 

sorry for long response time on this.

I'm afraid I don't have a linux-machine for compile and test.
Am planning on running this on HASSIO on Raspberry Pi. 

Is there anyway I can test anything - or otherwise help in this matter?

Best regards,

Bjørn

Smanar, 2021-05-19:
>Am planning on running this on HASSIO on Raspberry Pi.
>Is there anyway I can test anything - or otherwise help in this matter?

Bad luck ^^ HASSIO is not a full OS too.

BjornRonnestad, 2021-05-20:
> > Am planning on running this on HASSIO on Raspberry Pi.
> > Is there anyway I can test anything - or otherwise help in this matter?
> 
> Bad luck ^^ HASSIO is not a full OS too.

Well, can't have both sugar and cream, I see.. :)

Smanar, 2022-04-21:
Nope there is so much change since the last time.
DDF can support keypad now, you have a sample here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5352

There is lot of thing to read, but the final DDF is here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5850

Easier if you just want to edit it to support your device.

discodogge, 2022-04-22:
okay. Fantastic news for me :) I will start reading when work is over. probably ;P  hehehe tanks!

Smanar, 2022-04-22:
If it work for one keypad, can work for another (I don't think we need to edit c++ code)
Pls show your DDF if it don't work.

closed 2022-05-22 by github-actions[bot]

tejo68, 2023-01-19:
Hi,
have anyone this device  working in deconz/ha?
I've edited the ddf with correct endpoints and the keypad connects to deconz and updates the alarmsystem in rest-api correctly (arming/disarming). However the actual sensor state object (state/panel) never updates and is stuck at "exit_delay".
In home assistant I can disarm the panel, but panel state is stuck at "pending". Any ideas?
Best Regards,
Terje

Smanar, 2023-01-19:
Yep, it s not something simple. there is lot of step and they need to be done in the good order.

- Have you set the code 0
- can you check in the alarm endpoint if "config/configured" is true
- can you check in the alarm endpoint if the device is present in the "config/devices".

tejo68, 2023-01-20:
Thanks for helping out.
code 0 is set
configured is true
device is present
{
    "1": {
        "config": {
            "armed_away_entry_delay": 1,
            "armed_away_exit_delay": 1,
            "armed_away_trigger_duration": 1,
            "armed_night_entry_delay": 1,
            "armed_night_exit_delay": 1,
            "armed_night_trigger_duration": 1,
            "armed_stay_entry_delay": 1,
            "armed_stay_exit_delay": 1,
            "armed_stay_trigger_duration": 1,
            "armmode": "armed_away",
            "configured": true,
            "disarmed_entry_delay": 0,
            "disarmed_exit_delay": 0
        },
        "devices": {
            "00:12:4b:00:19:a4:a2:7d-01-0501": {
                "armmask": "none"
            }
        },
        "name": "default",
        "state": {
            "armstate": "armed_away",
            "seconds_remaining": 0
        }
    }
}
In the device state object panel state never changes:
 "2": {
        "config": {
            "battery": 0,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "c9c7671670d3858edc4bb6c1b38f72aa",
        "lastannounced": null,
        "lastseen": "2023-01-20T17:32Z",
        "manufacturername": "ClimaxTechnology",
        "modelid": "KP-ACE_00.00.03.12TC",
        "name": "AncillaryControl 2",
        "state": {
            "lastupdated": "2023-01-20T17:31:02.796",
            "lowbattery": false,
            "panel": "exit_delay",
            "tampered": true
        },
        "type": "ZHAAncillaryControl",
        "uniqueid": "00:12:4b:00:19:a4:a2:7d-01-0501"
It's always "exit_delay"
I have a paralell setup with the frient keypad and that is working in the Homeassistant integration, and the panel updates, not with this one. In deconz I can se this device does not publish any services on endpoint 0x0501 IAS ACE 
I can disarm from HA and the rest-api updates, but HA does not get state updates from deconz.
This is the DDF I'm using:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ClimaxTechnology",
  "modelid": "KP-ACE_00.00.03.12TC",
  "product": "KP-ACE_00.00.03.12TC",
  "sleeper": true,
  "status": "Bronze",
  "path": "/devices/kp-ace_00.00.03.12tc.json",
  "subdevices": [
    {
      "type": "ZHAAncillaryControl",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0501"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0003",
          "0x0500"
        ],
        "out": [
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0�100 %.",
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/action",
          "public": false,
          "default": "disarmed"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/panel",
          "default": "exit_delay"
        },
        {
          "name": "state/tampered"
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration",
          "description": "The duration until presence is automatically turned back to false."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "description": "True when presence is detected.",
          "default": false
        }
      ]
    }
  ]
}

Smanar, 2023-01-20:
```
"armed_away_entry_delay": 1,
"armed_away_exit_delay": 1,
"armed_away_trigger_duration": 1,
"armed_night_entry_delay": 1,
"armed_night_exit_delay": 1,
"armed_night_trigger_duration": 1,
"armed_stay_entry_delay": 1,
```

It's normal all setting are set a "1" ?

It's a parrallel setup with the  frient keypad ? So not the same alarmsystem ? Do you have the  frient keypad model id ? for I check the code for a difference.

>In deconz I can se this device does not publish any services on endpoint 0x0501 IAS ACE

Do you have an exemple.

Your DDF have "bronze" status, so not used on classic configuration.
The DDF miss too state/seconds_remaining, cf this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6189 (this issue was added to solve panel return)

If your frient keypad have a DDF, try to mimic it https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/frient/kepzb-110_keypad.json

tejo68, 2023-01-20:
Hi,
I’ve changed the default values of 120 sec to speed up things while testing.
No I run the frient on a separate system but config is the same. Deconz in a virtualbox running ubuntu 20.04.5. The climax manual:
[image]
I used the frient ddf and changed the endpoints, from 0x02 to 0x01 etc
I»ll add the missing parts to the ddf snd test.
Thanks!

Smanar, 2023-01-21:
If I m right the DDF look like
```

{
   "schema":"devcap1.schema.json",
   "manufacturername": "ClimaxTechnology",
   "product":"Climax Remote Keypad",
   "vendor": "Develco Products",
   "modelid": "KP-ACE_00.00.03.12TC",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"ZHAAncillaryControl",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0501"
         ],
         "fingerprint":{
            "profile":"0x0104",
            "device":"0x0401",
            "endpoint":"0x01",
            "in":[
               "0x0000",
               "0x0001",
               "0x0500"
            ],
            "out":[
               "0x0501"
            ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/action",
               "public":true
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
              "name": "state/seconds_remaining"
            },
            {
               "name":"state/panel",
               "default":"exit_delay"
            },
            {
               "name":"state/tampered"
            }
         ]
      },
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0406"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/presence"
            }
         ]
      }
   ]
}
```

But now I m seing the presence sensor, IDK how this one is working ....