dresden-elektronik/deconz-rest-plugin #4776: RasPi 4B + RaspBee II: Stops working when LAN cable attached

fubar-coder, 2021-04-15:
## Describe the question or issue you are having

I have a Raspberry PI 4B with a RaspBee II and I observed the strange behavior that communication with ZigBee devices doesn't work as soon as I plug in a LAN cable into the Raspberry PI. deconz can find (some) other devices, but is otherwise unable to communicate with them. I'm unable to query cluster information and query/set single values.

## Screenshots

None available. I could provide a VNC image where one can only see direct links to the devices, usually with a LQI of >200.

## Environment

- Host system: Raspberry Pi 4B
- Running method: Marthoc Docker container
- Firmware version: 26660700
- deCONZ version: 2.10.04
- Device: RaspBee II
- Is there any other USB or serial devices connected to the host system? If so: Which?
  - USB storage device

## deCONZ Logs

The logs are just showing discoveries, request starts, timeouts. I can provide a log if required.

## Additional context

No additional context.

senilio, 2021-04-15:
1. What specs are the power supply you use?
2. Do you see anything about voltage when you run a `sudo dmesg`?
3. Do you have any 0xE1 errors in your deCONZ log?

```To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.```

fubar-coder, 2021-04-15:
1. 5V 3A
2. No, I don't see any dmesg/syslog messages about a faulty/unstable power supply
3. Yes, I do indeed see E1 statuses/errors

```text
ESC[36mdeconz_1          |ESC[0m 20:58:57:795 APS-DATA.confirm id: 7, status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:58:58:011 Node 0x00158D000192D28A is known by 0 neighbors, last seen -1 s
ESC[36mdeconz_1          |ESC[0m 20:58:58:024 Poll APS confirm 9 status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:58:58:025    drop item attr/modelid
ESC[36mdeconz_1          |ESC[0m 20:58:58:025    drop item attr/swversion
ESC[36mdeconz_1          |ESC[0m 20:58:58:026 0x04CF8CDF3C764018 error APSDE-DATA.confirm: 0xE1 on task
ESC[36mdeconz_1          |ESC[0m 20:58:58:026 Erase task req-id: 9, type: 19 zcl seqno: 159 send time 0, profileId: 0x0104, clusterId: 0x0006
ESC[36mdeconz_1          |ESC[0m 20:58:58:027 APS-DATA.confirm id: 9, status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:58:58:296 enqueue event config/localtime for /config/
ESC[36mdeconz_1          |ESC[0m 20:58:58:492 Node 0x04CF8CDF3C803771 is known by 10 neighbors, last seen 308 s
ESC[36mdeconz_1          |ESC[0m 20:58:58:649 poll node ec:1b:bd:ff:fe:ab:93:5f-03
ESC[36mdeconz_1          |ESC[0m 20:58:58:649 Poll light node Steckdose 3-3
ESC[36mdeconz_1          |ESC[0m 20:58:58:697 Poll APS request to 0xEC1BBDFFFEAB935F cluster: 0x0006 dropped, values are fresh enough
ESC[36mdeconz_1          |ESC[0m 20:58:58:972 Node 0x00124B0007196493 is known by 9 neighbors, last seen 301 s
ESC[36mdeconz_1          |ESC[0m 20:58:59:296 enqueue event config/localtime for /config/
ESC[36mdeconz_1          |ESC[0m 20:58:59:451 Node 0x000D6F0014F715C1 is known by 8 neighbors, last seen 2 s
ESC[36mdeconz_1          |ESC[0m 20:58:59:621 poll node 00:15:8d:00:03:89:32:6e-01-0b04
ESC[36mdeconz_1          |ESC[0m 20:58:59:622 Poll ZHAPower sensor node Power 19
ESC[36mdeconz_1          |ESC[0m 20:58:59:883 Poll APS request to 0x00158D000389326E cluster: 0x0B04 dropped, values are fresh enough
ESC[36mdeconz_1          |ESC[0m 20:58:59:932 Node 0xEC1BBDFFFE326652 is known by 8 neighbors, last seen 35 s
ESC[36mdeconz_1          |ESC[0m 20:59:00:295 enqueue event config/localtime for /config/
ESC[36mdeconz_1          |ESC[0m 20:59:00:412 Node 0x842E14FFFEF5BDCD is known by 0 neighbors, last seen -1 s
ESC[36mdeconz_1          |ESC[0m 20:59:00:620 poll node 00:15:8d:00:03:89:24:1d-01-0b04
ESC[36mdeconz_1          |ESC[0m 20:59:00:620 Poll ZHAPower sensor node Power 21
ESC[36mdeconz_1          |ESC[0m 20:59:00:878 read attributes of 0x00158D000389241D cluster: 0x0B04: [ 20:59:00:879 0x0505 20:59:00:879 0x0508 20:59:00:880 0x050B 20:59:00:880 ]
ESC[36mdeconz_1          |ESC[0m 20:59:00:880 add task 153251 type 19 to 0x00158D000389241D cluster 0x0B04 req.id 25
ESC[36mdeconz_1          |ESC[0m 20:59:00:881 Poll APS request 25 to 0x00158D000389241D cluster: 0x0B04
ESC[36mdeconz_1          |ESC[0m 20:59:00:891 Node 0xEC1BBDFFFEA024AD is known by 1 neighbors, last seen 371 s
ESC[36mdeconz_1          |ESC[0m 20:59:00:895 APS-DATA.request id: 25, addrmode: 0x03, addr: 0x00158d000389241d, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 1 len: 9 tx.options 0x00
ESC[36mdeconz_1          |ESC[0m 20:59:01:160 Poll APS confirm 25 status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:59:01:161 0x00158D000389241D error APSDE-DATA.confirm: 0xE1 on task
ESC[36mdeconz_1          |ESC[0m 20:59:01:161 Erase task req-id: 25, type: 19 zcl seqno: 160 send time 0, profileId: 0x0104, clusterId: 0x0B04
ESC[36mdeconz_1          |ESC[0m 20:59:01:162 APS-DATA.confirm id: 25, status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:59:01:295 enqueue event config/localtime for /config/
ESC[36mdeconz_1          |ESC[0m 20:59:01:372 Node 0x588E81FFFED4AF0D is known by 9 neighbors, last seen 277 s
ESC[36mdeconz_1          |ESC[0m 20:59:01:793 poll node 00:15:8d:00:03:42:11:a0-01
ESC[36mdeconz_1          |ESC[0m 20:59:01:794 Poll light node Klimagerät
ESC[36mdeconz_1          |ESC[0m 20:59:01:845 read attributes of 0x00158D00034211A0 cluster: 0x0006: [ 20:59:01:846 0x0000 20:59:01:847 ]
ESC[36mdeconz_1          |ESC[0m 20:59:01:847 add task 153256 type 19 to 0x00158D00034211A0 cluster 0x0006 req.id 30
ESC[36mdeconz_1          |ESC[0m 20:59:01:848 Poll APS request 30 to 0x00158D00034211A0 cluster: 0x0006
ESC[36mdeconz_1          |ESC[0m 20:59:01:851 Node 0xEC1BBDFFFEB61BEF is known by 8 neighbors, last seen 49 s
ESC[36mdeconz_1          |ESC[0m 20:59:01:895 APS-DATA.request id: 30, addrmode: 0x03, addr: 0x00158d00034211a0, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
ESC[36mdeconz_1          |ESC[0m 20:59:01:948 APS-DATA.indication srcAddr: 0xf173, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0000, lqi: 207, rssi: -68
ESC[36mdeconz_1          |ESC[0m 20:59:01:949 APS-DATA.request id: 32, addrmode: 0x02, addr: 0xf173, profile: 0x0000, cluster: 0x8000, ep: 0x00 -> 0x00 queue: 2 len: 12 tx.options 0x04
ESC[36mdeconz_1          |ESC[0m 20:59:01:950 enqueue event attr/lastseen for /lights/8
ESC[36mdeconz_1          |ESC[0m 20:59:01:953 Websocket 172.18.0.1:52116 send message: {"attr":{"id":"8","lastannounced":"2021-03-21T09:53:33Z","lastseen":"2021-04-15T18:59Z","manufacturername":"Heiman","modelid":"SmartPlug","name":"Freier Zwischenstecker","swversion":"2018.1.11","type":"Smart plug","uniqueid":"00:0d:6f:00:14:f7:15:c1-01"},"e":"changed","id":"8","r":"lights","t":"event","uniqueid":"00:0d:6f:00:14:f7:15:c1-01"} (ret = 343)
[... cut ...]
ESC[36mdeconz_1          |ESC[0m 20:59:02:990 Poll APS confirm 38 status: 0xE1
ESC[36mdeconz_1          |ESC[0m 20:59:02:991    drop item attr/modelid
ESC[36mdeconz_1          |ESC[0m 20:59:02:992    drop item attr/swversion
ESC[36mdeconz_1          |ESC[0m 20:59:02:992 0x00158D000389326E error APSDE-DATA.confirm: 0xE1 on task
ESC[36mdeconz_1          |ESC[0m 20:59:02:993 Erase task req-id: 38, type: 19 zcl seqno: 162 send time 0, profileId: 0x0104, clusterId: 0x0006
ESC[36mdeconz_1          |ESC[0m 20:59:02:993 APS-DATA.confirm id: 38, status: 0xE1
```

senilio, 2021-04-15:
Does the 0xE1 errors stop when you remove the LAN cable?

You can read about this error in a bit more detail here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

fubar-coder, 2021-04-15:
Yes, they (almost) immediately stop. The whole network just starts working again.

senilio, 2021-04-15:
You write about having a USB storage device connected... SSD? USB3? Since you're having the Raspbee tethered to the RPi, you can try moving the storage device further away by connecting it with an extension cable.

fubar-coder, 2021-04-15:
I'll try, but this may take some time. However, everything works fine with the USB 3 stick attached, but I'll try your suggestion

ebaauw, 2021-04-15:
I've never heard/read/experienced interference due to LAN (wired ethernet); interference due to Wifi is conceivable (it shares the 2.4GHz band with Zigbee and bluetooth); maybe the Wifi goes crazy when you connect the network cable?  What happens if you disable the Wifi (after connecting the network cable)?

USB3 is a known source of interference.  I suppose, in theory, the USB voltage could be slightly different with LAN connected, changing the interference characteristics.  Indeed using an extension cable (or moving the storage device to a USB-2 port) should work around that.  I've also read reports about wrapping the USB-3 socket and cable in aluminium foil, but haven't tried that myself.

fubar-coder, 2021-04-15:
I tried deactivating WLAN, but the problem remained the same.

fubar-coder, 2021-04-17:
Update: I bought a USB hub (active) and here is the outcome of my tests:

LAN cable | USB Hub | USB stick placement | Outcome
------------|-----------|------------------------|-------------
Detached | Detached | USB 3.0 port (top) | OK
Attached | Detached | USB 3.0 port (top) | Error
Detached | Attached | Attached to hub | Error
Detached | Detached | USB 3.0 port (bottom) | Error

Any idea what I can try next?

SwoopX, 2021-04-17:
Hm, I guess using an active Hub probably eliminates the option of having too less power (we had reports that RPi4 + Conbee II @ 2.5A wasn't enough, with 3A it was ok).

More out of curiosity, what's the content of you `config.txt` for the RPi? Do you have BT active/need it?

fubar-coder, 2021-04-17:
This are the active entries:

```bash
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1
enable_uart=1
dtparam=sd_poll_once
dtoverlay=miniuart-bt
dtoverlay=pwm,pin=13,func=4
```

fubar-coder, 2021-04-17:
I disabled BT.

The USB hub started to interfer with the RaspBee II again, so using USB 2.0 isn't a solution either.

ebaauw, 2021-04-17:
You want to stick the ConBee in a USB-2 port, preferably using an extension cable to move it away from the USB-3 ports.  If it’s a bus-powered USB hub you might be asking too much of the power supply.  Check with `vcgencmd` if the Pi has experienced undervoltage, see https://github.com/ebaauw/homebridge-rpi/wiki/Hardware-Monitoring#system-status.

SwoopX, 2021-04-17:
@ebaauw RaspBee II this time !

Mimiix, 2021-04-20:
Just a taught here: What if the UTP cable just adds a obstacle in the area and thus pointing the usb noise into the RASPBEE's direction? 

Wondering if the same would happen if there's a UTP In there but not connected on the other side into the router/switch.

fubar-coder, 2021-04-30:
@Mimiix  I'll test this.

closed 2021-05-29 by github-actions[bot]