dresden-elektronik/deconz-rest-plugin #4780: LIDL Xmas lightstrip no longer works after deCONZ restart.

ebaauw, 2021-04-17:
## Describe the bug
On pairing, the LIDL Xmas lightstrip works fine.  The REST API Plugin creates the following resource:
```
{
  "etag": "bcdbd976e4a3cc572e042b7fc967a8c5",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2021-04-17T08:48Z",
  "manufacturername": "LIDL Livarno Lux",
  "modelid": "HG06467",
  "name": "Color dimmable light 13",
  "state": {
    "bri": 0,
    "colormode": "hs",
    "effect": "none",
    "hue": 0,
    "on": true,
    "reachable": true,
    "sat": 0
  },
  "swversion": null,
  "type": "Color dimmable light",
  "uniqueid": "58:8e:81:ff:fe:d8:83:cc-01"
}
```
However when deCONZ is restarted, the resource looks like:
```
{
  "etag": "98a3e6b225aecf4bad7f266c725ca29a",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2021-04-17T08:48Z",
  "manufacturername": "LIDL Livarno Lux",
  "modelid": "HG06467",
  "name": "Color dimmable light 13",
  "state": {
    "bri": 0,
    "colormode": "hs",
    "on": true,
    "reachable": true
  },
  "swversion": null,
  "type": "Color dimmable light",
  "uniqueid": "58:8e:81:ff:fe:d8:83:cc-01"
}
```
Obviously, the strip cannot be controlled for colour or effects with the missing attributes.

## Steps to reproduce the behavior
Pair the Xmas light strip and restart deCONZ.

## Expected behavior
After restart, the strip should be exposed as it was after pairing.

## Screenshots
n/a

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 0x26680700
 - deCONZ version: 2.11.0-beta
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
n/a

## Additional context
Was already mentioned in #4539.

LeoeLeoeL, 2021-04-17:
The title should be modified to include all Livarno smart lights because also HG06106C (E27), HG06106B (E14) and HG06106A (GU10) suffer this problem.

ebaauw, 2021-04-17:
My E27 doesn't, but please list the resource with the missing attributes.

closed 2021-04-18 by manup