dresden-elektronik/deconz-rest-plugin #4781: Trigger rule x seconds after device announces

thomas-wolf, 2021-04-17:
## Describe the question or issue you are having

My son keeps on coming to our bedroom at strange times asking if the night's already over. He has a nightlight at his bed with color capabilities. So what I'd like to achieve: 

If my son switches on his light between say 21:00 and 06:30, it turns red at reduced brightness. After x seconds, the light turns back to its usual color. 

I am able to achieve the first requirement by using a rule that sets xy color and brightness accordingly between the above mentioned times and the light is turned on (. However, I got stuck at the second part. I am aware that there is a condition operator "ddx", but I am not able to find any documentation about it. 

The light is turned of by its integrated switch, so it's powerless after turning it of. This is why I chose to base the logic on the lastannounced attribute.

What I tried - first part working, second part not triggering. Times changed to be able to test during daytime :)

```
{
        "actions": [
            {
                "address": "/lights/12/state",
                "body": {
                    "xy": [0.7,0.3]
                    ,"bri":120
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/lights/12/attr/lastannounced",
                "operator": "dx"
            },
            {
                "address": "/config/localtime",
                "operator": "in",
                "value": "T07:30:00/T22:00:00"
            }
        ]
         , "name": "First Part"
}
{
        "actions": [
            {
                "address": "/lights/12/state",
                "body": {
                    "xy": [0.35,0.35],
                    "bri": 80
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/lights/12/attr/lastannounced",
                "operator": "ddx",
                "value" : "PT00:00:15"
                
            },
            {
                "address": "/config/localtime",
                "operator": "in",
                "value": "T07:30:00/T22:00:00"
            }
        ]
     , "name": "Part 2 - Back to normal color"
}
```

Bonus questions:

- Is there an option to let the light blink?
- Can I differentiate times between weekdays and weekend? Using the W3 / W124 prefix as known from schedules does not seem to work here
- Could the second part with the ddx operator also use the last trigger time of the first rule (as it shall always trigger x seconds after the first rule has triggered)? In my opinion, this is a neater way to build the desired logic

## Screenshots

## Environment

 - Host system: Raspi 3
 - Running method: Raspbian / Conbee 2
 - Firmware version: 26660700
 - deCONZ version: (2.10.03)
 - Device: ConBee2
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No 

## deCONZ Logs

## Additional context

ebaauw, 2021-04-17:
> What I tried - first part working, second part not triggering

Are you sure it's not triggering (did you check `lasttriggered` and `timestriggered`)?

Syntactically, the second rule looks OK to me, but I don't think I ever tried a `ddx` condition on `lastannounced`.  I suppose, the light would send multiple _Device Annouce_ messages on boot, so the first rule would fire multiple times, possibly undoing the effects of the second rule.  deCONZ is probably trying to be smart and removing duplicate (or similar) commands from the queue, effectively canceling the action for the second rule.

I would probably use a CLIP sensor instead.  Set it on the first rule (together with the light state) and put a ddx condition on the CLIP sensor for the second rule.  Alternatively, you could try a `ddx` condition on the light brightness and/or colour.

> Is there an option to let the light blink?

Set `alert`, though lights by different vendors might react differently.

> Can I differentiate times between weekdays and weekend? Using the W3 / W124 prefix as known from schedules does not seem to work here

You'd need to provide a value like `W124/T07:30:00/T22:00:00`.  Not sure, but I think the weekday works on the first time, do W3/T23:00:00/T08:00:00` would be from 23:00 on Sunday evening to 08:00 Monday morning (which probably is not what you want).

> Could the second part with the ddx operator also use the last trigger time of the first rule (as it shall always trigger x seconds after the first rule has triggered)? In my opinion, this is a neater way to build the desired logic

No, we would need to create `ResourceItem`s for the rule's `lasttriggered` and `timestriggered` attributes.  I've been fantasising about that, so I can get web socket notifications for these as well, for monitoring/debugging the execution of rules, but the people already complaining about too many notifications for `lastseen` will probably go completely out of their minds.

thomas-wolf, 2021-04-17:
@ebaauw: Thanks a lot for the great feedback, I really appreciate it!

> Are you sure it's not triggering (did you check lasttriggered and timestriggered)?
Indeed, lasttriggered is set to none and timestriggered is 0

> I would probably use a CLIP sensor instead. Set it on the first rule (together with the light state) and put a ddx condition on the CLIP sensor for the second rule
Thanks for pointing that out. I hadn't read through the sensors section carefully yet, because I assumed these are only physical sensors. But I guess virtual sensors do help me even with some other ideas I have in my mind.

> Not sure, but I think the weekday works on the first time, do W3/T23:00:00/T08:00:00` would be from 23:00 on Sunday evening to 08:00 Monday morning (which probably is not what you want)
Thanks again for pointing that out. So I guess I'll give it a try with W6/T23:00:00/T08:00:00 then. This should start on Friday night and last to Sunday morning. At first try, the API did not accept these kind of values at all, but most likely I got the syntax wrong.

For now, I came up with the following:

Changing the light color to red - both actions working
```
 "15": {
        "actions": [
            {
                "address": "/lights/12/state",
                "body": {
                    "bri": 120,
                    "xy": [
                        0.7,
                        0.3
                    ]
                },
                "method": "PUT"
            },
            {
                "address": "/sensors/7/state",
                "body": {
                    "flag": true
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/lights/12/attr/lastannounced",
                "operator": "dx"
            },
            {
                "address": "/config/localtime",
                "operator": "in",
                "value": "W6/T07:30:00/T22:00:00"
            }
        ],
        "created": "2021-04-17T11:54:14",
        "etag": "1c6bb3569b3ab21720c45748feb3229b",
        "lasttriggered": "2021-04-17T12:07:02",
        "name": "Part 1",
        "owner": "4A92890DFC",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 5
    }
```

Changing the color back to normal 10 sec later - not being triggered
```
 "19": {
        "actions": [
            {
                "address": "/lights/12/state",
                "body": {
                    "bri": 80,
                    "xy": [
                        0.35,
                        0.35
                    ]
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/config/localtime",
                "operator": "in",
                "value": "W6/T07:30:00/T22:00:00"
            },
            {
                "address": "/sensors/7/state/lastupdated",
                "operator": "ddx",
                "value": "PT00:00:10"
            }
        ],
        "created": "2021-04-17T12:06:55",
        "etag": "40686c99243fa190ac4239d3bff7ce7c",
        "lasttriggered": "none",
        "name": "Part 2",
        "owner": "4A92890DFC",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 0
    }
```

And the virtual CLIP sensor - I notice the lastupdate attribute gets updated each time the part 1 rule triggers
```
   "7":{
      "config":{
         "on":true,
         "reachable":true
      },
      "etag":"1c6bb3569b3ab21720c45748feb3229b",
      "manufacturername":"Me",
      "modelid":"T1000",
      "name":"MySensor",
      "state":{
         "flag":true,
         "lastupdated":"2021-04-17T12:07:02.475"
      },
      "swversion":"1.0",
      "type":"CLIPGenericFlag",
      "uniqueid":"00:1f:ee:00:00:00:08:bb-01-1000"
   }
```

I started out with the second rule having also an action to set back the flag of the sensor to false, and with a condition that checks for the flag to be true - however I am not 100% sure if this works using the eq operator as the bool value I set for the condition got converted to string. Then I realized I do not need to toggle the bool value in order to get the lastupdated attribute updated.

Any ideas why the second rule does not trigger?

thomas-wolf, 2021-04-17:
okay ,it is working when I leave out the localtime condition in the second rule. I don't need it. It is sufficient to have this condition when setting the state of the CLIP sensor.

Thanks a lot for your help, @ebaauw !

closed 2021-04-17 by thomas-wolf

ebaauw, 2021-04-17:
> it is working when I leave out the localtime condition in the second rule.

I was just about to suggest checking that ;-)

Dunno why the second rule doesn't trigger with the added `localtime` condition.  The `ddx` condition should be treated as trigger, and the `in` condition as additional condition, just as with the first rule, where the `dx` condition is the trigger.  My guess would be that the API somehow evaluates the `in` as trigger instead of condition, so it would only be true at 7:30.  I don't think I have a rule that combines `in` and `ddx` conditions.

thomas-wolf, 2021-04-17:
Only one issue left: I guess the kid just won't stay in bed when the light's red. Any suggestions on that?

Nevermind, it was just a little fun project anyway. The last time he came around to ask if the night is already over at 11pm, I told him I'd turn the light red to let him now without disturbing us. Just kidding, but he insisted.