dresden-elektronik/deconz-rest-plugin #4786: Inside Alexa App wrong Device is added to Group

jannnfe, 2021-04-19:
## Describe the question or issue you are having
I learned a light source in Phoscon and then searched for new devices using Alexa. The lamp appears in the Alexa app. So far so good. A Hue bridge is also taught in the Alexa app, which also provides lights. If I now want to add the light that was taught in the Phoscon app to a group in the Alexa app, then another lamp is simply added from the Alexa app. The selected lamp is not added to the group, but simply another one that exists in the Alexa app. The problem only exists with lamps from the Phoscon app. How can I solve the problem so that I can add the lamp to the correct group in the Alexa app?

Is there a way to deactivate the search for lamps from Phoscon for Alexa?

Thanks! :)

## Screenshots
None

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian)
 - Firmware version: (26390500)
 - deCONZ version: (2.10.04 / 14.3.2021)
 - Device: (ConBee I)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: NO

## deCONZ Logs
None

## Additional context
None

jannnfe, 2021-05-03:
I am the only one with this issue?

Mimiix, 2021-05-04:
I am not sure if this is something that's rerelated to deCONZ or phoscon. I think you should contact Amazon on it.