dresden-elektronik/deconz-rest-plugin #4788: Content-Length header is missing a space

froogl, 2021-04-20:
## Describe the bug

The REST API returns an invalid HTTP response, which some parsers cannot parse. The `Content-Length` header violates the HTTP spec, as it's missing a space between the colon character and the actual content length.

## Steps to reproduce the behavior
1. Open a shell
2. Issue the curl request: ```curl -i --raw http://deconz.local/api```
3. Check response:
```
HTTP/1.1 403 Forbidden
Access-Control-Allow-Origin: *
Content-Type: application/json; charset=utf-8
Content-Length:70

[{"error":{"address":"/","description":"unauthorized user","type":1}}]d
```

As you can see **Content-Length:70** is an invalid header.

## Expected behavior

The header should be as follows:

 **Content-Length: 70**

## Screenshots

## Environment

 - Host system: Cubietruck
 - Running method: Debian 10
 - Firmware version: (26xxyy00)
 - **deCONZ version: (2.09.03)**
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-04-21:
But this not depend of deconz, but the client that do the request.
Deconz make a request of 70 Lenght, and this is displayed differently by the client.

Try with your browser, with console dev, you will see the espace.
[image]

But you have provlem with some parsers ?

ebaauw, 2021-04-21:
I can confirm the bug.  Above commit fixes all references to `Content-Length` I could find.

ebaauw, 2021-04-21:
```
$ curl -i --raw http://localhost/api
HTTP/1.1 403 Forbidden
Access-Control-Allow-Origin: *
Content-Type: application/json; charset=utf-8
Content-Length: 70

[{"error":{"address":"/","description":"unauthorized user","type":1}}]
```

manup, 2021-04-22:
Good catch

closed 2021-04-22 by manup