dresden-elektronik/deconz-rest-plugin #4791: MEGOS Smart Home remote

Jonas-wav, 2021-04-21:
## Device

- Product name: MEGOS Smart Home remote
  - https://www.megos-light.com/eng/led-assortment/ "LED STARTER SET"
- Manufacturer: _TZ3000_ztrfrcsu
- Model identifier: TS1001
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch
  - (Light)
[image]

## Screenshots

[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-04-21:
Missing screenshots.

Jonas-wav, 2021-04-22:
Sorry, so far I didn't know where to find the other information, and I also thought that this would be sufficient, as shown in the [example](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support). :) 
That's all I've found so far, hope it's what you need.

Mimiix, 2021-04-22:
Hi,

Screenshots of Identity and Groups clusters are still missing.

Kind regards,

Jonas-wav, 2021-04-22:
I'm sorry, but I think you need to clarify [there](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support) which screenshots are really needed ("You can typically spare indentify...").

Mimiix, 2021-04-22:
Further clusters. Meaning  all clusters.

Jonas-wav, 2021-04-30:
Is it still incomplete?

Mimiix, 2021-04-30:
Seems fine to me :)

closed 2021-05-29 by github-actions[bot]