dresden-elektronik/deconz-rest-plugin #4792: LIDL smart plugs and smart extensions

andys73, 2021-04-22:
## Describe the question or issue you are having
LIDL smart plugs and the smart extensions (UK / IRL version) are reported as unavailable from time to time. I have to unplug the smart plug / smart extensions and then the item is available again in Deconz and home assistant. Reported the issue to LIDL and ended up talking the the support from the manufacturer that basically is not giving me any support as I am not using the LIDL HUB

## Screenshots
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method:  Home Assistant deCONZ Add-on 
 - Firmware version: 26660700
 - deCONZ version: 2.10.04 
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: SSD in enclosure - connected with a small cable. The Conbee 2 stick is on an USB extension cable as needs to stay away for interference...
 -
## deCONZ Logs
I do not have any logs

## Additional context
about 100 devices in the system, 4 LIDL smart plugs, 3 LIDL smart extensions, 1 Heimann smartplug, 1 OSRAM smart plug, 20+ ZBMINI, 1 x Sonoff basic zigbee, 6 INNR Smartlights, rest Aquara sensors battery operated - only the LIDL go offline

Mimiix, 2021-04-22:
Hi,

Are you able to add logs? Otherwise we can't say anything.

andys73, 2021-04-22:
Hello,
I am running this in Home Assistant OS with a raspbery pi, and it is happening randomly, could be once every few weeks. How can I produce logs without interfering with the "live" operations?

Thanks

Mimiix, 2021-04-23:
You should be able to enable logs in the addon. Not sure how to enable those. Check their documentation for the best guidance on that.

In terms of deconz logs: Please provide some debug logs from running a few minutes. That can already help in some cases.
To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

The HA addon has a bug in the noVNC (browser) view of deCONZ: You can't copy paste logs. I recommend using realvnc. Again, check the HA Addon documentation on how to enable it.

Please paste the logs in Pastebin and post that url here :)

andys73, 2021-04-23:
it actually happen again and I can see in logs!
 APS-DATA.request rejected, destination 0x60a423fffe0949d0 is zombie node
[deconz-logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6365058/deconz-logs.txt)

Mimiix, 2021-04-23:
What powerSupply are you using?

It's odd, it could be that this is not working properly though.

andys73, 2021-04-23:
I am using the standard raspbery pi power supply and I use the HA plug in that tells me it is ok., It is saying that it was ok whole day

I have umplugged one of the impacted plugs and plug it back and cames back online.

Rebooted the PI host and whatever was not connected before is not connected now. It only comes back after it is unplugged and plugged back.  Can this Zombie status be reset somehow so it retries more?

[image]

andys73, 2021-05-06:
Today a Heimann smartplug went offline just after before me updating the home assistant (noticed in the morning only). Had to power off the smart plug and put it back in, that was the only way to get it back online. Everything else was online.  So is not only LIDL ones that are going offline like that...

[image]

andys73, 2021-05-10:
anyone else having similar problems, within 1 h I had a Lidl smart extension, 1 smart plug and 1 heiman gas sensor going "unavailable" while everything else is fine. I had plug them off and on again and they reconnect fine. Could the power devices that are not set to ON for a while being ejected from the network somehow?

andys73, 2021-06-07:
ok, it was a message on the discord that executing the deCONZ: Device refresh service from home assistant it will repair the offline devices and it did (it is not instant, you need to wait few minutes).
I am glad is a workaround and hope is fixed in the new release.

closed 2021-06-07 by andys73