dresden-elektronik/deconz-rest-plugin #4798: Schedule returns explicit `null` for time when localtime is set

justin-caldicott, 2021-04-23:
## Describe the question or issue you are having

Schedule with localtime set returns time as null in the response.

I'm fine with this in principle, no harm in being explicit. But since I switched my schedules to use localtime, the Hue Essentials app refuses to connect to deconz, due to these null values. I've emailed Hue Essentials support, asking if they can be more flexible here, but not heard anything back so far.

I don't have a genuine Hue Bridge to test on, but I suspect it might not return a time property at all, when localtime is set. I understand that, where possible, you're being api-compatible with the Hue Bridge?

As a sidenote, I noticed that localtime doesn't seem to be documented here: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/schedules/

## Screenshots
N/A

## Environment
 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.10.04 / 14/03/2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
N/A

## Additional context
Let me know if you need any more details.

Mimiix, 2021-04-23:
@Thomas-Vos Might want to reply on this from Hue Essentials.

Thomas-Vos, 2021-04-23:
I received an email about this and am working on a fix. Hue Essentials does not want to connect if time is null which will be fixed in the next update.

By the way, on a Hue bridge, time is never null. It will just convert localtime to utc and use that for time. So both time and localtime are provided for schedules. That is the reason Hue Essentials does not want to connect, as it does not expect null. But I can fix this in the app for deCONZ.

Mimiix, 2021-04-23:
@Thomas-Vos I have no clue about any of this, but is this something we need to fix too?

Thomas-Vos, 2021-04-23:
Maybe, as a Hue bridge does not return null for `time`. I still need to look into this further to see in what situations `null` is returned for time.

justin-caldicott, 2021-04-26:
Many thanks @Mimiix & @Thomas-Vos. Let me know if I can help any further.

Sounds good if deconz responses line up with the Hue bridge. Though auto-conversion to time does sound unexpected. It's nice to be able to GET what you PUT.

Thomas-Vos, 2021-04-26:
I just updated Hue Essentials with a fix so it can now pair if `time` is null. Version 1.21.1 on Android and 1.15.1 on iOS. The update is currently going through Google/Apple review and should be available soon.

I still need to check if the schedules within Hue Essentials still work as expected, and if needed I will fix those in another update. But at least you can pair the app again now.

justin-caldicott, 2021-04-26:
Many thanks @Thomas-Vos. I guess this issue now just documents the difference between the Hue bridge and Deconz. I'll leave you to decide if that needs addressing. :)

closed 2021-05-25 by github-actions[bot]