dresden-elektronik/deconz-rest-plugin #4806: Blinds Tuya AM43 deviceDevice name

Wimaware, 2021-04-24:
I noticed the new beta release V2.11.1 beta.

There is an addition of the Moes AM43.
Initial support for Moes ZigBee Smart DIY Motorized Roller AM43-0.45/40-ES-EZ(TY) #4674
Is there a way to add the _TZE200_zah67ekd to it, it's exactly the same device.
See #4310

Thanks

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-04-24:
It seem this device is already in the code

>    {"_TZE200_zah67ekd", "TS0601", "MoesHouse / Livolo", "Tuya_COVD AM43-0.45-40"},

Wich one deconz version are you using ? Remember you need 2 inclusions without deleting anything.

Smanar, 2021-04-24:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4807

Are you able to test it ? (need a linux machine with full OS)

Mimiix, 2021-04-24:
@Smanar is there a chance this device is different then the others?

Wimaware, 2021-04-25:
I am using the last beta V2.11.1 under windows so I can't test, sorry.
What do you mean with: Remember you need 2 inclusions without deleting anything
It's battery operated with adapter to charge the batteries

Smanar, 2021-04-25:
@mimix, impossible to known, tuya magic, but I have already added 2/3 models of this device and only 1 with the battery tips, because an user have opened an issue just for that.

It help a lot to reconize the device, deconz create a battery device, and the "light" part can use this data to reconize the "light" entry

@Wimaware this device can work without this PR, but it s harder, try that
- set phoscon in permit join (add new light)
- read the basic attribute of the device 2 times in deconz, cluster 0x0000, button "read"

Wimaware, 2021-04-25:
Again tried to join and read the basic attribute twice but with no luck
Tried with adapter connected and without, makes no difference

[image]
[image]

Mimiix, 2021-04-25:
> @mimix, impossible to known, tuya magic, but I have already added 2/3 models of this device and only 1 with the battery tips, because an user have opened an issue just for that.
>
> It help a lot to [...]

In that case I'd like the user to add all screenshots.

@Smanar please wait with any integration until all screens are provided and you can confirm the functionality 
@Wimaware Can you provide all screenshots as per template?

Wimaware, 2021-04-25:
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-04-25:
Nothing more, the PR is already validated https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4807

Can try with the next version.

Wimaware, 2021-04-25:
Oké, thanks I will try with the next release.

Wimaware, 2021-05-01:
Hi guys, just installed the new beta release but the _TZE200_zah67ekd is still not recognized.
The cluster readings are the same as mentioned above.
[image]

Smanar, 2021-05-01:
Yep, that you see in deconz is the reality, the code don't change the device characteristic.

Try 2 inclusions, without deleting anything, the first one create a sensor entry, a ZHABattery, and the second one will create the "light" ressource.

Wimaware, 2021-05-01:
Yesss, that's it. Works great !! Just to know how to do it.
Thank you for the effort and quick reply.

Smanar, 2021-05-01:
Np ^^
But close the issue pls, I forget to link the PR to this issue, so much problem recently with all tuya covering on battery, It will clean a little the issue list.

closed 2021-05-01 by Wimaware

chris199512, 2021-05-23:
Hey @Smanar , i own the AM43 with battery and i cant add it completely in my system. I use Phoscon 2.11.05 and Fhem. I add the shutter as a sensor two times. FHEM displayed one as battery and the other one as daylight. But it didnt add a "light" . In fhem it displayed in line manufacture: _TZE200_rddyvrci
Is something missing in the code for my blinds driver?
[image]
[image]

Smanar, 2021-05-24:
I have perhaps found an issue, you don't have the GUI to check for manufacture number ?

Have you tried a second inclusion without deleting the battery entry ?

chris199512, 2021-05-24:
I inserted the stick in my pc and read it out. Hope this is what you need. Yes i tried a second inclusion without deleting.
[image]
[image]
[image]

Smanar, 2021-05-24:
All seem fine ...

Can you share logs during the inclusion ? with "info" and "info_l2" ?

chris199512, 2021-05-26:
i deleted the node and added it new. now i have the light.  over fhem now i can open and close over bri.
the debug shows a warning: [WARNING] - Missing cluster in sensor fingerprint: 0x5C0272FFFEC9C9DC - 0xA447 (TS0601), endpoint: 0x01, cluster: 0x000A, payload: 0700, zclSeq: 117

`0:42:45:718 [ND] All good, nothing to do...
10:42:45:718 FP indication 0x0000 / 0x8002 (0x5C0272FFFEC9C9DC / 0x3884)
10:42:45:718                       ...     (0x5C0272FFFEC9C9DC / 0x3884)
10:42:45:719 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) cluster 0x8002
10:42:45:720 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) clear timeout on cluster 0x8002
10:42:45:730 don't use deleted sensor and node 0x5C0272FFFEC9C9DC as candidate
10:42:45:731 [2] get active endpoints for 0x5c0272fffec9c9dc
10:42:45:906 CTRL skip polling while permit join is set
10:42:45:930 APS-DATA.indication from child 0x3884
10:42:45:930 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x3884
10:42:45:931 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0x3884)
10:42:45:931                       ...     (0x5C0272FFFEC9C9DC / 0x3884)
10:42:45:932 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) cluster 0x8005
10:42:45:932 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) clear timeout on cluster 0x8005
10:42:45:938 don't use deleted sensor and node 0x5C0272FFFEC9C9DC as candidate
10:42:45:939 [3] get simple descriptor 0x01 for 0x5c0272fffec9c9dc
10:42:46:140 poll node 00:15:8d:00:04:6e:1a:43-01
10:42:46:141 Poll light node Extended color light 2
10:42:46:161 CTRL skip polling while permit join is set
10:42:46:416 CTRL skip polling while permit join is set
10:42:46:487 enqueue event config/localtime for /config/
10:42:46:665 CTRL skip polling while permit join is set
10:42:46:689 APS-DATA.indication from child 0x3884
10:42:46:690 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x3884
10:42:46:691 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0x3884)
10:42:46:691                       ...     (0x5C0272FFFEC9C9DC / 0x3884)
10:42:46:692 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) cluster 0x8005
10:42:46:697 don't use deleted sensor and node 0x5C0272FFFEC9C9DC as candidate
10:42:46:697 wait response fastEnddeviceProbe() 0x5C0272FFFEC9C9DC, elapsed 758 ms
10:42:46:907 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
10:42:46:908 COM: \\.\COM4 : Standardm��gige Seriell-�ber-Bluetooth-Verbindung (0x0000/0x0000)
10:42:46:908 COM: \\.\COM7 : Serielles USB-Ger�t (0x1CF1/0x0030)
10:42:46:909 COM: \\.\COM3 : Standardm��gige Seriell-�ber-Bluetooth-Verbindung (0x0000/0x0000)
10:42:46:919 CTRL skip polling while permit join is set
10:42:47:042 poll node 7c:b0:3e:aa:0a:08:e4:f1-03
10:42:47:043 Poll light node On/Off plug-in unit 3
10:42:47:171 CTRL skip polling while permit join is set
10:42:47:183 APS-DATA.indication from child 0x3884
10:42:47:184 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 5, node: 0x3884
10:42:47:184 Tuya debug 7 : Missing manufacture name for 0x5c0272fffec9c9dc
10:42:47:185 sql exec SELECT manufacturername FROM nodes WHERE mac LIKE '5c:02:72:ff:fe:c9:c9:dc%' COLLATE NOCASE
10:42:47:186 sql exec SELECT * FROM nodes WHERE mac='5c:02:72:ff:fe:c9:c9:dc-01' COLLATE NOCASE AND state != 'deleted'
10:42:47:187 sql exec SELECT * FROM nodes WHERE mac='0x5c0272fffec9c9dc' COLLATE NOCASE AND state != 'deleted'
10:42:47:188 sql exec SELECT * FROM nodes
10:42:47:189 LightNode 9: Smart plug 9 added
10:42:47:190 enqueue event event/added for /lights/9
10:42:47:190 enqueue event attr/lastseen for /lights/9
10:42:47:191 DB pushZdpDescriptorDb()
10:42:47:192 DB UPDATE device_descriptors SET data = x'0104015100010400000400050000ef0219000a00', timestamp = 1622018567 WHERE device_id = (SELECT id FROM devices WHERE mac = '5c:02:72:ff:fe:c9:c9:dc') AND endpoint = 1 AND type = 4
10:42:47:193 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104015100010400000400050000ef0219000a00', 1622018567 FROM devices WHERE mac = '5c:02:72:ff:fe:c9:c9:dc'
10:42:47:204 FP indication 0x0000 / 0x8004 (0x5C0272FFFEC9C9DC / 0x3884)
10:42:47:205                       ...     (0x5C0272FFFEC9C9DC / 0x3884)
10:42:47:206 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) cluster 0x8004
10:42:47:206 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0x3884) clear timeout on cluster 0x8004
10:42:47:207 read attributes of 0x5C0272FFFEC9C9DC cluster: 0x0000: [ 
10:42:47:207 0x0005 
10:42:47:208 ]
10:42:47:208 add task 660 type 19 to 0x5C0272FFFEC9C9DC cluster 0x0000 req.id 160
10:42:47:211 Websocket 192.168.178.36:53028 send message: {"e":"added","id":"9","light":{"etag":"b4b0572fc6dbd1751f72b830b91236b7","hascolor":false,"lastannounced":null,"lastseen":"2021-05-26T08:42Z","manufacturername":"_TZE200_rddyvrci","modelid":null,"name":"Smart plug 9","state":{"alert":"none","bri":0,"lift":0,"on":false,"open":false,"reachable":true},"swversion":null,"type":"Window covering device","uniqueid":"5c:02:72:ff:fe:c9:c9:dc-01"},"r":"lights","t":"event","uniqueid":"5c:02:72:ff:fe:c9:c9:dc-01"} (ret = 455)
10:42:47:223 don't use deleted sensor and node 0x5C0272FFFEC9C9DC as candidate
10:42:47:227 [4.1] Get manufacturer code
10:42:47:227 [4.2] get basic cluster attr 0x0004 for 0x5c0272fffec9c9dc
[... cut ...]
10:43:15:485 Wait 20s till query finished
10:43:15:672 Erase task req-id: 113, type: 41 zcl seqno: 198 send time 1, profileId: 0x0104, clusterId: 0xEF00
10:43:15:752 poll node 84:2e:14:ff:fe:35:a2:2f-01
10:43:15:753 Poll light node Smart plug 8
10:43:15:816 read attributes of 0x842E14FFFE35A22F cluster: 0x0006: [ 
10:43:15:817 0x0000 
10:43:15:817 ]
10:43:15:818 add task 781 type 19 to 0x842E14FFFE35A22F cluster 0x0006 req.id 115
10:43:15:818 Poll APS request 115 to 0x842E14FFFE35A22F cluster: 0x0006`

Smanar, 2021-05-26:
Don't worry 

>10:42:50:364 [WARNING] - Missing cluster in sensor fingerprint: 0x5C0272FFFEC9C9DC - 0x3884 (TS0601), endpoint: 0x01, cluster: 0x000A, payload: 0700, zclSeq: 115

It s just because the device is making request on Time cluster (0x000A), but I don't use it, haven't found an utility to that yet.

And BTW, what was the magic trick for the inclusion working finaly ?

chris199512, 2021-05-26:
okay.
i had to delete the node in the gui. there was a fault maybe, then i paired it two times. now it works.

Smanar, 2021-05-31:
@chris199512 I have same strange issue with the same device https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4942

Can you share some debug output to compare with yours, on his one he have relay long data

> 13:04:44:659 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 000001**04000102**
> 13:04:44:663 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846594

If I m right on your side it will be something like

> 13:04:44:659 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 000001**0100**

Smanar, 2021-06-01:
BTW, there is a special procedure for calibration on this device ?

chris199512, 2021-06-01:
> BTW, there is a special procedure for calibration on this device ?

The calibration is easy.

[image][image]

Smanar, 2021-06-01:
Ha nice, the complete procedure, thx ^^

Smanar, 2021-06-25:
Hello, no one else have side issue with the device ? Or no issue ?

tnowak, 2021-06-25:
Hi everybody.
Please take a look at issue I've created: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5049
I believe there is a bug in AM43 blinds support.

AM43 blinds motor has two physical buttons: Up and Down. Up button (located higher and with the arrow heading up) is meant to open the blind, obviously. You should also configure motor rotation to fit your mounting place / rotation requirements, so that this button opens the blind.

After you press button Up and wait for your blind to fully open, deCONZ API reports:
- open = false (should be: open = true)
- lift / level = 100% (should be:  0% - according to API docs blind level is "% of closed" so if blinds is open it is 0% closed)

So I find those datapoints to report the opposite values, either not true to real world or not compliant with general API docs.
Could anyone confirm my observations?
I believe that's required to consider it a bug.
Thanks in advance.

chris199512, 2022-03-03:
hey,
i restored factory settings because i would like to add it to a new stick. Now i have the same issue as before. only battery is added but the light not. In log is a message like: "don't create binding for attribute reporting of sensor Battery 7"
i added the log:
`Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
09:22:57:858 unknown node 0x5C0272FFFEC9C9DC (0xDBEF), lqi: 255
09:22:57:859 APS-DATA.indication from unknown node 0x5C0272FFFEC9C9DC
09:22:57:864 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0xDBEF
09:22:57:865 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0xDBEF
09:22:57:866 new node - ext: 0x5c0272fffec9c9dc, nwk: 0xDBEF
09:22:57:868 device announce 0x5C0272FFFEC9C9DC (0xDBEF) mac capabilities 0x80
09:22:57:869 set fast probe address to 0x5C0272FFFEC9C9DC (0xDBEF)
09:22:57:870 FP indication 0x0000 / 0x0013 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:57:870                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:57:871 device announce 0x5C0272FFFEC9C9DC (0xDBEF) mac capabilities 0x80
09:22:57:921 ZDP get node descriptor for 0xDBEF
09:22:57:923 ZDP get node descriptor for 0xDBEF
09:22:58:112 APS-DATA.indication from child 0xDBEF
09:22:58:113 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0xDBEF
09:22:58:114 DB pushZdpDescriptorDb()
09:22:58:114 DB save zll database items 0x00000801
09:22:58:116 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('2', 'normal', '00:21:2e:ff:ff:05:b8:f2-01', 'Configuration tool 2', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"2","attr/lastannounced":null,"attr/lastseen":"2022-03-03T08:22Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 2","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:05:b8:f2-01","state/reachable":true}')
09:22:58:118 DB sql exec UPDATE devices SET nwk = 56303 WHERE mac = '5c:02:72:ff:fe:c9:c9:dc';INSERT INTO devices (mac,nwk,timestamp) SELECT '5c:02:72:ff:fe:c9:c9:dc', 56303, strftime('%s','now') WHERE (SELECT changes() = 0);
09:22:58:236 DB saved in 122 ms
09:22:58:238 DB UPDATE device_descriptors SET data = x'0240800210525200002c520000', timestamp = 1646295778 WHERE device_id = (SELECT id FROM devices WHERE mac = '5c:02:72:ff:fe:c9:c9:dc') AND endpoint = 0 AND type = 2
09:22:58:240 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240800210525200002c520000', 1646295778 FROM devices WHERE mac = '5c:02:72:ff:fe:c9:c9:dc'
09:22:58:256 FP indication 0x0000 / 0x8002 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:257                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:257 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) cluster 0x8002
09:22:58:312 ZDP get active endpoints for 0xDBEF
09:22:58:315 [2] get active endpoints for 0x5c0272fffec9c9dc
09:22:58:316 ZDP get active endpoints for 0xDBEF
09:22:58:365 APS-DATA.indication from child 0xDBEF
09:22:58:366 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0xDBEF
09:22:58:367 DB pushZdpDescriptorDb()
09:22:58:369 FP indication 0x0000 / 0x8002 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:369                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:370 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) cluster 0x8002
09:22:58:402 wait response fastEnddeviceProbe() 0x5C0272FFFEC9C9DC, elapsed 85 ms
09:22:58:617 APS-DATA.indication from child 0xDBEF
09:22:58:618 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xDBEF
09:22:58:619 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:620                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:621 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) cluster 0x8005
09:22:58:622 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) clear timeout on cluster 0x8005
09:22:58:653 ZDP get simple descriptor 0x01 for 0xDBEF
09:22:58:657 [3] get simple descriptor 0x01 for 0x5c0272fffec9c9dc
09:22:58:658 ZDP get simple descriptor 0x01 for 0xDBEF
09:22:58:871 APS-DATA.indication from child 0xDBEF
09:22:58:872 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xDBEF
09:22:58:873 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:874                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:58:875 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) cluster 0x8005
09:22:58:899 [3] get simple descriptor 0x01 for 0x5c0272fffec9c9dc
09:22:58:901 ZDP get simple descriptor 0x01 for 0xDBEF
09:22:59:125 APS-DATA.indication from child 0xDBEF
09:22:59:127 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xDBEF
09:22:59:127 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:59:128                       ...     (0x5C0272FFFEC9C9DC / 0xDBEF)
09:22:59:129 ZDP indication search sensors 0x5C0272FFFEC9C9DC (0xDBEF) cluster 0x8005
09:22:59:153 wait response fastEnddeviceProbe() 0x5C0272FFFEC9C9DC, elapsed 250 ms
09:22:59:376 APS-DATA.indication from child 0xDBEF
09:22:59:377 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xDBEF
09:22:59:378 FP indication 0x0000 / 0x8005 (0x5C0272FFFEC9C9DC / 0xDBEF)
[... cut ...] ":"5c:02:72:ff:fe:c9:c9:dc-01-ef00"} (ret = 2123813652)
09:23:06:856 don't create binding for attribute reporting of sensor Battery 7
09:23:06:859 MAC poll fastEnddeviceProbe() 0x5C0272FFFEC9C9DC
09:23:06:860 wait response fastEnddeviceProbe() 0x5C0272FFFEC9C9DC, elapsed 5881 ms
09:23:06:861 skip check bindings for client clusters (no group)
09:23:06:862 don't create binding for attribute reporting of sensor Battery 7
09:23:06:863 don't create binding for attribute reporting of sensor Battery 7`

Smanar, 2022-03-03:
Hello, so you have inclusion issue ?
Can you try to make a DDF for this device ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

I can't say if it will work because there is a hack in the code for this device
Try to use as product name "Tuya_COVD AM43-0.45-40"

You just need to create the DDf and drag and drop a window covering type, no configuration, no bind, no report.

chris199512, 2022-03-03:
yes. i used it on my old system. now i have a new deconz and it dont work. i will test it with ddf next days. thank you

henkdenerd, 2022-05-23:
> Yesss, that's it. Works great !! Just to know how to do it. Thank you for the effort and quick reply.

What was the trick exactly? I added the device in the deconz app  through phoscon, pressed "read" two times and then press find new light in phoscon.. nothing happens. How does this work?

Smanar, 2022-05-24:
You general need to  make 2 inclusions without deleting the device between.

Else if you can start a DDF I can finish it (I need at least a skeleton)

henkdenerd, 2022-05-24:
I just saw the link for the DDF instructions, I will take a look at this tomorrow and send you feedback  if  it worked.

Smanar, 2022-05-25:
No, on deconz there is only 1 node by device.

Deconz have problem with tuya stuff, because they don't respect zigbee rules, the firsts informations deconz obtain are useless and deconz have problem to reconise the device.
There is 2 situations.
- deconz have created the battery device, so this entry is now in the API, so if you don't delete it, on the next inclusion deconz will use this entry in the database to reconize the light entry (the covering)
- Unfortunately there is some situation where this device block the inclusion, so if it realy don't want to work, delete the battery entry and restart fom scratch.

Honnestly the inclusion on the legacy code is realy random, I haven't a better procedure than another for tuya stuff.

And another bad news for you, you can forget the DDF, this device use too the tuya cluster and this one is not ready yet to be used with DDF for covering.

Johan-Arens, 2023-01-11:
Hi, I'm trying to pair some Moes AM43 to Deconz, one motor worked, the other ones don't. Did we have bug fixed recently with this device ? The process that worked for one motor was to add it as a switch then restart the pairing process to add as a light, it showed up as a smart plug and is exposed to HA. I can't reproduce the method to my other motors. Any idea the step I would have missed ? Thanks !

Smanar, 2023-01-11:
No, they are using the old system, and IDK what can happen if we make a DDF to include it. The DDF can help for inclusion but not possible to use the DDF during the working mode.

Generaly, you need to include the device at least one time for deconz have the manufacture name, then include it a second time and deconz will use the manufacture name already in database to reconize it.
So try to include it 2 time without deleting it between both inclusion.

Are you sure all device are the same model ? Tuya use lot of clones.

Johan-Arens, 2023-01-11:
I've tried to include them 2 times in a row, multiple times. 
Regarding the Manufacturer : _TZE200_zah67ekd
Model TS0601
They looked to the be the same model, reported by Deconz (in the basic information) and from the casing (doubled check as well). 
I'm running 2.19.3 (installed with the HA add-on), my Conbee runs firmware 26580700. 

Thanks !

Smanar, 2023-01-11:
This is the sensor part.
On the light part you need to have in log (visible with info_l2 and "info)

```
DBG_Printf(DBG_INFO_L2, "Tuya debug 7 : Missing manufacture name for 0x%016llx\n", node->address().ext());
DBG_Printf(DBG_INFO, "DeviceAnnce of LightNode: %s Permit Join: %i\n", qPrintable(i->address().toStringExt()), gwPermitJoinDuration);
```

If you have the GUI and if you have the ZHAbattery
- set phoscon in permit join (add new light)
- read the descriptors (on the menu, on the first circle on the node title)

Johan-Arens, 2023-01-11:
This one ? 

Johan Arens
***@***.***

Johan-Arens, 2023-01-11:
[image]
[image]

Johan-Arens, 2023-01-11:
I don't have ZHA-Battery, is it part of the zha integration in HA ?

Johan-Arens, 2023-01-11:
Ok I figured out with the node descriptor thing. Here is what I've tried, deleting the node, re-adding the node, click on read description, re-starting the pairing process as light. No luck, the motor is stays as battery. 
I've tried to also only try to add the node as a light, same, the node shows up as battery device.

Johan-Arens, 2023-01-12:
OK found the way to do this. When adding a new motor, once it is detected and seen as a battery, right-click and point to read simple descriptor, that turns the node into a Smart Plug. 
Where could I push that information so that would be available for everyone ?

Smanar, 2023-01-12:
>Where could I push that information so that would be available for everyone ?

I think it's the good place.
The problem is this old system is not reliable, it have worked for you but still can fail.
The DDF core is better but not available yet for lot of tuya stuff.
On my side I hope for a tuya support  by DDF faster as possible.

Johan-Arens, 2023-03-16:
Hey guys, I have a strange issue with 2 motors, I'd like to pickup your brain. With the motors paired to deConz and exposed to HA, I was able to use the motors. 
However, since last week, one motor can't be controlled anymore, it is exposing it's state to HA (open, close, position) but any up/down orders are ignored. I've tried to delete the node, repairing it, factory default, repairing it, rebooting the deconz add-on, restarting the integration in HA,  nothing. 

Another motor is seen as disconnected in HA, but it is well there in deConz, I see it as active, I can read descriptors etc. 

Any idea what could be going on ? 

Thanks ! 

Johan

Mimiix, 2023-03-16:
Hi,

Please open a forum topic for device specific help.

Locking this up.

Kind regards,