dresden-elektronik/deconz-rest-plugin #4810: 4 Gang Tuya ZigBee Wireless 12 Scene Switch

mbriney, 2021-04-25:
## Device

- Product name: 4 Gang Tuya ZigBee Wireless 12 Scene Switch
- Manufacturer: unknown
- Model identifier: blank
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch
- URL: https://www.aliexpress.com/item/1005001639819248.html?spm=a2g0s.9042311.0.0.5da14c4dsUQrOq
- 

## Screenshots

- Device: https://www.awesomescreenshot.com/image/7626845?key=0d740606d13e2707e34264fdd431076c
- Basic: https://www.awesomescreenshot.com/image/7626790?key=15fa02139d432ee5713a1292576c2944
- Power: https://www.awesomescreenshot.com/image/7626805?key=3835e6a37fcdf7dfe8728c3bdfb6c642
- Identify: https://www.awesomescreenshot.com/image/7626812?key=842b8c7e11697d386c0d15e4be7f347d
- On/Off: https://www.awesomescreenshot.com/image/7626828?key=d0e16ebb3403d95568e86e1024e4e8c5
- OTAU: https://www.awesomescreenshot.com/image/7626828?key=d0e16ebb3403d95568e86e1024e4e8c5

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

mbriney, 2021-04-25:
Someone reported it works well with Hubitat recognizing each button with the three actions (single, double, hold)
https://www.amazon.com/gp/customer-reviews/R1COWVHU2VTJKA/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B08P4SRGFQ

Smanar, 2021-04-25:
Hello, I assume the manufacture number is 0x1002 visible on node info panel ?

Are you able to compile the code to test it ? You need a linux machine with full OS (There is some thing that look strange on this device)

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_switch_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

Remember you need to edit your json file, path = /usr/share/deCONZ/devices/button_maps.json

mbriney, 2021-04-26:
Yes it might take a few days to get started as I want to order a second conbee so that I don't have to pull my production one out of service.

Smanar, 2021-04-27:
You can just switch the conbee from the production to the dev machine for 10mn , time for tests.

But yes I understand, I can wait, I can too make the PR in blind mode if it s realy not possible to test it, but results can be random like it.

easybeat, 2021-04-28:
I have integrated one of them succesfully today:

[image]

Smanar, 2021-04-28:
Yep, but yours is in the code, no his one ^^.
But they work same, 99% de chance for the PR will work.

easybeat, 2021-04-28:
Well, I'm using ioBroker with deconz Adapter and in there it works without any problems

luka6000, 2021-05-22:
any news on this one? On 2.11.5 this switch doesn't work. Detected as single light (not switch/button!). And physically pushing button 1 turns on every light. Button 2 switches everything off. How can I check or disable this behavior?

Smanar, 2021-05-22:
Hello, I m still waiting for some one to test the code ?
BTW do you have a linux machine with full OS ?

luka6000, 2021-05-22:
```
pi@raspberrypi:~/deconz-rest-plugin $ make
make -f Makefile.Release
make[1]: Entering directory '/home/pi/deconz-rest-plugin'
/usr/lib/qt5/bin/uic -no-stringliteral de_web_widget.ui -o ui_de_web_widget.h
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.11.01\" -DGW_SW_DATE=1619356943 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"4c99730c1a62ffff9edb0299fe242b9b253ac557\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/air_quality.o air_quality.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.11.01\" -DGW_SW_DATE=1619356943 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"4c99730c1a62ffff9edb0299fe242b9b253ac557\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/authorisation.o authorisation.cpp
authorisation.cpp: In member function ‘bool DeRestPluginPrivate::allowedToCreateApikey(const ApiRequest&, ApiResponse&, QVariantMap&)’:
authorisation.cpp:80:39: error: no matching function for call to ‘QHttpRequestHeader::hasKey(const char [14]) const’
     if (req.hdr.hasKey("Authorization"))
                                       ^
In file included from /usr/include/deconz.h:39,
                 from de_web_plugin_private.h:24,
                 from authorisation.cpp:13:
/usr/include/deconz/qhttprequest_compat.h:47:10: note: candidate: ‘bool QHttpRequestHeader::hasKey(const QLatin1String&) const’
     bool hasKey(const QLatin1String &key) const;
          ^~~~~~
/usr/include/deconz/qhttprequest_compat.h:47:10: note:   no known conversion for argument 1 from ‘const char [14]’ to ‘const QLatin1String&’
authorisation.cpp:82:55: error: no matching function for call to ‘QHttpRequestHeader::value(const char [14]) const’
         QStringList ls = req.hdr.value("Authorization").split(' ');
                                                       ^
In file included from /usr/include/deconz.h:39,
                 from de_web_plugin_private.h:24,
                 from authorisation.cpp:13:
/usr/include/deconz/qhttprequest_compat.h:54:19: note: candidate: ‘QLatin1String QHttpRequestHeader::value(const QLatin1String&) const’
     QLatin1String value(const QLatin1String &key) const;
                   ^~~~~
/usr/include/deconz/qhttprequest_compat.h:54:19: note:   no known conversion for argument 1 from ‘const char [14]’ to ‘const QLatin1String&’
authorisation.cpp: In member function ‘void DeRestPluginPrivate::authorise(ApiRequest&, ApiResponse&)’:
authorisation.cpp:166:48: error: no matching function for call to ‘QHttpRequestHeader::hasKey(const char [11]) const’
                 if (req.hdr.hasKey("User-Agent"))
                                                ^
In file included from /usr/include/deconz.h:39,
                 from de_web_plugin_private.h:24,
                 from authorisation.cpp:13:
/usr/include/deconz/qhttprequest_compat.h:47:10: note: candidate: ‘bool QHttpRequestHeader::hasKey(const QLatin1String&) const’
     bool hasKey(const QLatin1String &key) const;
          ^~~~~~
/usr/include/deconz/qhttprequest_compat.h:47:10: note:   no known conversion for argument 1 from ‘const char [11]’ to ‘const QLatin1String&’
authorisation.cpp:168:62: error: no matching function for call to ‘QHttpRequestHeader::value(const char [11]) const’
                     i->useragent = req.hdr.value("User-Agent");
                                                              ^
In file included from /usr/include/deconz.h:39,
                 from de_web_plugin_private.h:24,
                 from authorisation.cpp:13:
/usr/include/deconz/qhttprequest_compat.h:54:19: note: candidate: ‘QLatin1String QHttpRequestHeader::value(const QLatin1String&) const’
     QLatin1String value(const QLatin1String &key) const;
                   ^~~~~
/usr/include/deconz/qhttprequest_compat.h:54:19: note:   no known conversion for argument 1 from ‘const char [11]’ to ‘const QLatin1String&’
make[1]: *** [Makefile.Release:579: release/authorisation.o] Error 1
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2
pi@raspberrypi:~/deconz-rest-plugin $

```

Smanar, 2021-05-23:
Ha yes sorry, it s normal, the dev package have moved.

```
cd deconz-rest-plugin
git pull
make clean
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Smanar, 2021-05-23:
Have submitted too the PR, as manup is validating PR ATM, I don't think we will have problem, there is this kind of addition every week.

The only thing where I m not sure is 

>And physically pushing button 1 turns on every light. Button 3 switches everything off. How can I check or disable this behavior?

luka6000, 2021-05-24:
ok, got it. So the situation is like this

 - Host system: Raspberry Pi3
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.11.5-beta + included fix from this issue
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

[image]
In Phoscon it's light
[image]
in the API it's light and sensor
```
{
    "config": {
        "UTC": "2021-05-24T07:24:05",
        "apiversion": "1.16.0",
        "backup": {
            "errorcode": 0,
            "status": "idle"
        },
        "bridgeid": "00212EFFFF0725AA",
        "datastoreversion": "93",
        "devicename": "ConBee II",
        "dhcp": true,
        "disablePermitJoinAutoOff": false,
        "factorynew": false,
        "fwversion": "0x26680700",
        "gateway": "192.168.0.1",
        "internetservices": {
            "internet": "connected",
            "remoteaccess": "disconnected",
            "swupdate": "connected",
            "time": "connected"
        },
        "ipaddress": "192.168.0.31",
        "lightlastseeninterval": 60,
        "linkbutton": false,
        "localtime": "2021-05-24T07:24:05",
        "mac": "b8:27:eb:30:a5:ab",
        "modelid": "deCONZ",
        "name": "Phoscon-dev",
        "netmask": "255.255.255.0",
        "networkopenduration": 180,
        "panid": 63009,
        "portalconnection": "disconnected",
        "portalservices": false,
        "portalstate": {
            "communication": "disconnected",
            "incoming": false,
            "outgoing": false,
            "signedon": false
        },
        "proxyaddress": "none",
        "proxyport": 0,
        "replacesbridgeid": null,
        "rfconnected": true,
        "starterkitid": "",
        "swupdate": {
            "checkforupdate": false,
            "devicetypes": {
                "bridge": false,
                "lights": [],
                "sensors": []
            },
            "notify": false,
            "text": "",
            "updatestate": 0,
[... 277 lines cut ...]
07:59:56:422 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0006/0x0000 after 0 s
07:59:56:425 enqueue event event/checkgroupanyon for /groups/65520
07:59:56:427 enqueue event event/checkgroupanyon for /groups/1
07:59:56:429 ZCL attribute report 0x680AE2FFFE8DF77D for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
07:59:56:431 	payload: 00001000
07:59:56:433 MS CORE_APS_IND (processPacked:985) took 19 ms
07:59:56:434 enqueue event state/all_on for /groups/65520
07:59:56:436 enqueue event state/any_on for /groups/65520
07:59:56:498 enqueue event state/all_on for /groups/1
07:59:56:505 enqueue event state/any_on for /groups/1
07:59:56:516 don't create binding for attribute reporting of sensor TS004F 2
07:59:56:544 Websocket 192.168.0.36:61494 send message: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 2126787808)
07:59:56:556 discard group state push for 65520: state/any_on (already pushed)
07:59:56:576 Websocket 192.168.0.36:61494 send message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 2126787808)
07:59:56:585 don't create binding for attribute reporting of sensor TS004F 2
07:59:56:598 discard group state push for 1: state/any_on (already pushed)
07:59:57:072 don't create binding for attribute reporting of sensor TS004F 2
07:59:57:231 enqueue event config/localtime for /config/
07:59:57:233 Wait 0s till query finished
07:59:57:304 poll node 60:a4:23:ff:fe:f6:f8:93-01
07:59:58:754 MS CORE_APS_IND (processPacked:985) took 0 ms
08:00:01:153 MS CORE_APS_IND (processPacked:985) took 0 ms

```
@Smanar, let me know if you would like to see some more.

Smanar, 2021-05-24:
Ha yep, sorry,have forget to say something.

>07:59:55:065 [INFO] - No button map for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 92

This is because you need to edit the json file
On raspian it s /usr/share/deCONZ/device/button_maps.json

The modification is here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4908/files first part.
After that when pressing the button you will see values in "buttonevent"

The light entry is a bug, but it s "normal", ATM deconz have problem to reconize tuya device, it s for that we need 2 inclusions, but if I m right, now deconz is now able to reconize it (already in database), so you can remove the "light" entry, and re-include the device is needed.

For ikea reaction, still searching.

luka6000, 2021-05-24:
Aqara OPPLE has this mode
[image]
Looks similar.

For now I've used group removal and this seems to work and not switching lights
[image]
Can you add this step to provisioning/onboarding while including switch to the network?

Smanar, 2021-05-24:
Add button are working now ?

>Can you add this step to provisioning/onboarding while including switch to the network?

That's my problem, he is by defaut, when you include a device, you reset it, so you loose all previous group/setting/ect ...
The opple have the "group command" enabled by defaut, and it need to be disabled using code, but I never see it for tuya device.

It s for that, I have some worry, Idk if your device need special code, or if the issue is from previous inclusions.
For me it s from inclusions, because it use API group command (the hidden group) and not native zigbee like the opple.

luka6000, 2021-05-24:
> Add button are working now ?
> 

After:
1. Tyua switch Basic cluster "Reset to factory defaults"
2. Gateway reset
3. Setting up everything from scratch
I have this
[image]

and buttons control light without automation as previously.
Button map is working
```
15:14:41:454 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 45
15:14:41:495 ZCL attribute report 0x680AE2FFFE8DF77D for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:14:41:739 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 46
15:14:42:006 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 00330A00, zclSeq: 47
15:14:42:316 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 48
15:14:42:494 ZCL attribute report 0x680AE2FFFE8DF77D for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

```

> It s for that, I have some worry, Idk if your device need special code, or if the issue is from previous inclusions.
> For me it s from inclusions, because it use API group command (the hidden group) and not native zigbee one like the opple.

It looks like by default not only ONOFF works but also Level control
```
15:20:00:508 MS CORE_APS_IND (processPacked:985) took 2 ms
15:20:02:113 MS CORE_APS_IND (processPacked:985) took 3 ms
15:20:03:815 MS CORE_APS_IND (processPacked:985) took 3 ms
15:20:05:508 MS CORE_APS_IND (processPacked:985) took 2 ms
15:20:07:111 MS CORE_APS_IND (processPacked:985) took 3 ms
15:20:07:418 enqueue event config/localtime for /config/
15:20:07:622 poll node 60:a4:23:ff:fe:f6:f8:93-01
15:20:08:196 poll node 68:0a:e2:ff:fe:8d:f7:7d-01
15:20:08:198 Poll light node Color temperature light 2
15:20:08:249 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0006 dropped, values are fresh enough
15:20:08:420 enqueue event config/localtime for /config/
15:20:08:561 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0008 dropped, values are fresh enough
15:20:08:659 read attributes of 0x680AE2FFFE8DF77D cluster: 0x0300: [ 
15:20:08:660 0x0003 
15:20:08:661 0x0004 
15:20:08:662 0x0007 
15:20:08:662 0x0008 
15:20:08:663 0x4001 
15:20:08:664 ]
15:20:08:664 add task 78981 type 19 to 0x680AE2FFFE8DF77D cluster 0x0300 req.id 251
15:20:08:665 Poll APS request 251 to 0x680AE2FFFE8DF77D cluster: 0x0300
15:20:08:857 Poll APS confirm 251 status: 0x00
15:20:08:858 Erase task req-id: 251, type: 19 zcl seqno: 206 send time 0, profileId: 0x0104, clusterId: 0x0300
15:20:08:916 Node data 0x680ae2fffe8df77d profileId: 0x0104, clusterId: 0x0300
15:20:08:919 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0300/0x0003 after 0 s
15:20:08:920 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0300/0x0004 after 0 s
15:20:08:921 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0300/0x0007 after 0 s
15:20:08:922 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0300/0x0008 after 0 s
15:20:08:923 0x680AE2FFFE8DF77D: update ZCL value 0x01/0x0300/0x4001 after 0 s
15:20:08:925 MS CORE_APS_IND (processPacked:985) took 10 ms
15:20:09:418 enqueue event config/localtime for /config/
15:20:09:519 poll node 60:a4:23:ff:fe:f6:f8:93-01
15:20:10:042 poll node 68:0a:e2:ff:fe:8d:f7:7d-01
15:20:10:043 Poll light node Color temperature light 2
15:20:10:095 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0006 dropped, values are fresh enough
15:20:10:414 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0008 dropped, values are fresh enough
15:20:10:434 enqueue event config/localtime for /config/
15:20:10:526 read attributes of 0x680AE2FFFE8DF77D cluster: 0x0300: [ 
15:20:10:529 0x0003 
15:20:10:531 0x0004 
15:20:10:532 0x0007 
15:20:10:533 0x0008 
15:20:10:534 0x4001 
[... 168 lines cut ...]
15:28:13:762 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 83
15:28:13:763 MS CORE_APS_IND (processPacked:985) took 6 ms
15:28:13:813 Websocket 192.168.0.10:1027 send message: {"e":"changed","id":"2","r":"lights","state":{"bri":252,"colormode":"ct","ct":370,"on":false,"reachable":true},"t":"event","uniqueid":"68:0a:e2:ff:fe:8d:f7:7d-01"} (ret = 35824848)
15:28:13:819 enqueue event event/checkgroupanyon for /groups/65520
15:28:13:823 enqueue event event/checkgroupanyon for /groups/1
15:28:13:826 poll node 68:0a:e2:ff:fe:8d:f7:7d-01
15:28:13:828 Poll light node Color temperature light 2
15:28:13:926 enqueue event state/all_on for /groups/65520
15:28:13:934 enqueue event state/any_on for /groups/65520
15:28:13:941 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0006 dropped, values are fresh enough
15:28:14:003 enqueue event state/all_on for /groups/1
15:28:14:013 enqueue event state/any_on for /groups/1
15:28:14:088 Websocket 192.168.0.10:1027 send message: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 2125841632)
15:28:14:135 discard group state push for 65520: state/any_on (already pushed)
15:28:14:184 Websocket 192.168.0.10:1027 send message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 2125841632)
15:28:14:227 discard group state push for 1: state/any_on (already pushed)
[... cut ...] nfig/localtime for /config/
15:28:15:105 poll node 68:0a:e2:ff:fe:8d:f7:7d-01
15:28:15:106 Poll light node Color temperature light 2
15:28:15:158 Poll APS request to 0x680AE2FFFE8DF77D cluster: 0x0006 dropped, values are fresh enough
15:28:15:415 enqueue event config/localtime for /config/
15:28:16:247 MS CORE_APS_IND (processPacked:985) took 0 ms
15:28:32:035 MS CORE_APS_IND (processPacked:985) took 0 ms
```
Bulb is not changing state - it's off all the time. But state shown by Phoscon changes.

Smanar, 2021-05-24:
Just to be sure, your device is a battery one ?

You have edited the Json file ? If yes, it s strange you still have on log
>15:28:13:762 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 83

Do not reset the gateway, deconz need to use previous data.
So you can include the device, delete the "light" entry if you want but re-include it. If I m right on the second inclusion you will have only the switch entry.

Will check if I found information to see if this device need "magic" after inclusion.

luka6000, 2021-05-24:
> Just to be sure, your device is a battery one ?
> 

> You have edited the Json file ? If yes, it s strange you still have on log
> 
> > 15:28:13:762 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 83
> 

```
pi@raspberrypi:~ $ grep -B3 -A15 xabckq1v /usr/share/deCONZ/devices/button_maps.json
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_xabckq1v"],
            "map": [
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],
                [1, "0x01", "ONOFF", "0xfd", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "B1 double"],
                [1, "0x01", "ONOFF", "0xfd", "2", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "B1 long"],
                [1, "0x02", "ONOFF", "0xfd", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 short"],
                [1, "0x02", "ONOFF", "0xfd", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "B2 double"],
                [1, "0x02", "ONOFF", "0xfd", "2", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "B2 long"],
                [1, "0x03", "ONOFF", "0xfd", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3 short"],
                [1, "0x03", "ONOFF", "0xfd", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "B3 double"],
                [1, "0x03", "ONOFF", "0xfd", "2", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3 long"],
                [1, "0x04", "ONOFF", "0xfd", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4 short"],
                [1, "0x04", "ONOFF", "0xfd", "1", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "B4 double"],
                [1, "0x04", "ONOFF", "0xfd", "2", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4 long"]
            ]
        },
pi@raspberrypi:~ $
```

> Do not reset the gateway for future test, deconz need to use previous data (now we know the group command is not from API stuff as you still have it after the reset)
> So you can include the device, delete the "light" entry if you want but re-include it. If I m right on the second inclusion you will have only the switch entry.
> 
> Will check if I found information to see if this device need "magic" after inclusion.
> 
> Tips:
> If you still have broadcast request that trigger the bulb, you try to bind the remote to the coordinator direclty, cluster on/off and setlevel.

ok. I can see this is already closed, so let me grab the newest code and check that one. I'll open new issue if anything will go wrong, ok?

Smanar, 2021-05-24:
Ok, so it s realy a different tuya device, need to rewrite a part of the code, from the last official.
Have re opened the issue (was my fault, was sure it was a standard tuya switch)

Give me 15 mn to send you  a new code.

Smanar, 2021-05-24:
Ok, so new code is online, sorry I have added perhaps 10 tuya switch so for me yours was same.
The code is based on last official so to switch branch

```
cd deconz-rest-plugin
git pull
git checkout tuya_switch_new
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

So have added a "magic" code to disable the broadcast command (the thing that make the ikea buld react) during inclusion, have started the json file according to log I have.

You have all changes here https://github.com/Smanar/deconz-rest-plugin/commit/ad48441426a20703c7a29949da506ac155c03af4

The new button map here https://github.com/Smanar/deconz-rest-plugin/blob/master/button_maps.json

It s just a json file, easy to understand all it work else
- enable log "info"
- press a button
- take a look at line 
>15:20:15:237 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 53

I need the button you have used and the log according to it to finish the json file.

luka6000, 2021-05-24:
> Have added a "magic" code to disable the broadcast command (the thing that make the ikea buld react) during inclusion, have started the json file according to log I have.
> 

ok, fresh start. Gateway reset, devices added with new plugin.
It's the same - switch added as light and ikea bulb reacting to buttons.
After a while and deconz restart, light Dimmer switch is gone leaving only sensor 
```
        "2": {
            "config": {
                "battery": 100,
                "group": "65520",
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "fa6137922ef92d823a07a35c4179204c",
            "lastseen": "2021-05-24T22:34Z",
            "manufacturername": "_TZ3000_xabckq1v",
            "mode": 1,
            "modelid": "TS004F",
            "name": "TS004F 2",
            "state": {
                "buttonevent": 1004,
                "lastupdated": "2021-05-24T22:34:18.994"
            },
            "type": "ZHASwitch",
            "uniqueid": "60:a4:23:ff:fe:f6:f8:93-01-1000"
        }
```

> I need the button you have used and the log according to it to finish the json file.

Buttons pushed in the following order

short click
```
22:38:35:000 [INFO] - Button 1004 - TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), action: B2, payload: None, zclSeq: 86
22:38:35:492 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 87
22:38:35:935 [INFO] - Button 1003 - TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 88
22:38:36:405 [INFO] - Button 1002 - TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1, payload: None, zclSeq: 89
```

double click
no actions in the log

long press and release on 1 and 4 does nothing. Buttons 2 and 3 as follows
```
22:54:08:925 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0133, zclSeq: 124
22:54:11:784 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 125
22:54:16:268 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0033, zclSeq: 126
22:54:18:473 [INFO] - No button handler for: TS004F, broadcast to: 0xFFF0, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 127
```
So it's not a "12 scene" switch after all :-/

Smanar, 2021-05-25:
>So it's not a "12 scene" switch after all :-/

IDK, still missing the "double clic" and there is nothing about the cluster Scene 0x0005.

So first, have updated the JSON file > https://github.com/Smanar/deconz-rest-plugin/commit/d252096f8f08241a2413117fa61cd96eb9982aa0

>It's the same - switch added as light and ikea bulb reacting to buttons.

Ok, do you know how to make bind ? (something like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916)

To prevent the device make braodcast command I make bind from it to the coordinator for cluster on/off + setlevel+ scene.
But it seem not working (and your device continu to make request in broadcast instead of unicast) and Scene cluster don't work at all.

So if you can try to re-do them manualy with the GUI (you probably need to awake the device in same time) to find the usefull one.

Im searching information about this device on other project, but nothing, this device haven't a special C button like the opple ?

luka6000, 2021-05-25:
> IDK, still missing the "double clic" and there is nothing about the cluster Scene 0x0005.
> 

double click does nothing - no log entries.
There is no Scene server cluster, look my screenshot here
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4810#issuecomment-846859026

> So first, have updated the JSON file > [Smanar@d252096](https://github.com/Smanar/deconz-rest-plugin/commit/d252096f8f08241a2413117fa61cd96eb9982aa0)
> 

I think for LEVEL_CONTROL STOP it should be S_BUTTON_ACTION_LONG_RELEASED

> Ok, do you know how to make bind ? (something like [#2077 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916))
> 
> To prevent the device make braodcast command I make bind from it to the coordinator for cluster on/off + setlevel+ scene.
> But it seem not working (and your device continu to make request in broadcast instead of unicast) and Scene cluster don't work at all.
> 
> So if you can try to re-do them manualy with the GUI (you probably need to awake the device in same time) to find the usefull one.
> 

What about this?
```
        "2": {
            "config": {
                "battery": 100,
                "group": "65520",
                "on": true,
                "reachable": true
            },
```
Isn't this "group" entry making it broadcast?
I've made it stop broadcasting be removing all groups, as mentioned here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4810#issuecomment-846914401

> Im searching information about this device on other project, but nothing, this device haven't a special C button like the opple ?

No additional buttons. It's pretty clean design
[image]

Smanar, 2021-05-26:
>There is no Scene server cluster, look my screenshot here

You have it, blue are input cluster, grey are output, you have it in grey.

>I think for LEVEL_CONTROL STOP it should be S_BUTTON_ACTION_LONG_RELEASED

Yes, right but for the moment I m trying to retreive all events, I need to change too "B1" "B1L" and ect ....

>I've made it stop broadcasting be removing all groups

And only this is working but work in all time ?
The problem is if the device use the grey cluster (output) for scene, you need to bind this cluster to the coordinator, else no devices can see it.
It s for that will be nice if you can try it too to be sure, else I can use the "clean group" method, but no device use this method yet.

>Isn't this "group" entry making it broadcast?

Oups, I have see I have enabled this feature too. This is the "magic" group, all light in this group will be binded with the remote, but I don't think the ikea bulb is inside this one.

luka6000, 2021-05-26:
> Yes, right but for the moment I m trying to retreive all events, I need to change too "B1" "B1L" and ect ....
> 

So it's like that
short release for 1-2-3-4
```
19:34:29:358 [INFO] - Button 1004 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B2, payload: None, zclSeq: 108
19:34:31:161 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 109
19:34:32:039 [INFO] - Button 1003 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 110
19:34:32:826 [INFO] - Button 1002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1, payload: None, zclSeq: 111
```
hold for 2-3
hold release for 2-3
```
19:36:22:329 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0133, zclSeq: 123
19:36:25:068 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 124
19:36:29:368 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0033, zclSeq: 125
19:36:32:104 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 126
```
double click does nothing

> And only this is working but work in all time ?
> The problem is if the device use the grey cluster (output) for scene, you need to bind this cluster to the coordinator, else no devices can see it.
> It [...]

Started all over again.
reset gateway
add bulb
add switch
This is state after
[image]
```
{
    "config": {
        "UTC": "2021-05-26T19:26:54",
        "apiversion": "1.16.0",
        "backup": {
            "errorcode": 0,
            "status": "idle"
        },
        "bridgeid": "00212EFFFF0725AA",
        "datastoreversion": "93",
        "devicename": "ConBee II",
        "dhcp": true,
        "disablePermitJoinAutoOff": false,
        "factorynew": false,
        "fwversion": "0x26680700",
        "gateway": "192.168.0.1",
        "internetservices": {
            "internet": "connected",
            "remoteaccess": "disconnected",
            "swupdate": "connected",
            "time": "connected"
        },
        "ipaddress": "192.168.0.31",
        "lightlastseeninterval": 60,
        "linkbutton": false,
        "localtime": "2021-05-26T19:26:54",
        "mac": "b8:27:eb:30:a5:ab",
        "modelid": "deCONZ",
        "name": "Phoscon-dev",
        "netmask": "255.255.255.0",
        "networkopenduration": 180,
        "panid": 42882,
        "portalconnection": "disconnected",
        "portalservices": false,
        "portalstate": {
            "communication": "disconnected",
            "incoming": false,
            "outgoing": false,
[... 491 lines cut ...]
        "3": {
            "config": {
                "battery": null,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "d7a1ef0150bda510a6686a942084b272",
            "lastseen": "2021-05-26T19:31Z",
            "manufacturername": "_TZ3000_xabckq1v",
            "mode": 1,
            "modelid": "TS004F",
            "name": "TS004F 3",
            "state": {
                "buttonevent": 1002,
                "lastupdated": "2021-05-26T19:31:13.050"
            },
            "type": "ZHASwitch",
            "uniqueid": "60:a4:23:ff:fe:f6:f8:93-01-1000"
        }
    }
}
```

It's all good now, switch is not turning on the bulb.

Smanar, 2021-05-26:
>It's all good now, switch is not turning on the bulb.

So delete the device (light + switch), and re-include it (only switch appear), works ?

But I don't understand how you can have exactly the same log when pressing button ? I have changed 3 button / 4, and you have exaclty same logs than before my change ?
https://github.com/Smanar/deconz-rest-plugin/commit/d252096f8f08241a2413117fa61cd96eb9982aa0

romanho, 2021-06-06:
Hope you don't mind me jumping in here...

I have the same Tuya switch" _TZ3000_xabckq1v" and I've pulled + compiled the newest REST plugin as described above, but still the switch isn't working.

Effects here are:
- it's not shown in Phoscon web app, but I can see a TS004F device in deCONZ
- in deCONZ info output I see the same "no button handler" as luka600, f.i. (short press for all 4 buttons):

 `20:34:13:925 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 72`
 `20:34:17:065 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 73`
 `20:34:18:236 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 00330A00, zclSeq: 74`
 `20:34:19:268 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 75`

I've checked that my button_map.json is listing the xabckq1v device and that deconz is actually reading this file. Also libde_rest_plugin.so contains at least the string, so is probably the modified version.

But still all seems as nothing changed compared to before me applying the patch...

Smanar, 2021-06-07:
@romanho 
Hello, you have used the branch tuya_switch_new ?
You have replaced the json file too ?

Because he have at least the ONOFF working

```
19:34:29:358 [INFO] - Button 1004 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B2, payload: None, zclSeq: 108
19:34:31:161 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 109
19:34:32:039 [INFO] - Button 1003 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 110
19:34:32:826 [INFO] - Button 1002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1, payload: None, zclSeq: 111
```

romanho, 2021-06-07:
> you have used the branch tuya_switch_new ?

I'm sorry, actually no. I've seen "tuya_switch_1" somewhere in this thread and found it already merged, so I had cloned the the official distribution, not your fork.

Now I've pulled that one in and checked out the "tuya_switch_new" branch, build + copied libde_rest_plugin.so and button_maps.json, and now I have a similar but not equal output:

  `23:14:35:180 [INFO] - Button 1002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1, payload: None, zclSeq: 102`
  `23:14:36:158 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B2, payload: None, zclSeq: 103`
  `23:14:36:952 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 104`
  `23:14:37:635 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 105`

Also button 4 is missing, but my button numbers are different: 2002 vs. 1002 and 3002 vs. 1003. Does that matter, or what does it tell?

Double press still triggers no reaction at all, and long press on the two right buttons gives:

  `23:24:21:531 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0033, zclSeq: 122`
  `23:24:23:477 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 123`
  `23:24:28:604 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 0133, zclSeq: 124`
  `23:24:32:521 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: None, zclSeq: 125`

Could you tell me what the fields in the button map are for? Then I could try a bit for myself to maybe fix the mappings.

Smanar, 2021-06-08:
>Also button 4 is missing, but my button numbers are different: 2002 vs. 1002 and 3002 vs. 1003. Does that matter, or what does it tell?

Yep, don't worry, it s just because the json file have changed since.

This part is for short press

```
[1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1"],
[1, "0x01", "ONOFF", "OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2"],
[1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3"],
[1, "0x02", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"],
```

The not working is this one
`[1, "0x02", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"]`

But seriously all seem fine

> 23:14:37:635 [INFO] - No button handler for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01330A00, zclSeq: 105

- cluster: LEVEL_CONTROL (0x0008)
- command: STEP (0x02)
- payload: 01330A00 (it use only the first byte, so 0x01)

romanho, 2021-06-08:
> `Have found the problem, the endpoint, try with replacing all 0x02 by 0x01`

I had added some DBG_Printf()s in checkSensorButtonEvent() today, and basically found out the same thing now. As I wanted to post it here now, I've seen your edit -- obviously I didn't recheck soon enough... :-O

Anyway, I can confirm that all 4 buttons deliver an action for short press now:

  `00:01:09:582 [INFO] - Button 1002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1, payload: None, zclSeq: 87`
  `00:01:10:759 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B2, payload: None, zclSeq: 88`
  `00:01:13:089 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 89`
  `00:01:13:987 [INFO] - Button 4002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B4, payload: 01330A00, zclSeq: 90`

Even the two right buttons give an action for long press and release of this:

  `00:01:42:170 [INFO] - Button 2001 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B2H, payload: 0033, zclSeq: 91`
  `00:01:42:939 [INFO] - Button 2002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B2R, payload: None, zclSeq: 92`
  `00:02:05:665 [INFO] - Button 3001 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3L, payload: 0133, zclSeq: 93`
  `00:02:06:821 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3R, payload: None, zclSeq: 94`

So quite some progress :-) But still nothing on double press or long press of the left buttons. Can anything be done about those? Does it make sense to play around with the output clusters, or something else?

Smanar, 2021-06-09:
Perhaps its just not supported.

>Does it make sense to play around with the output clusters, or something else?

Yes can try to bind other cluster than onoff and levelcontrol (rest the scene cluster), to check if you have more return in log, but on the code It already done.

```
    else if (sensor->modelId().startsWith(QLatin1String("RGBgenie ZB-5")) || // RGBgenie remote control
             sensor->manufacturer() == QLatin1String("_TZ3000_xabckq1v") || // 4 Gang Tuya ZigBee Wireless 12 Scene Switch
             sensor->modelId().startsWith(QLatin1String("ZBT-DIMController-D0800"))) // Mueller-Licht tint dimmer
    {
        clusters.push_back(ONOFF_CLUSTER_ID);
        clusters.push_back(LEVEL_CLUSTER_ID);
        clusters.push_back(SCENE_CLUSTER_ID);
        srcEndpoints.push_back(sensor->fingerPrint().endpoint);
    }
```

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916

romanho, 2021-06-09:
> Perhaps its just not supported

At least it has been advertised as being able to set 3x4 scenes, i.e. 3 different things per button.

> Yes can try to bind other cluster than onoff and levelcontrol (rest the scene cluster), to check if you have more return in log, but
> on the code It already done.

I can see this code, yep. A new button definition for SCENE didn't change anything.

Then I've experimented a bit in deCONZ GUI, selected the "scene" cluster and tried to add a scene response. But I cannot select anything in the "Status" dropdown, and any combination of values just triggers

  `Command info, failed to get parameter for 0x0000, invalid user input`

I could at least define a group in the device; if defining again it says something like "already defined", but it seems not to store the group name. Ok, not important... 

Do you know how the "Status" could be filled with something meaningful?
Or is that a wrong track anyway?

Smanar, 2021-06-10:
Ha, we are not speaking about same thing, was just saying to bind the scene cluster, but you are right, the device probably don't use scene feature because they are not set.

Never used them on my side, but I have that as information

> In most cases scenes are associated with a particular group ID. Scenes MAY also exist without a group, in
> which case the value 0x0000 replaces the group ID. Note that extra care is required in these cases to avoid a
> scene ID collision, and that commands related to scenes without a group MAY only be unicast, i.e., they
> MAY not be multicast or broadcast.

The "status" dropbox is the answer when you use the request "add scene", the first 3 lines are to set the request, the 3 last are for the answer.

But I have relay no clues what we need to to with that ....

romanho, 2021-06-11:
I hope I understand what you mean... And I think I found out the following:

My device only has a grey Scene cluster, and you said somewhere above that blue is for sending/output, grey is for receiving/input from the device. No blue Scene cluster, no output :-/

In the grey one, the frame border title really reads "Add scene response", so it makes sense that this is the answer ... but where's the command? I search all clusters up and down -- nothing. I also defined a group, hoping that this might unhide something, but also no success.

Maybe some special magic by the original Tuya bridge or app is needed before those scenes can be defined ... I give up on this.

Just a last remark: In button_maps.json, the MOVE and STOP events should be mapped to correctly corresponding names, I'd say. Currently it's B2H/B2R/B3L/B3R but the buttons are 3 and 4, not 2 and 3.

-        [1, "0x02", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "B2H"],
-        [1, "0x02", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2R"],
-        [1, "0x02", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "B3L"],
-        [1, "0x02", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3R"]
+       [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "B3H"],
+       [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3R"],
+       [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "B4L"],
+       [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4R"]

Smanar, 2021-06-12:
For remote, we are using output cluster (the grey one).
But if you realy nothing on deconz logs, it mean we forget to configure something on zigbee side, can't do more with API.

For the modification, it s as you want, can make the PR yourself if you want ?

romanho, 2021-06-13:
> mean we forget to configure something on zigbee side, can't do more with API

You mean one would have to send some magic Zigbee packets to enable something? Could you give me a pointer where/how to do that?
Maybe I'll get back to this sometimes when I'm bored ;-)
 
> For the modification, it s as you want, can make the PR yourself if you want ?

Done, #5004 

And many thanks for all your patience and support! :-)

Smanar, 2021-06-13:
Np ^^.

>You mean one would have to send some magic Zigbee packets to enable something? Could you give me a pointer where/how to do that?

That's the problem, no information yet on other project about this device, to be sure it s possible we need at least someone that have alrady do that with the original gateway, or better that have sniffed it, to have the "magic receipe"

closed 2021-07-13 by github-actions[bot]

phhe, 2021-08-02:
Running 2.12.03, I was able to stop updates to all lights by clearing light entries for group 65520.

```
❯ curl -X 'PUT' 'http://<HOST_IP>:40850/api/<API_KEY>/groups/65520' -H "Content-Type: application/json" -d '{ "lights": [] }'
[{"success":{"/groups/65520/lights":[]}}]%
```

I'm now getting `deconz_events` for single presses (neither hold nor double click seems to do anything) in HA:
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "switch_moes_1",
        "unique_id": "XXX,
        "event": 1002,
        "device_id": "YYY"
    },
    "origin": "LOCAL",
    "time_fired": "2021-08-02T00:09:09.258572+00:00",
    "context": {
        "id": "ZZZ",
        "parent_id": null,
        "user_id": null
    }
}
```

For the record: I was not able to delete the group and 'Remove all groups' didn't work for me either.

Smanar, 2021-08-04:
@igorrotxategi 
I can't add your device finally, you have a typo in the manufacture name (one more letter)
and your capture are unusable, can I can't read the manufacture name

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5172

igorrotxategi, 2021-08-04:
> @igorrotxategi
> I can't add your device finally, you have a typo in the manufacture name (one more letter)
> and your capture are unusable, I can't read the manufacture name
> 
> #5172

Hi @Smanar 
Im sorry for the cuality of the screenshots i have attached at the issue. 
I have entered to home assistant on my tv, to have a image with a better quality. 
I hope this will be better
[image]

Thank you very much

Smanar, 2021-08-05:
Ok, so there is relay nothing to do ^^
It s exaclty the same than on this issue, you just need a deconz version >= 2.12

igorrotxategi, 2021-08-06:
> Ok, so there is realy nothing to do ^^
> It s exaclty the same than on this issue, you just need a deconz version >= 2.12 to use it.

I am running  v2.11.5 on hassio. I don´t know how to upgrade the version. Can you help me, please?

Smanar, 2021-08-06:
Nope, sorry, I realy don't know HA.

hgross, 2021-08-06:
Pleasse reopen, double press and long press is still not working.

Mimiix, 2021-08-06:
@hgross On version 2.12?

hgross, 2021-08-06:
Yes. Should it work? I am only able to receive the 1002, 2002, 3002 and 4002 events. From the conversation above I understood that long press and double press is not supported at the moment.

[image]

Mimiix, 2021-08-06:
@smanar should be able to answer that

Smanar, 2021-08-06:
Deconz not receive request when making double or long press.
You are sure they are working natively ?

hgross, 2021-08-07:
> 
> 
> Deconz not receive request when making double or long press.
> You are sure they are working natively ?

Yes, ordered it through Amazon. It is marketed as[ 4 buttons with tap, long-tap, double-tap](https://www.amazon.de/gp/product/B08J3TMGJH/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1). Even some reviews mention it working with other ZigBee gateways (12-way).

Smanar, 2021-08-07:
>Even some reviews mention it working with other ZigBee gateways

Do you have some link ? I m searching for "_TZ3000_xabckq1v" on other zigbee project but I don't found something usefull.

hgross, 2021-08-07:
TS004F is the correct one. It is also the name phoscon came up with after pairing (name). The z2m mapping of events seems weird. It is a tuya device meant to be configured with the Tuya gateway and a the Smartlife app.

phhe, 2021-08-07:
Not sure if it helps, 2.12.03, I just did a quick test with the remote, this is what I get in the logs:

### Single Click -> shows up in HA as 3002
```
11:55:32:073 [INFO] - No button handler for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 5
11:55:32:073 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:32:323 [INFO] - No button map for: SML001, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0402, command: 0x0A, payload: 0000293D08, zclSeq: 156
11:55:32:323 ZCL attribute report 0xZZZZZZZZZZZZZZZ for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000
11:55:32:490 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 6
11:55:32:493 rule event /sensors/64/state/lastupdated: 0 -> 0
11:55:33:074 [INFO] - No button handler for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 5
11:55:33:075 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:33:778 Discard duplicated zcl.cmd: 0x0A, cluster: 0x0001 with zcl.seq: 5 for _TZ3000_xabckq1v / TS004F
11:55:33:778 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
```

### Double Click
doesn't show in HA
```
11:55:37:426 [INFO] - No button handler for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 7
11:55:37:426 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:38:131 Discard duplicated zcl.cmd: 0x0A, cluster: 0x0001 with zcl.seq: 7 for _TZ3000_xabckq1v / TS004F
11:55:38:132 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:38:821 Discard duplicated zcl.cmd: 0x0A, cluster: 0x0001 with zcl.seq: 7 for _TZ3000_xabckq1v / TS004F
11:55:38:821 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:40:214 [INFO] - No button handler for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 8
11:55:40:215 ZCL attribute report 0xXXXXXXXXXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
11:55:40:577 ZCL attribute report 0xVVVVVVVVVVVVVV for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000
```

### Hold until LED turns off
shows up in HA as 3001 when the led turns off, followed by 3002 when the button is released
```
11:56:15:737 [INFO] - Button 3001 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3H, payload: 0033, zclSeq: 9
11:56:15:741 rule event /sensors/64/state/lastupdated: 0 -> 0
11:56:16:531 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3R, payload: None, zclSeq: 10
11:56:16:535 rule event /sensors/64/state/lastupdated: 0 -> 0
11:56:18:388 0xYYYYYYYYYYYYYYY error APSDE-DATA.confirm: 0xE9 on task
11:56:19:075 Device TTL 732 s flags: 0x7
```

hgross, 2021-08-07:
@phhe you get more events than I do on the same version. I don't receive the 3001 event. I am not using HA. I am working with a [NodeRED integration](https://github.com/deconz-community/node-red-contrib-deconz) to the rest api.

Smanar, 2021-08-08:
You have the event list here
```
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1"],
                [1, "0x01", "ONOFF", "OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "B3H"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3R"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "B4L"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4R"]
            ]
```

>11:55:40:214 [INFO] - No button handler for: TS004F, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq:

This one is not used as event, because the cluster 0x0001 is a battery event, you have it for all event on the devices, not usable.

phhe, 2021-08-08:
Thanks @Smanar for looking into this. So we simply don't get anything on double click that is what I thought.

That means that mapping for short press/release makes sense: `B3` -> 3002 (`S_BUTTON_ACTION_SHORT_RELEASED`)
```
11:55:32:490 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3, payload: 00330A00, zclSeq: 6
```

We also get the the right mapping for hold: `B3H` -> 3001 (`S_BUTTON_ACTION_HOLD`)
```
11:56:15:737 [INFO] - Button 3001 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3H, payload: 0033, zclSeq: 9
```

But then a wrong mapping (at least I'd say so) `B3R` -> 3002 which is the same code we got for `B3`, and also called `S_BUTTON_ACTION_SHORT_RELEASED` in the map.
```
11:56:16:531 [INFO] - Button 3002 - TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: B3R, payload: None, zclSeq: 10
```

Could we remap `B3R` to 3003 (which is probably `BUTTON_ACTION_HOLD_RELEASED` or something along those lines) ?

Smanar, 2021-08-09:
Ha yes, right, there is some problem in the JSON. And haven't finf something about that in the PR queue list.

You can correct them and submit a PR ?

oat1, 2021-08-24:
Hi,
Just bought  TS004F 4-button switch,  installed the latest version of deconz 2.12.06 and problems still appear.

Short click - works just fine on every button
Double press - does not work at all on any button
Long press – does release the “long press event”  but right away it releases “short click event” in logs
So does not work properly also.

Also just like the persons above my 1 and 3 button are some kind combined with one of my Lidl LED strips. A short press of 1 button turns the LED on and a short press of 3 button turns them off. 
In deconz they are not linked at all.

Are there any news when there will be a fix for this?

Smanar, 2021-08-24:
Oups, tottaly forget it.
@phhe , can you make the PR with your corrective ? Or do you want I make it ?

[55 more replies]