dresden-elektronik/deconz-rest-plugin #4815: IKEA TRÅDFRI Fernbedienung rechts / links (<>) Taste ohne Funktion

TobiasH2010, 2021-04-26:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: MSI Cubi 3 Silent mit ConBee II
 - Running method: Debian Headless in Proxmox
 - Firmware version: 26680700
 - deCONZ version: 2.11.01 / 14.3.2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## deCONZ Logs

## Additional context

jan666, 2021-04-26:
Try removing the battery for 10 seconds and re-insert it. I can't remember if I had the same issue but it fixed something related to the <> buttons for me. Just give it a try :)

TobiasH2010, 2021-04-26:
I've tried it several times without effect. The system also will not recognize a long press of the middle button... At the moment it won't even show the firmware of the remote. But I checked before and it was the latest

Mimiix, 2021-04-26:
Hi,

Can you please translate to English? This is a English git.

Verteo, 2021-04-26:
I have the same problem with the Tradfri Remote. The < and > buttons no longer work with 2.11.01, the button presses - neither short or long - show up in the logs. They were still working with 2.11.00.

ebaauw, 2021-04-26:
I can confirm.  The left and right buttons of the STYRBAR (Remote Control N2) don't work either.

Mimiix, 2021-04-26:
@ebaauw please remain from adding labels. We have guidelines for issues and one of them is that they should be in English.

ebaauw, 2021-04-26:
Ok you go fix it

Mimiix, 2021-04-26:
@ebaauw I think it's for the best if we solve the issue you have with me in discord.

@TobiasH2010 my first request is the same: please translate your issue to English.

jazza1234567, 2021-04-27:
have the same issues, I am more than glad to provide any information needed to help out, Just need to be told what I can provide to help team :)

jazza1234567, 2021-04-27:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: NAS
 - Running method: Marthoc Docker container but this links through to my home assistant platform
 - Firmware version: 26660700
 - deCONZ version: 2.11.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2021-04-28:
@jazza1234567 It's a nice gesture, but i'd rather have the opening post from @TobiasH2010 to be in english...

jazza1234567, 2021-04-28:
> @jazza1234567 It's a nice gesture, but i'd rather have the opening post from @TobiasH2010 to be in english...

As much as it is a gesture, its also due to self-interest - because it will be handy for me to have fixed 😂

Mimiix, 2021-04-28:
@jazza1234567 Feel free to open a seperate issue and copy paste your report in there. I'll close this one then.

jazza1234567, 2021-04-28:
> @jazza1234567 Feel free to open a seperate issue and copy paste your report in there. I'll close this one then.

Done 🙂 ill be around for a few hours should you need any troubleshooting on my end

Mimiix, 2021-04-28:
Closing this one, moving on in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4821.

closed 2021-04-28 by Mimiix