dresden-elektronik/deconz-rest-plugin #4819: Wyze Door Lock

junalmeida, 2021-04-27:
## Device

- Product name: Wyze Door Lock
- Manufacturer: Yunding.
- Model identifier: Ford.
- Device type : Lock

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above
[image]

[image]

[image]

[image]

Smanar, 2021-04-27:
Hello, I m on a PR just for doorlock, are you able to test it ? You need a linux machine with a complete OS.

junalmeida, 2021-04-27:
Currently my setup runs on the docker image in a openhabian (openhab image for raspberry).  I don't have a x64 full machine with linux running. If you could add a dev tag to the docker image I can pull it and give try.

Smanar, 2021-04-27:
Docker are not realy "full OS", I m not skilled enought to do it on docker, perhaps the openhab docker is more complete than the HA one, but seriously I prefer avoid.
If you realy don't have other linux machine (even x32), just for tests, I can add the device in blind mode, but without guarantees.

junalmeida, 2021-04-27:
Yeah, I really don't have other linux boxes, only this ARM openhabian image (which is a debian distro that runs the docker I mentioned).

I'm ok to try your blind mode version :)

Smanar, 2021-04-28:
Have added it on the future doorlock PR.

junalmeida, 2021-04-29:
Ah, I forgot to mention, this lock has also a door sensor that  indicates if the door is open or closed. I believe this comes from the FC00 cluster, is there any log messages I can send you to implement this cluster?

[image]

Smanar, 2021-04-30:
I have already added a "door state" but I read it from the cluster 0x0101. IDK if it s the same ...

junalmeida, 2021-04-30:
hmmm interesting, I will test then when it is available. 
thanks :)

Smanar, 2021-05-01:
Yep it s this one
```
                                    if (ia->numericValue().u8 == 0)
                                    {
                                        str = QLatin1String("open");
                                    }
                                    else if (ia->numericValue().u8 == 1)
                                    {
                                        str = QLatin1String("closed");
                                    }
                                    else if (ia->numericValue().u8 == 2)
                                    {
                                        str = QLatin1String("error jammed");
                                    }
                                    else if (ia->numericValue().u8 == 3)
                                    {
                                        str = QLatin1String("error forced open");
                                    }
                                    else if (ia->numericValue().u8 == 4)
                                    {
                                        str = QLatin1String("error unspecified");
                                    }
                                    else
                                    {
                                        str = QLatin1String("undefined");
                                    }
```

junalmeida, 2021-05-03:
wow that's perfect :D

junalmeida, 2021-05-04:
Just checking in to ask if this is supposed to be in v2.11.2-beta (Apr 30th) or in the next release. Thank you.

Smanar, 2021-05-05:
The PR is not submitted at all yet, we are finishing the pin management first https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3750

closed 2021-06-03 by github-actions[bot]

villasenor, 2022-06-09:
@Smanar any updates on this? What is the current state of Wyze Lock support in deconz?

Smanar, 2022-06-09:
Hello, if you can try this DDF

{
```
  "schema": "devcap1.schema.json",
  "manufacturername": "Yunding",
  "modelid": "Ford",
  "vendor": "Wyze",
  "product": "YWyze Door Lock",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DOOR_LOCK",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/lock",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val == 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lockstate",
          "read": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0101",
            "ep": 11,
            "eval": "if (Attr.val == 1) { Item.val = 'open' } else if (Attr.val == 2) { Item.val = 'closed' } else { Item.val = 'error' }",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0003",
            "cl": "0x0101",
            "ep": 11,
            "eval": "Item.val = Attr.val != 1",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "cl": "0x0101",
      "report": [
        {
          "at": "0x0003",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

I m not sure it's the last version for doorlock, if you can tell me what is working or not ?