dresden-elektronik/deconz-rest-plugin #4821: Ikea TRÅDFRI 5 Button remote control not registering Left and Right button presses

jazza1234567, 2021-04-28:
## Describe the bug

  Since 2.11 my ikea 5 button switches are no longer triggering an action on the left and right buttons, these have previously had no issue. 

## Steps to reproduce the behavior

Attempting to use either the right or left buttons of the remote on deconz 2.11. 

I have reproduced this on both my 5 button controls

## Expected behavior

  Trigger the relevant actions for my lights that the buttons are assigned to.

## Screenshots

  Glad to provide any necessary screenshots later on request. Am not currently home. 

## Environment

 - Host system: NAS
 - Running method: Marthoc Docker container but this links through to my home assistant platform
 - Firmware version: 26660700
 - deCONZ version: 2.11.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

 Happy to provide these when i am back home if needed

## Additional context

Don't think theres anything else. Seems to be impacting multiple people so im guessing its a small change that caused a slip up 🙂 thanks guys

Mimiix, 2021-04-28:
As @ebaauw mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4815#issuecomment-827063476 , it's a confirmed bug.

Mimiix, 2021-05-05:
IT seems that multiple devices are affected. Stryrbar and also the Sengled e1e-g7f switches.

manup, 2021-05-05:
I'll make a test with the IKEA 5 button remote and check if something in the button handler code from the last release causes this.

>Stryrbar and also the Sengled e1e-g7f switches.

These I don't have for testing, but perhaps it's the same cause/bug.

jazza1234567, 2021-05-05:
Im glad to do whatever i can on my end to help test - have both the ikea 5
button and the sengled remote - that said, if its anything too technical
ill need to be walked through it 😂

manup, 2021-05-05:
I've tested the IEKA 5 button remote with latest version v2.11.2, here all buttons do work. In the test I've configured dim up/down actions for the side buttons via Phoscon App Switch Editor and this works as expected as well.

Since you're using v2.11.1, can you please check if the problem still exists with v2.11.2 in your setup?

manup, 2021-05-05:
Hmm odd, even with v2.11.1 my remote works normally :thinking: 

The remote firmware version is 1.2.214 with date code 20170302.

jazza1234567, 2021-05-05:
Ill have a look tomorrow and let you know (late night here, need my beauty
sleep)

Verteo, 2021-05-05:
It's not working for me on 2.11.2. My remote has the firmware 2.3.014 with date code 20190708. 
I tried the phoscon app and hue essentials neither work. There are no button presses for < > showing up in the deconz log.

jazza1234567, 2021-05-06:
> Hmm odd, even with v2.11.1 my remote works normally 🤔
> 
> The remote firmware version is 1.2.214 with date code 20170302.

My firmware, like verteo above, is 2.3.014 and yep no < > actions showing up

truidix, 2021-05-06:
Same here since v2.11.1 (curetly i'm on 2.11.2) no actions with <>, with all 5 of my ikea trafdi remotes.

m-s-b, 2021-05-06:
I have two Tradfri remotes, both 2.3.014. With latest beta left and right buttons without any functions. Downgrade to latest stable 2.10.04 the first remote works with left an right button only short button prressed. Second remote has short and long pressed button for left and right

manup, 2021-05-07:
Thanks everyone for checking, we found the issue now fix should be included in v2.11.3.

jazza1234567, 2021-05-07:
Awesome stuff!

Would this issue (and fix) have any link to other remotes like the sengled
e1e-g7f? Just need to work out if i need to get around to testing some
things /posting a seperate issue about it

Cheers for the quick fix, you guys rock.

EpicLPer, 2021-05-08:
Was going crazy and thought I misconfigured something in Home Assistant, glad other people also have this issue. Will wait for an update :)

SwoopX, 2021-05-09:
> 
> Awesome stuff! Would this issue (and fix) have any link to other remotes like the sengled e1e-g7f? Just need to work out if i need to get around to testing some things /posting a seperate issue about it Cheers for the quick fix, you guys rock.
> […](#)

Yes. Same story, same fix ;)

jazza1234567, 2021-05-09:
> > Awesome stuff! Would this issue (and fix) have any link to other remotes like the sengled e1e-g7f? Just need to work out if i need to get around to testing some things /posting a seperate issue about it Cheers for the quick fix, you guys rock.
> > […](#)
> 
> Yes. Same story, same fix ;)

Legends, thanks to the entire team. Will be good to load up the new revision when its out! The sengled remotes were part of a work in progress for my home so looking forward to perfecting their uses!

closed 2021-05-09 by manup