dresden-elektronik/deconz-rest-plugin #4822: LIDL Livarno Lux G10 Lamps

Guardempire, 2021-04-28:
## Describe the bug

## Steps to reproduce the behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Openhabian)
 - Firmware version: (26680700)
 - deCONZ version: (2.11.01 / 26.4.2021)
 - Device: (ConBee III)
 - Do you use an USB extension cable: (yes )

Mimiix, 2021-04-28:
I am unsure if this is something in deConz. 

Can you please check if you can change color using the rest api https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/?  

Is the (Color)light working in Openhab?

Guardempire, 2021-04-28:
In openhab it was working, when the lamp was shown in the phoscon app with colors... But i know what you mean, its a phoscon/deconz bug, not a bug in the api, right?!

Mimiix, 2021-04-28:
The API is deCONZ. Phoscon is, like OpenHAB, a user of the API. 

If the functionality works with the rest api directly (as per my URL), the issue is within Phoscon/Openhab.

If it doesn't, the issue is within deCONZ.

Guardempire, 2021-04-28:
Ok, i´ll try it out and write when i have more infos

LeoeLeoeL, 2021-04-28:
I have those lamps.
Phoscon reports:
Vendor: LIDL Livarno Lux
Model: HG06106A
I confirm Phoscon doesn't show the RGB sliders.

Guardempire, 2021-04-28:
So i tried some things:

The Phoscon Version where the RGB part works is 2.09.03. When the lamp is already connected the slider is also in the newest version 2.11.01 BUT when i delete the lamp and connect it like a new lamp the RGB slider is not there. Only the colortemperature slider. 

I made some things with the api from your link. First time but i could manage to get some numbers... what i am not able to is to change the "colormode" from "ct" <- i think it stands for colortemperature to xy <-- that i found when RGB was "active"

perhabs someone got the "code" to change the colormode so i can try if its the working with this version

Thats the numbers when i am on Version 2.11.01
{
  "ctmax": 65279,
  "ctmin": 0,
  "etag": "555eb08e2b4cd7c68ae44e84fcf2234a",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2021-04-28T15:07Z",
  "manufacturername": "LIDL Livarno Lux",
  "modelid": null,
  "name": "Extended color light 4",
  "state": {
    "alert": "none",
    "bri": 255,
    "colormode": "ct",
    "ct": 262,
    "on": true,
    "reachable": true,
    "xy": [
      0.7,
      0.3
    ]
  },
  "swversion": null,
  "type": "Extended color light",
  "uniqueid": "58:8e:81:ff:fe:d2:6f:c6-01"
}

acid115, 2021-05-01:
The same problem i have with the LIDL LED stripe HG06104A
No colorchange with phoscon oder amazon alexa (but i think this is an other problem, because the led stripe does not support hue - maybe it is a config issue in deconz plugin)
But colorchange through api (set x and y) is possible, and also through the old web interface Open Wireless Light Control (2016).

Guardempire, 2021-05-03:
I changed the Firmware today to 266b0700  and the slider is back. It seams like it is working now.

acid115, 2021-05-05:
Just upgraded firmware to 26390500 (latest for conbee), but no change. Maybe they have changed something in latest firmware for conbee2... But i think there is a problem in phoscon....

Mimiix, 2021-05-05:
The firmware of conbee/raspbee  doesn't change anything on how lights behave. A reboot or incorrect cluster reading is probably the reason why it changed.

I've asked Manup to reply.

manup, 2021-05-05:
I think the problem is that in the REST API output the light says it is an "Extended color light" but the hue/saturation and colorcapabilities values are somehow missing.

I can't tell yet if the plugin is not querying them properly or if the lights actually don't have these.

Can you please provide screenshots of the *Color Control*  cluster after reading the attributes.
Description on how to query the clusters can be found in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

We need to look at attribute 0x400A:

[image]

And Hue / Saturation:

[image]

Note these are examples of the IKEA CWS light not related to LIDL Livarno.

Guardempire, 2021-05-05:
> The firmware of conbee/raspbee doesn't change anything on how lights behave. A reboot or incorrect cluster reading is probably the reason why it changed.
> 
> I've asked Manup to reply.

I have tried this more than once and it makes a difference.

manup, 2021-05-05:
I found a LIDL Livanro RGB E27 HG06106C in my testing box which I think works the same as the GU10.
The light doesn't expose `colorcapabilities`, we should set this to a fixed value for `xy`.

There are hue and saturation attributes, but commands to them bring an error 0xC3 or will be just ignored.

So all we have is `xy`values, the Phoscon App can work with these but is likely confused of the missing hue and saturation. I'll check how to fix this.

manup, 2021-05-05:
Fixed now in the Phoscon App it should be available tomorrow in http://phoscon.de/pwabeta in the local version in v2.11.3.

The color and saturation control for these lights is a bit clunky but that seems to be a limitation of the lights.

LeoeLeoeL, 2021-05-05:
Is the brightess issue fixed too ?

LeoeLeoeL, 2021-05-05:
Maybe interesting.
The brightness issue is present to Hue Bridge too.

manup, 2021-05-05:
What's the brightness issue? At least for the RGB E27 HG06106C light changing brightness works, but it looks like similar to the colors there are rather coarse offsets between brightness steps.

LeoeLeoeL, 2021-05-05:
The fastest way to get this issue is:
Turn on the light
Put at minimum brightness
Turn off the light
Switch off by physical wall switch
Wait for some minutes
Switch on the wall switch
Turn on the light
Now you have a lamp with the brightness at minimum but the cluster 0008 (Level Control) 0x0000 (Current Level) reports 255.

manup, 2021-05-07:
> The fastest way to get this issue is:
> Turn on the light
> Put at minimum brightness
> Turn off the light
> Switch off by physical wall switch
> Wait for some minutes
> Switch on the wall switch
> Now you have a lamp with the brightness at minimum but the cluster 0008 (Level Control) 0x0000 (Current Level) reports 255.

Unfortunately this appears to be a bug in the lights firmware which needs to be fixed by LIDL. There is not much we can do as deCONZ and the REST API can only rely on the state which the lights report.

Can anybody confirm that the GU10 Livarno Lux color sliders work again with http://phoscon.de/pwabeta ?

Guardempire, 2021-05-11:
Some for me! The already connected GU10 was without color slider. I deleted it and paired it again, now the slider is there. I will use it some days an report if there is any issue.

Guardempire, 2021-05-12:
Ok, new findings. 
After a while, the RGB lamp is no longer switched by the Openhab rule  and just stays on, while the white ones (modelHG06492A) work normally. When I go into the Phoscon web interface I can switch the lamp BUT the colour slider has disappeared again. After switching once via Phoscon, Openhab also works again, but the colour slider stays disappeared.

acid115, 2021-05-13:
Same here with the led stripe.
After deletion and repairing the led, the slider for color change appears and can control the color. In phoscon-beta AND in phoscon.
But, after restarting the gateway, the slider disappears again. If i look on the json, the colorcapabilities also disappears...maybe it is not persistent?

`{"ctmax":65279,"ctmin":0,"etag":"8b65efc89350ae844d0501506f19df25","hascolor":true,"lastannounced":null,"lastseen":"2021-05-13T07:38Z","manufacturername":"LIDL Livarno Lux","modelid":"HG06104A","name":"LEDBand","state":{"alert":"none","bri":59,"colormode":"ct","ct":464,"on":true,"reachable":true,"xy":[0.4557,0.5175]},"swversion":null,"type":"Extended color light","uniqueid":"84:2e:14:ff:fe:2c:44:f1-01"}`

Mimiix, 2021-05-13:
To be fair: I never had this issue with my Ledstrip. Not on current or older versions. I use it with HA and it seems fine. Not sure if this could be due to another version or so.

acid115, 2021-05-13:
Yes, because HA just set x and y value. 
I have thirdparty app on mY smartphone, with that i can change color and ct. But it is not possible with phoscon or with alexa...

LeoeLeoeL, 2021-05-13:
You are lucky.
In Alexa I switch on a lamp but it stays off and a plug goes on !!!!!!!

acid115, 2021-05-13:
Then alexa talks to another deviceID...
Or you have to gateway with same ID range...
Or you have another problem.
I have to open a new issue, but no time at the moment... Since  a couple of weeks i have the issue, that when a device got powerloss (plug it off or sth.) Deconz forget the device after 5-10min. After rescan the device appears again, but the name is removed And the device has a new ID. So it is possible, that alexa has another id for a light and a plug has got the id from light....

LeoeLeoeL, 2021-05-13:
the strange thing is it's always a plug (never another lamp) to take the command of the lamp i wanted to switch on.

Guardempire, 2021-05-26:
Since it doesn´t work properly i hooked up my hue bridge and connected the LIDL RGB lamp to it. Since then no proplems with it. So it looks like a deconz problem for me?! I didn't really want to use the HUE Bridge as I only want to use one gateway, so while that is a possibility at the moment, it shouldn't be the solution.

closed 2021-06-25 by github-actions[bot]

oywino, 2022-08-30:
> 
@LeoeLeoeL 
With the two LED lamps that I have; **LIDL GU10 Model HG07878A and model HG06492A** this did not come out as you said. Cluster 0008 and attribute 0x0000 Current level show the number 8 when on minimum level and the value 254 on maximum light level.  Controlling the color temperature seems impossible. It randomly changes in all directions when trying to change it.  These bulbs offer no color control.