dresden-elektronik/deconz-rest-plugin #4824: 2.11.00 Raspbee II can not find devices

GenuineJakob, 2021-04-28:
## Describe the bug

## Steps to reproduce the behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4B 4GB)
 - Running method: (Raspbian)
 - Firmware version: (26610700)
 - deCONZ version: (2.11.00)
 - Device: (RaspBee II)
 - Is there any other USB or serial devices connected to the host system? (No)

Mimiix, 2021-04-28:
Hi,

Can you please add some logs?

GenuineJakob, 2021-04-28:
```
16:59:56:292 dev /dev/ttyS0 (/dev/ttyS0)
16:59:56:297 GW firmware version: 0x26610700
17:00:06:291 dev /dev/ttyS0 (/dev/ttyS0)
17:00:06:293 GW firmware version: 0x26610700
17:00:16:324 dev /dev/ttyS0 (/dev/ttyS0)
17:00:16:326 GW firmware version: 0x26610700
17:00:26:320 dev /dev/ttyS0 (/dev/ttyS0)
17:00:26:323 GW firmware version: 0x26610700
17:00:29:751 enqueue event config/localtime for /config/
17:00:30:271 Master: read param with arg 0x19
17:00:30:293 Current channel 20
17:00:30:303 CTRL got nwk update id 1
17:00:30:313 Device TTL 3386 s flags: 0x4
17:00:30:751 enqueue event config/localtime for /config/
17:00:31:751 enqueue event config/localtime for /config/
17:00:32:751 enqueue event config/localtime for /config/
17:00:33:751 enqueue event config/localtime for /config/
17:00:34:251 sql exec SELECT * FROM config2 WHERE key='wifitype'
17:00:34:252 Load config wifitype: accesspoint from db.
17:00:34:252 sql exec SELECT * FROM config2 WHERE key='wifi'
17:00:34:253 Load config wifi: configured from db.
17:00:34:751 enqueue event config/localtime for /config/
17:00:35:751 enqueue event config/localtime for /config/
17:00:36:291 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:00:36:291 COM: /dev/ttyS0 :  (0x0000/0x0000)
17:00:36:292 dev /dev/ttyS0 (/dev/ttyS0)
17:00:36:293 GW firmware version: 0x26610700
17:00:36:751 enqueue event config/localtime for /config/
17:00:37:751 enqueue event config/localtime for /config/
17:00:38:751 enqueue event config/localtime for /config/
```
INFO + INFO_L2
APS, ZDP, ZCL are empty

Mimiix, 2021-04-28:
Can you run logs while trying to add a device?

GenuineJakob, 2021-04-28:
```
17:07:14:293 dev /dev/ttyS0 (/dev/ttyS0)
17:07:14:295 GW firmware version: 0x26610700
17:07:18:751 APS-DATA.request id: 4, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
17:07:18:752 send permit join, duration: 59
17:07:18:753 APS-DATA.request id: 5, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 1 len: 6 tx.options 0x00
17:07:18:754 send GP proxy commissioning mode
17:07:24:292 dev /dev/ttyS0 (/dev/ttyS0)
17:07:24:296 GW firmware version: 0x26610700
17:07:34:330 dev /dev/ttyS0 (/dev/ttyS0)
17:07:34:333 GW firmware version: 0x26610700
17:07:44:313 dev /dev/ttyS0 (/dev/ttyS0)
17:07:44:316 GW firmware version: 0x26610700
17:07:48:293 Current channel 20
17:07:48:303 CTRL got nwk update id 1
17:07:48:314 Device TTL 3483 s flags: 0x4
17:07:54:315 dev /dev/ttyS0 (/dev/ttyS0)
17:07:54:318 GW firmware version: 0x26610700
17:08:04:292 dev /dev/ttyS0 (/dev/ttyS0)
17:08:04:294 GW firmware version: 0x26610700
17:08:14:325 dev /dev/ttyS0 (/dev/ttyS0)
17:08:14:328 GW firmware version: 0x26610700
17:08:17:751 APS-DATA.request id: 6, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 2 len: 3 tx.options 0x00
17:08:17:751 send permit join, duration: 0
17:08:17:754 Search sensors done
17:08:18:023 aps request id: 4 prf: 0x0000 cl: 0x0036 timeout (confirmed: 0) to 0x0000000000000000 (0xFFFC)
17:08:18:024 aps request id: 5 prf: 0xA1E0 cl: 0x0021 timeout (confirmed: 0) to 0x0000000000000000 (0xFFFC)
17:08:18:103 aps request id: 4 failed, erase from queue
17:08:18:103 emit artificial APSDE-DATA.confirm id: 4
17:08:18:183 aps request id: 4 failed, erase from queue
17:08:18:264 aps request id: 5 failed, erase from queue
17:08:18:264 emit artificial APSDE-DATA.confirm id: 5
17:08:18:343 aps request id: 5 failed, erase from queue
17:08:24:332 dev /dev/ttyS0 (/dev/ttyS0)
17:08:24:356 GW firmware version: 0x26610700
```
With INFO, APS, ZDP and ZCLselected

Mimiix, 2021-04-28:
Can you please let it run the full duration of the search?

GenuineJakob, 2021-04-28:
After rebooting the pi I can now see the Raspbee in the right part of the window:
[image]
But it still fails do add devices:
[image]

Thanks for helping me help!

```
17:19:35:295 Mgmt_Lqi_req zdpSeq: 63 to 0x00212EFFFF061CE6 start index 0
17:19:35:296 APS-DATA.request id: 224, addrmode: 0x03, addr: 0x00212effff061ce6, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x04
17:19:35:363 APS-DATA.confirm id: 224, status: 0x00 SUCCESS
17:19:35:364 APS-DATA.confirm request id: 224 -> confirmed, timeout 8670640
17:19:35:384 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
17:19:35:385 APS-DATA.indication request id: 224 -> finished
17:19:35:386 APS-DATA.request id: 224 erase from queue
17:19:35:387 ZDP status = 0x00 -> SUCCESS
17:19:35:388 ZDP Mgmt_Lqi_rsp zdpSeq: 63 from 0x00212EFFFF061CE6 total: 0, startIndex: 0, listCount: 0
```
[log_full-run.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6393645/log_full-run.txt)

GenuineJakob, 2021-04-28:
I am so sorry for wasting your time.
Reading [Update-deCONZ-manually](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) and updating my firmware did the trick.

closed 2021-04-28 by GenuineJakob