dresden-elektronik/deconz-rest-plugin #4826: Moes 4-gang switch

marithpl, 2021-04-29:
## Device

- Product name: Moes 4-gang Switch
- Manufacturer: Moes _TZ3000_vp6clf9d
- Model identifier: TS0044
- Device type : Switch
- Link to device: https://www.aliexpress.com/i/1005001504861368.html?spm=a2g17.12057483.1000001.8.22893bdeP2K6OL

Device paired with phoscon with success but in HA it show as a normal 4 plug switch.
So I'm not able to make any automations in Home Assistant.
In Phoscon it also not showing on group linking with switch.

I have other Aqara Wall Switch and not present toggles in HA and is visible in Phoscon group linking.

[image]
[image]

## Screenshots
[image]

### Basic
[image]
### Identify

### Power Configuration
[image]

### On/Off
[image]

### Other clusters that are not mentioned above

Mimiix, 2021-04-29:
Hi! 

there's already a issue on this switch https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3611

I think it's the same (at least the model ID is).

Mimiix, 2021-04-29:
Is it the same? if so, i suggest we close this issue and proceed there :)

marithpl, 2021-04-29:
Seems to be the same

closed 2021-04-29 by marithpl