dresden-elektronik/deconz-rest-plugin #4829: Do all the routers forward to the coordinator correctly ?

LeoeLeoeL, 2021-05-01:
## Describe the question or issue you are having
 I have 2 Vimar Covering Windows.
They are at the same floor.
One works perfectly; For the other one it's almost impossible to read the clusters.
When I open both windows, the first one shows me the % of opening while the second says it's still closed. After hours, it tells me the %.
They have different links in the mesh of course.

## Screenshots

## Environment
 v2.11.1-beta
Qt: 5.7.1
GCC: 6.3.0 (C++ 201402)
Commit: 51086e
Firmware: 0x26680700

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26680700)
 - deCONZ version: (2.11.1-beta)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC ups

## deCONZ Logs

## Additional context
For info, the working Vimar has a direct link to Conbee.
Could be a problem Vimar is Zigbee HA and often the strongest link is a ZLL device?
Is there a way to force the device to connect to Conbee directly ?

Smanar, 2021-05-01:
You don't have the mesh picture from the GUI ?

LeoeLeoeL, 2021-05-01:
[image]

Smanar, 2021-05-01:
Effectively, realy strange, you have realy good mesh, and the device "Finestra centrale" seem have lower quality connexion than the problematic one ....

>Is there a way to force the device to connect to Conbee directly ?

No, but there is perhaps something to help you, never tried (and I realy don't understand what is it) but take a look at that https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Source-Routing.

Perhaps on logs you can find a device with connexion or firmware issue ? (With error and error_l2)

LeoeLeoeL, 2021-05-01:
Same routers are offline because they are under physical switch.
I tried source routing but it didn't solve the problem.A strange thing I saw in Phoscon are the name Deconz gave to the device:
Finestra Centrale was "cover.window_covering_device_26"
Finestra Dispensa was "cover.window_covering_device_26_2"
As the same device has been re-paired.

One question. What are Max buffer size and Max incoming/outcoming Transfer Size ? Vimar devices have higher values (107/128) than the others. Osram plugs have (64/0).

The only errors are:
19:50:41:680 command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_INDICATION2, seq: 123
19:55:42:720 command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_INDICATION2, seq: 168
20:00:43:716 command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_INDICATION2, seq: 239
20:05:41:288 command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_INDICATION2, seq: 16

LeoeLeoeL, 2021-05-01:
I forgot to write that before the 2nd covering window device I had a Vimar switch. It had the same problem. I believed it was faulty so that I gave back and bought a 2nd covering window switch.
It seems Deconz doesn't like more than a Vimar device. :-)

Smanar, 2021-05-01:
>Some routers are offline because they are under physical switch.

Can be a possible issue too, the mesh is dynamic, but it s not something recommanded.

For the "magic numbers" https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/LQI-explained

For the error message, I will ask to other devs, before saying craps.

>It seems Deconz doesn't like more than a Vimar device. :-)

Yeah, but I don't see why a device will not work like the second one ...

LeoeLeoeL, 2021-05-01:
The values I wrote are in Node Info window and are "steady".
Max Buffer Size for:
Vimar 108
Osram 64
Lidl 82
Sonoff 82
Conbee 71

Smanar, 2021-05-02:
>I know but I have to choose between a good mesh and my wife.

Lol, the WAF problem.

Values, like color, are just an indication, and they not look bad, about the error log, manup have answered

> I think this is an timing /race condition  issue in core when the indication is received at the same time it is queried.
> Have this on the todo list to check, shouldn't cause any harm other then the misleading debug message

So bad luck, not realy usefull too to explain the issue.

Same values for reporting value ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2196#issuecomment-695843587

LeoeLeoeL, 2021-05-02:
Do you know what MAC Capabilities means in Node Info?
Beacause I found something strange.
The coordinator has 0x0f.
All the end devices (sensors and Tradfri ON/OFF switches) have 0x80.
All the routers have 0x8e except Vimar Finestra Centrale that has 0x0e.

This week, if I have time, I disconnect Finestra Centrale in order to realize if Deconz doesn't like 2 Vimar devices.

SwoopX, 2021-05-03:
Oh, if you got 0x0E in mac capabilities, than this is most likely the reason for you trouble. In those cases, deconz considers the node descriptor to be invalid, which impacts cluster bindings and attribute reporting and pretty much explains why you don't get anything. The issue here is that the allocate short address bit is set to 0.

I, personally, consider this a bug in deconz core as the specs do not demand it to be set to 1 in the node descriptor, but it's a must on IEEE 802.15.4 level.

As I see it, a potential fix would require 2 whitelisting entries in `bindings.cpp`. However, the root cause of this should be eliminated once and for all.

Smanar, 2021-05-03:
Ha nice, ths a lot, realy usefull, to put on a post-it.

It s not possible to solve this problem using GUI command ? re-ask the node descriptor up to have good values ?

LeoeLeoeL, 2021-05-03:
I clicked on READ NODE DESCRIPTOR  and now I have 0x8e.

SwoopX, 2021-05-03:
I'd assume now the troublesome device works as expected (again)?

LeoeLeoeL, 2021-05-03:
The device with the wrong MAC Capability was the working one (Finestra Centrale) . Before fixing it, I tried to read something in the other device (Finestra Dispensa) and was OK.
Perhaps because yesterday night I put an IKEA range extender very close to it (70 cm) and now this one links Vimar to Conbee correcly.

Smanar, 2021-05-03:
So the working one was the one with  0x0E, and the good reconized  not working ?

LeoeLeoeL, 2021-05-03:
Affirmative.

LeoeLeoeL, 2021-05-05:
The trick of Ikea range extender worked until I changed 2 Osram plugs (ZLL) with 2 Lidl plugs (HA).
Then I deleted FINESTRA CENTRALE (the working one) in order to verify if Deconz doesn't like more than 1 Vimar. No difference.
Then I deleted FINESTRA DISPENSA too and re-paired both.
It happened:
[image]
The working device paired correctly (even if with another name) but the other one remained as 0x9C98

Smanar, 2021-05-05:
>but the other one remained as 0x9C98

This one is grayed, not connnected, and badly included previously (have only 1 circle on node title)

LeoeLeoeL, 2021-05-05:
At last, to have the device working, I had to:
1) Delete the dimmable actuator iCasa  ICZB-IW11D "Luce Sala Cinema" (is it the guilty ?)
2) Use an old installation in Windows where there were less devices.
3) Pair the Vimar
4) Come back to RPi3.
Now it works OK hoping there will not be another device with that problem.

I think zigbee is more a protocol than a standard.
Some devices fight with others.
But the main limitation is each brand claims its bridge to update firmwares.

Smanar, 2021-05-05:
> I think zigbee is more a protocol than a standard.
> Some devices fight with others.

Big story ...

>But the main limitation is each brand claims its bridge to update firmwares.

No you can update all firmwares with deconz, ...., if you have the file.

LeoeLeoeL, 2021-05-05:
... But almost nobody publishes them.
It's a lack in the zigbee alliance.

LeoeLeoeL, 2021-05-05:
Damn, the issue again!

LeoeLeoeL, 2021-05-08:
I made many other tests.
I also disconnected by wire the working vimar (Finestra Centrale) keeping in the mesh only the other one (Finestra Dispensa).
At first it worked but, when I add other devices, after some hours the problem reappears.
Only one attempt remains; Deleting all the mesh and starting again.
If I use RESET GATEWAY in Phoscon does Deconz forget everything?
Do devices remember "something" from the previous link?

Smanar, 2021-05-08:
> If I use RESET GATEWAY in Phoscon does Deconz forget everything?

Useless for me. You can see the information if you have a sql editor.

> Do devices remember "something" from the previous link?

Not sure to understand what you are talking, perhaps about neighbour table ? but yes, some data are memorised.

>At first it worked but, when I add other devices, after some hours the problem reappears.

But it s realy kike the network can support only 1 vimar ...

LeoeLeoeL, 2021-05-08:
And I should buy other 3 vimars for my windows!!!!!!

LeoeLeoeL, 2021-05-11:
Unbelievable.
It worked for 24 hours flawlessy and now it doesn't read again.

Is it normal this number in the "read neighbor table"?
[image]

LeoeLeoeL, 2021-05-14:
At last, it seems to work properly only when linked straightly to the coordinator.

Smanar, 2021-05-14:
I know some end device not compatible with some routers, but from my memory, no other situation with routers not compatible with other routers ...

LeoeLeoeL, 2021-05-14:
I think I buy an extension cord USB to put the Conbee II in a better place for Vimar(s).
But it would be a better place for everyone.
How many links can Conbee II manage directly?

Smanar, 2021-05-14:
>How many links can Conbee II manage directly?

32 if I m right.

LeoeLeoeL, 2021-05-14:
> I know some end device not compatible with some routers, but from my memory, no other situation with routers not compatible with other routers ...

I just wrote to Vimar about the inability to communicate the position of the window if not connected directly to the coordinator.
I hope they don't reply Deconz is guilty.  :-)

LeoeLeoeL, 2021-05-25:
Vimar didn't respond yet.
Anyway I observed since 2.11.3 more links than before generally and now also the 2nd Vimar is firmly connected to the coordinator.

LeoeLeoeL, 2021-06-04:
Vimar answered the product is certified Zigbee 3.0 so the problem must be the coordinator.
I'm sure they didn't test anything.

closed 2021-06-04 by LeoeLeoeL