dresden-elektronik/deconz-rest-plugin #4832: IKEA STYRBAR "Remote Control N2" Type E2001

zustellr, 2021-05-02:
## Device

- Product name: IKEA STYRBAR Remote Type E2001
- Manufacturer: IKEA of Sweden
- Model identifier: Remote Control N2
- Device type : Please remove all unrelated device types. 
  - Remote

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

zustellr, 2021-05-02:
Also posted this to https://github.com/dresden-elektronik/phoscon-app-beta/issues/380 two days ago, probably in error. Sorry about that.

SwoopX, 2021-05-09:
This device is already supported, check the release notes.

closed 2021-05-09 by SwoopX