dresden-elektronik/deconz-rest-plugin #4833: Color temperature issue with Singled Zigbee Lights

kkilchrist, 2021-05-02:
## Describe the bug

Singled RGB + Color temp bulbs no longer work properly. RGB colors can be selected just fine, but I am no longer able to activate color temperatures. This previously worked. 

Vendor: singled
Model: E12-N1E
Version: 0x00000015
Link: https://www.amazon.com/Sengled-E12-N1E-Smart-Light-White/dp/B07J6JK2BD

## Steps to reproduce the behavior

## Expected behavior

I should be able to control color temperature of these lights, in addition to RGB.

## Screenshots

Attached some screenshots.

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
Raspberry Pi

 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine): 
Home Assistant DeCONZ add-on 6.8.0

 - Firmware version: (26xxyy00)
26680700

 - deCONZ version: (2.xx.yy)
2.10.04 

 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
ConBee II

 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
Yes

 - Is there any other USB or serial devices connected to the host system? If so: Which?
No

## deCONZ Logs

## Additional context

Attached photos of PhosCON info for bulb, as well as HA state for both a single bulb and a group of these bulbs.
[image]
[image]
[image]

kkilchrist, 2021-05-02:
I should mention: if I manually call the light.turn_on service, e.g., 
```
service: light.turn_on
data:
  kelvin: 3100
target:
  area_id: living_room

```
The lights properly update to warm white

closed 2021-05-31 by github-actions[bot]

akusokuzan, 2021-08-04:
I am also having this problem! Is this issue being worked on or should I just return them? Also manually setting them doesn't work instead they turn a light pink looking color. The only way to reset them I've found is to delete them and re-add them.

kkilchrist, 2021-08-04:
The issue has actually gotten worse for me, as I can no longer manually
call the light turn on command with a color temperature to fix it.

It’s quite frustrating, as they originally did work.

I wish that the developers would at least comment on the issue, or suggest
some places to start, especially as they originally did work, and something
broke. And then it broke worse with a subsequent update.

akusokuzan, 2021-08-04:
Well that's annoying. I bought them on sale and since many bulbs cant be flashed anymore I was hoping this would be a good alternative. Also the design and look are similar to my regular bulbs. I was hoping the issue would be getting better not worse! Ugh well thank you for the update. Seems like no one cares about this issue. I've seen more than one mention of this problem with no response. I guess ill keep looking for a bulb that will work. If you have any recommendations let me know preferably a bulb that look similar to this one.

Smanar, 2021-08-05:
I m not sure to understand.
You can send a RGB color temperature but not a temperature color one ?
Can you show the device JSON ? 
Have you try with phoscon ?
Can you try direclty with the API (I can explain how to do if needed ?)

kkilchrist, 2021-08-05:
Hi @Smanar 

The lights are RGBWW, based on the fact that "white 100%" is significantly dimmer than "warm white 100%" or "cool white 100%" 

For a time, the lights worked fine and Phoscon properly saw these as color temp lights, which allowed me to set color temperature in HomeAssistant. 

At some point, something broke, and the lights were no longer showing up as color temp capable, but if I manually sent them a color temperature with a HomeAssistant script, e.g., 
```
service: light.turn_on
data:
  kelvin: 2100
  white_value: 254
target:
  area_id:
    - living_room
```
They would properly turn on the white and warm white LEDs to produce bright light. 

At some point, however, this stopped working as well, and now the lights only use their RGB bulbs. If I hard reset the bulbs, they automatically go into a bright warm light, so I know the LEDs are not defective. 

I just tried manually setting the color temp through the Deconz API, and this did work! 

So the issue seems to be that Deconz does not properly read the capabilities of the bulbs and report it to HomeAssistant. 

Here's the GET for a bulb (before manually futzing with it via API): 
```
{
"etag": "9b275239c324baa9d4c066072e668be4",
"hascolor": true,
"lastannounced": "2021-06-26T18:58:38Z",
"lastseen": "2021-08-05T17:59Z",
"manufacturername": "sengled",
"modelid": "E12-N1E",
"name": "Living room bulb 2",
"state": {
"alert": "none",
"bri": 1,
"colormode": "xy",
"effect": "none",
"hue": 2838,
"on": true,
"reachable": true,
"sat": 133,
"xy": [
  0.3127,
  0.329
],
},
"swversion": "0x00000015",
"type": "Color dimmable light",
"uniqueid": "b0:ce:18:14:03:52:ba:97-01"
}
```

kkilchrist, 2021-08-05:
Edit: I gave up trying to get the Deconz plugin to work. I deleted the plugin and switched to ZHA (https://www.home-assistant.io/integrations/zha/) which automatically has the correct behavior. 

@akusokuzan

akusokuzan, 2021-08-05:
Yeah thanks for explaining that its exactly the same for me. Great so my only solution is to switch to ZHA. I was hoping I wouldn't have to do that as I might run into a totally different problem. Quick question do you get any effects with the bulbs with ZHA? Would love a color cycle random or something. 

@kkilchrist

kkilchrist, 2021-08-05:
Not sure about color effects! I never really saw a use for the colorloop
color effect

akusokuzan, 2021-08-05:
Oh well if you have a chance to check at some point I would appreciate it. I use it for my kids who love to dance and do party nights with friends so we have a disco light show and color changing bulbs and music going the kids love it lol.

Smanar, 2021-08-06:
```
I just tried manually setting the color temp through the Deconz API, and this did work!
So the issue seems to be that Deconz does not properly read the capabilities of the bulbs and report it to HomeAssistant.
```

I have see lot some issue/code about that, if I m right the better method is just checking the "param" available in the json, use "colormode" is not precise enought.
IDK how o HA but you have used wich one command to change the temperature ?
I don't see the "ct" command in the json.

kkilchrist, 2021-08-06:
To set the color temp through the API I used this command:
````
{
  "on": true,
  "bri": 255,
  "ct": 250,
}
````

Smanar, 2021-08-06:
And you don't have error message ?
You haven't "ct" on the JSON ..... (and I think it's for that HA don't use them too)

akusokuzan, 2021-09-25:
Any update on this. I decided to keep them maybe one day they will fully work. fingers crossed