dresden-elektronik/deconz-rest-plugin #4834: Add Button Map for lumi.switch.b1naus01

rexbron, 2021-05-02:
## Device

- Product name: Xiaomi Aqara Single Rocker In-Wall Switch 
- Manufacturer:  LUMI
- Model identifier: lumi.switch.b1naus01
- Device type : Please remove all unrelated device types. 
  - Light
  - Sensor
  - Switch

## Logs

`12:30:14:587 [INFO] - No button map for: lumi.switch.b1naus01, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 000025E01600000000, zclSeq: 28`

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Alarms
[image]

### Device Temperature
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering
[image]

### Diagnostics

### Other clusters that are not mentioned above
[image]

Smanar, 2021-05-02:
But according to your capture, not sure you need it ?
It s a router, and I think it create only a "light" entry, no ?

rexbron, 2021-05-02:
Perhaps this is a misunderstanding on my part, I saw it in the log and figured that deconz wouldn't emit the log entry unless it needed more information.

In an ideal world, It would be great to be able to use the physical button to also trigger other automations in Home Assistant. Perhaps you already can?

Smanar, 2021-05-02:
IDK, Aqara davice that permit that have so much endpoint.

[image]

1 endpoint can be used for light and another for switch hen they are "decoupled". But your device relay don't look same.
If you can't "decouple" the command from the relay, you can just use the light state to know if the device is on or off, no ?

> 12:30:14:587 [INFO] - No button map for: lumi.switch.b1naus01, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 000025E01600000000, zclSeq: 28

Cluster 0x0702 is the simple metring, for the light consumption, it s probably for that the sensor was ignored.
It s something coded in the API, I don't know the reason but there is something in code to ignore this cluster for this device.

SwoopX, 2021-05-20:
The device would need to expose a ZHASwitch resource for the button presses to emit button events. I'm not aware it is currently whitelisted for creating any sensor resources. If it would, it would generate further such INFO messages for the other clusters which then could be used to define a button map.

closed 2021-06-19 by github-actions[bot]