dresden-elektronik/deconz-rest-plugin #4835: Aqara/Xiaomi Multi Sensor not detected or falling out

AbDhops, 2021-05-03:
## Describe the bug

## Steps to reproduce the behavior
 
  1. Login to Deconz.
  2. Put the gateway in paring mode (navigating to sensor -> Add new sensor -> other
  3. Go to Aqara/Xiaomi Multi Sensor
  4. Long press Aqara / Xiaomi reset button (3~4 seconds) to put the device in pariing mode and to connect with Deconz
  5. Observe device LED blinks indicating device is entering in pairing mode
  6. Even after multiple connect attempts on multiple devices (tried 3 devices), device is not getting connected.

Issue can be reproducible atleast in my system every time.
## Expected behavior

Atleast after some attempts, device shall be connected.
## Screenshots

## Environment

 - Host system: NAS
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.10.04
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-05-03:
Hello, after the step 5, press shortly the button every 2/3s, to prevent the device going to sleep.

Mimiix, 2021-05-03:
Changed labels.

Did you read this? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

AbDhops, 2021-05-04:
Thanks for the link. Will try out. I am using the setp now for thsn 2
years. And in between, I also had reboots but never had any issue. Is this
new ?

sammycontrols, 2021-05-05:
Same problem, I have 6 of these and I could get 1 connected once and never again. 
Environment

    Host system: NAS
    Running method: Marthoc Docker container
    Firmware version: 266B0700
    deCONZ version: 2.11.02 / 3/14/2021
    Device: ConBee II
    Do you use an USB extension cable: Yes
Also tried on Windows 10 but it didn't help me
Aqara door sensor and motion sensor works
Above tip by Smanar doesn't help nor the link Mimiix posted

Mimiix, 2021-05-05:
@sammycontrols can you share a screen of your deCONZ?

sammycontrols, 2021-05-05:
Like this?
[image]

AbDhops, 2021-05-05:
So, same here. Tried the steps mentioned by @Mimiix. But no sucess. From the steps (step no. 4), node was not getting visible in Deconz GUI. 
I also made the video.
Then I removed the sensor and put the new battery in one of them. Tried repairing and it got connected. I used the battery removed from sensor 1 to another sensor, and tried re-pairing it. It also got connected. Like this, I connected all 4 sensors, which was not getting connected before. Don't know what I achived when I swap the battery from one sensor to another.

Smanar, 2021-05-05:
I realy have no ideas.
Can you share a log with "info"/"info_l2"/"error" with the inclusion procedure, start the logging somes seconds before active it.

sammycontrols, 2021-05-06:
Yes! 
Thanks to AbDhops tip on battery I could pair the Multi Sensor.
The battery on the two I tried where around 50% and when I replaced them with a 90% battery from the one that worked I could pair them. After pairing I put back the 50% battery and they still work.
Thank you AbDhops!

Smanar, 2021-05-06:
You can change the battery, a battery at 50 % is dead, not a 50% of his life.

disaster123, 2021-05-15:
I've the same problem running Conbee II with Firmwarre 266B0700 - might it be related to that FW? My batteries showed 90% on each Aqara sensor.

Smanar, 2021-05-15:
You looses all your 6 sensors ? You have others ones working ?
Can you show a picture of your mesh ? Are you sure devices are full included, (visible on node picture), remember aqara is not compatible with all routers.

disaster123, 2021-05-15:
They were working fine for about 2 weeks i then tried to include another sensor... and than suddently started the mess and i lost all. I can't show any mesh and there even wasn't one at all as all sensors were directly near the stick. I already tried different versions, upgrades, downgrades... i've now deleted and resettet everything and will start from a fresh new system.

Smanar, 2021-05-15:
If you can't take a look on mesh, can try with logs, connection issues are visible on it.
Try with avoiding usb 3.0 and using an USB extension cable.

disaster123, 2021-05-15:
After resetting everything it started to work again... let's see how long its lasting.

AbDhops, 2021-05-15:
Where to see the logs, if I am running deconz in the docker container...

AbDhops, 2021-06-06:
Issue must be because of low battery. So, issue can be close. New issue / question will be raise to see the logs in docker container.

closed 2021-06-06 by AbDhops