dresden-elektronik/deconz-rest-plugin #4836: Philips Hue: Motion sensor. Value for temperature and light level is always 0 and will not be updated.

xADDRx, 2021-05-04:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

xADDRx, 2021-05-04:
[image]

cllatser, 2021-05-06:
Hi, delete it and pair it again. Sometimes it doesn't work right the first time.

xADDRx, 2021-05-06:
> Hi, delete it and pair it again. Sometimes it doesn't work right the first time. 

Hi, I tried so many times. Sometime without values and sometimes without Temp and light values. Doesn't work

cllatser, 2021-05-06:
Okay, now leave it for a bit. Sometimes it takes a few minutes. from time to time you see passing in front to activate it.

xADDRx, 2021-05-06:
I left it like that the day before yesterday, but the values ​​still apply today. I'll wait another day to see if something happens.

xADDRx, 2021-05-07:
> Okay, now leave it for a bit. Sometimes it takes a few minutes. from time to time you see passing in front to activate it.

Unfortunately does not work. I have integrated over 100 devices, all without any problems. Only this sensor drives me crazy

cllatser, 2021-05-07:
I understand you perfectly, I have 5 indoor and 3 outdoor and no problem. In some I had to test the link 5 times. Another thing you have to know is that it refreshes the temperature and the lux when it detects movement.

bee-con, 2021-05-19:
Sounds familiar. Have you tried reading the temperature and illuminance values manually from within deConz? That's what usually helped me when I experienced that.

[image]

closed 2021-06-18 by github-actions[bot]