dresden-elektronik/deconz-rest-plugin #4838: Tuya Smart Plug missing metering data (_TZ3000_cphmq0q7)

muratcorlu, 2021-05-05:
## Describe the bug

I bought 4 [Tuya Zigbee smart plugs](https://www.aliexpress.com/item/1005002344798281.html?spm=a2g0s.9042311.0.0.30fe4c4dnraeY4) and easily connected to Phoscon (running with Conbee II  on Rpi4). I use them on my HomeKit setup via Homebridge. One of them worked perfectly with metering functionality but 3 of them doesn't report metering data. Working one has `_TZ3000_8nkb7mof` vendor name and others has `_TZ3000_cphmq0q7`.

## Steps to reproduce the behavior

1. Start pairing a smart plug with _TZ3000_cphmq0q7 vendor name.
2. Pair with with Phoscon GW
3. Restart HomeBridge
4. Check if you have consumption data on Eve Home app. (Or in HomeAssistant)

## Expected behavior

I expect to see consumption data on Eve Home App or HomeAssistant.

## Screenshots

[image]

[image]

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian / HomeBridge / homebridge-hue plugin / Eve Home App
 - Firmware version: 266b0700
 - deCONZ version: 2.10.04
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: P1 adapter, USB Bluetooth

## deCONZ Logs

I use headless deconz. I currently couldn't figure out how to fetch debug logs. I'll update if I can.

## Additional context

NijsKei, 2021-05-19:
I have exactly the same problem. Why it always must be difficult. 2 are working well a 2 are working not. Only the switches are doing their job but no measering.

oywino, 2021-05-19:
I have exactly the same problem. 
Setup: 
Conbee Firmware: 266B0700
deConz Version: 2.11.02 beta / 4/30/2021

Device is registered in Phoscon as a Light. No measuring data is present. On/Off works fine.
Here are my deConz screen dumps:

[image]

[image]

[image]

On/Off Cluster:
[image]

Simple Metering Cluster:
[image]

[image]

REST-API:
```
{
    "etag": "db47fd907f3381714dc29b86101ac2f3",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-19T18:38Z",
    "manufacturername": "_TZ3000_cphmq0q7",
    "modelid": "TS011F",
    "name": "On/Off Tuya 1",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "67",
    "type": "On/Off plug-in unit",
    "uniqueid": "84:71:27:ff:fe:8d:96:1d-01"
}
```

Smanar, 2021-05-19:
For me the code is blocking for the reason exposed by @muratcorlu, the device have
- No support attribute 0x4000 
- Empty value for 0x0006

And the code don't like that, there is a hack for light device but not for sensor.
There is many way to help

Ugly one like

```
        if (node->nodeDescriptor().manufacturerCode() == VENDOR_SAMJIN)
        {
            swBuildIdAvailable = false; // empty string
            dateCodeAvailable = false; // unsupported attribute
        }
        else if (modelId == QLatin1String("HG06323")) // LIDL Remote Control
        {
            swBuildIdAvailable = false; // unsupported attribute
            dateCodeAvailable = false; // empty string
        }
```

But for me it s possible with using dateCode.isNul(), but in this situation we will have swversion with null value.

Need to ask first the autorisation at others devs.

oywino, 2021-05-19:
@Smanar, thanks for stepping in - but what you just said is "gibberish" to me. Could you "translate" it into layman's English? 😊

Smanar, 2021-05-20:
Lol.
I have a code to test

```
sudo apt install deconz-dev
git clone --branch tuya_plug_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

This code (if it work) try to get the Swbuild, then the date code and finaly the Application version to fill the "swversion" field.

You will have in log with "info"

```
DBG_Printf(DBG_INFO, "[4.1] Get sw build id\n");
DBG_Printf(DBG_INFO, "[4.1] Get date code\n");
DBG_Printf(DBG_INFO, "[4.1] Get Application version\n");
```

And after the device will be included.

oywino, 2021-05-20:
Sorry, I don't understand.
The issue in this thread is: **Tuya Smart Plug missing metering data (_TZ3000_cphmq0q7)**
That is my problem too.
Is your "code" going to solve this problem if your "test" is successful ?

muratcorlu, 2021-05-20:
@oywino Smanar has a fix on his/her fork and asks for test it if it fixes our problem.

But I also couldn't understand this part:

> This code (if it work) try to get the Swbuild, then the date code and finaly the Application version to fill the "swversion" field.

@Smanar What is *swbuild*?

oywino, 2021-05-20:
_swbuild_ is Software Build number I guess. 
I'd love to test whatever it is @Smanar want us to test, but I have no clue how to do this.
Is it possible to prove a step-by-step in human understandable terms ?  😊

Smanar, 2021-05-21:
Sorry.

To make short, the deconz code ATM, prevent sensor creation for device without "swbuild", it s a field visible in the Json.
By defaut it s the attribute 0x4000 visible on capture on first post, if this attribute is not available, the code take the datacode (0x0006) but for this device the value is empty, so the code stop here.
With my chnage it will use the application version.

The procedure is explained here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

But you need a linux OS with full OS, I can't explain how to do on docker.

oywino, 2021-05-21:
Ok, I see.
But what did @Mimiix mean when he wrote that "It needs to be whitelisted" ?
I was hoping that meant that a quick-fix was imminent.... 😢

Smanar, 2021-05-21:
And not sure my fix will work too ^^.
In all situation, we realy need a test, and for that we need you, we haven't the device.

oywino, 2021-05-22:
I would need another ConBee dongle and a test system to do this. ANd I don't have that. My existing set up is a productions system that I cannot take down. It would be easier for me to send a device to you.
But it would be nice if @Mimiix would join us again and clarify his opinion.

Smanar, 2021-05-22:
Don't worry, you are not alone with this bug.
No one here have the device, the problem, and a linux machine ?

Mimiix, 2021-05-22:
@oywino where do you need my opinion on?

oywino, 2021-05-23:
> @oywino where do you need my opinion on?

Hi @Mimiix , you wrote (if I understood correctly) that all that was needed was to whitelist the device (?)
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4838#issuecomment-844002948

I was hoping that this could turn out to be an easy fix ?

Mimiix, 2021-05-23:
> Hi @Mimiix , you wrote (if I understood correctly) that all that was needed was to whitelist the device (?) [...]
Well that's what I thought. But @Smanar says it isn't that easy, so I'm with him on this one. I'm not a developer so I can't change anything about this unfortunately.

oywino, 2021-05-23:
> Well that's what I thought. But @Smanar says it isn't that easy, so I'm with him on this one. I'm not a developer so I can't change anything about this unfortunately.

Ok, I see. 
@Smanar - are you a developer (maintainer) ? 
I wish I could be of any help....

Smanar, 2021-05-23:
For the moment need someone with the device, with the issue and with a linux machine to test the code modification.

oywino, 2021-05-23:
I guess a "linux machine" can be arranged by simply installing Linux on a laptop. Which Linux do you recommend, and what would be the preferred installation config?

Smanar, 2021-05-24:
I think all Linux version are supported, I have see some issues with dev package not found for some X64 but I think now it s corrected.
There is that in doc

```
Ubuntu
Supported Ubuntu versions

    Ubuntu 16.04 LTS 64-Bit PC (AMD64)
    Ubuntu 18.04 LTS 64-Bit PC (AMD64)
    Ubuntu 20.04 LTS 64-Bit PC (AMD64)
```

Classic install, not "light" "custom", headless (the desktop can be usefull for debugging) or docker.

oywino, 2021-05-24:
Ok, I need to find one more ConeBee II because I cannot disconnect the one I have in production.
I hope setting up two ConBee coordinators in my house isn't going to cause any problems (interference) with my existing network (?)

Smanar, 2021-05-24:
No, no danger, but just to make test, you can just unplug the conbee from your production machine, make the test, and replug it.
You just need 2 mn to check if the code is working (if you have the sensor or not), and 15 mn to test it.

oywino, 2021-05-24:
Ok, do you have step-by-step on how to install and config deConz on Ubuntu 20.04 (inkluding links)?

Smanar, 2021-05-24:
Yep, first update the OS then 

https://phoscon.de/en/conbee2/install#ubuntu

Then after with command line

```
sudo apt install deconz-dev
git clone --branch tuya_plug_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

oywino, 2021-05-24:
Should I do all this first:

[image]

Smanar, 2021-05-24:
Yep, it s to configure the package stuff.

oywino, 2021-05-24:
Ok, and you are 100% sure that using my production Conbee isn't going to screw up my existing config when I plug it back again in my production server (after our "experiment") ?  
I have a large number of ZigBee devices connected to it now. My whole house will stop working when I disconnect it 😢

Smanar, 2021-05-25:
>Ok, and you are 100% sure that using my production Conbee isn't going to screw up my existing config when I plug it back again in my production server (after our "experiment") ?

Sure, the network configuration is stored on the device (but not modified if you just plug/unplug/replug, and devices configations are stored on host.

>I have a large number of ZigBee devices connected to it now. My whole house will stop working when I disconnect it

But yes, all your devices will stop working, and the mesh can need some time to be full working. But like I have said, can wait some days to see if someone else can test the code, for exemple @muratcorlu  or @NijsKei

SwoopX, 2021-05-25:
Just make a backup via Phoscon as a safety measure.

oywino, 2021-05-25:
I tried to follow the procedure.
But when I got the command: 
```
sudo apt install deconz-dev
```
I just got these error messages:
```
E: Malformed entry 1 in list file /etc/apt/sources.list.d/deconz.list ([option] not assignment)
E: The list of sources could not be read

```

So, now what?

Smanar, 2021-05-25:
I think it have an error here

`sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz $(lsb_release -cs) main' > /etc/apt/sources.list.d/deconz.list"`

oywino, 2021-05-25:
I used the **beta** version. Can that cause problems?

Smanar, 2021-05-25:
>I can launch deconz, and it sees the coordinator, and immediately begins to populate the dashboard with all my devices in my house. It's a mess and it's impossible to identify these devices because they have only hexadecimal numbers and no names.

This is normal, remember the devices config is on the host, not on the gateway.
So somes devices appear on deconz, because devices are trying to connect, but deconz can't reconize them without the database.
Just ignore them.

For the error compilation, IDK, first time I see that.

Try to retry it from clean project (I don't use sudo on my side)
```
make clean
qmake && make
```

oywino, 2021-05-25:
I tried, but I get the `make` error just as described above (and I don't understand what this error means).
In addition, the copy command fails:
```
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```
because the directory does not exist.
So, I do not know if I did get your code installed in deconz properly, or not.
I can see the Tuya in the deconz, dashboard and in Phoscon, but I do not know how to check if the REST-API has all the attributes or not.

Smanar, 2021-05-25:
>So, I do not know if I did get your code installed in deconz properly, or not.

No, if it don't compile, the file is not created, so it can't be copied, so for the moment you can let the conbee on your production machine.

>I can see the Tuya in the deconz, dashboard and in Phoscon, but I do not know how to check if the REST-API has all the attributes or not.

If the compilation fails, you can't have it.

You have updated the OS ?

```
sudo apt update
sudo apt full-upgrade
```

oywino, 2021-05-25:
I did 
```
sudo apt update
sudo apt upgrade
```
but not
```
sudo apt full-upgrade
```

Smanar, 2021-05-26:
and same error message ?

oywino, 2021-05-28:
So, what now?

Smanar, 2021-05-28:
```
Could not create output file
make[1]: *** [Makefile.Release:543: ui_de_web_widget.h] Error 1
make[1]: *** Waiting for unfinished jobs...
Assemler messages:
Fatal error: can't create release/air_quality.o: permission denied
make[1]: *** [Makefile.Release:573: release/air_quality.o] Error 2
make[1]: Leaving directory '/deconz-rest-plugin'
make: *** [Makefile:42: release] Error 2
```

Sorry, realy have no idea where it can be from.

Mimiix, 2021-05-28:
Not running sudo could be it?

oywino, 2021-05-28:
> @oywino smanar is doing all he can. He's trying his best. That would go beyond what he should do and he is going above and beyond already.

Sorry - I tried earlier to find out his position in this so that could adjust my expectations accordingly, but I guess I never got an answer, so I assumed he was the maintainer with all privileges and the ultimate knowledge (which I myself don't have ).
I am prepared to do what ever is being asked of me in order to assist, but as much of this is beyond my expertise, I need a step-by-step in order to succeed.  
And yes, I did after a while understand that I had to add 'sudo' to every command, and I did.
Isn't there anyone who can step in shortly and assist us so that I can complete the test that smanar wants me to do?
I am prepared. I'm just a bit lame. 😶

Smanar, 2021-05-28:
I realy don't understand, your procedure is good, you are not the first one that do that, but it s the first time I m seing this result.
Try to restart but in another folder (with more permissions)
Just this part

```
sudo apt install deconz-dev
git clone --branch tuya_plug_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

oywino, 2021-05-29:
Hi @Smanar, sorry for being so ignorant, but please know that I'm not a Linux guy.  So, you must provide more guidance; What folder do you think I should use?  How do I assign more permissions on  folder level and subfolder and files ?

Smanar, 2021-05-30:
I m not a linux guy too ^^.
Try to mke a new folder, on the "basic root" that open when you run the console.
And try to change it's right using

`sudo chmod -R 777 folder_name`

then go in it
`cd folder_name`

and retry

```
sudo apt install deconz-dev
git clone --branch tuya_plug_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

oywino, 2021-06-11:
Hi @Smanar,  Sorry for my late reply. I have been on a business trip.
But I haven' given up yet. Hope to find some more time next week.
Meanwhile. - I don't understand why HA is unable to detect these values, when they are easily visible in deConz:
[image]

If i plug/unplug something, I can sit and watch the values change in deConz.

Smanar, 2021-06-11:
Ha take value in the API, not in deconz.

Smanar, 2021-06-12:
It s just because I have make the remark as device with versions are working and oher not.
So just for test, the last code force a versions for not working devices.

thorschtn, 2021-06-15:
@Smanar, I have the same Plugs, and a conbee 2 and a Linux environment, so I might be able to help.

Unfortunately the  compilation fails:

`root@smartnuc:/tmp/deconz-rest-plugin# qmake && make -j2
Info: creating stash file /tmp/deconz-rest-plugin/.qmake.stash
make -f Makefile.Release
make[1]: Entering directory '/tmp/deconz-rest-plugin'
/usr/lib/qt5/bin/uic de_web_widget.ui -o ui_de_web_widget.h
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.11.03\" -DGW_SW_DATE=1621531975 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"71a99bb44e2ce19fa1eeafba347342d70829a60d\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/air_quality.o air_quality.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.11.03\" -DGW_SW_DATE=1621531975 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"71a99bb44e2ce19fa1eeafba347342d70829a60d\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/authorisation.o authorisation.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.11.03\" -DGW_SW_DATE=1621531975 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"71a99bb44e2ce19fa1eeafba347342d70829a60d\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/backup.o backup.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.11.03\" -DGW_SW_DATE=1621531975 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"71a99bb44e2ce19fa1eeafba347342d70829a60d\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/bindings.o bindings.cpp
bindings.cpp: In member function ‘void DeRestPluginPrivate::checkLightBindingsForAttributeReporting(LightNode*)’:
bindings.cpp:2581:93: error: conversion from ‘std::vector<deCONZ::ZclCluster>::const_iterator’ {aka ‘__gnu_cxx::__normal_iterator<const deCONZ::ZclCluster*, std::vector<deCONZ::ZclCluster> >’} to non-scalar type ‘QList<deCONZ::ZclCluster>::const_iterator’ requested
[... cut ...]
bindings.cpp:3396:35: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<deCONZ::SimpleDescriptor>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
 3396 |                 for (int j = 0; j < sensor->node()->simpleDescriptors().size(); j++)
      |                                 ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile.Release:617: release/bindings.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/tmp/deconz-rest-plugin'
make: *** [Makefile:42: release] Error 2`

oywino, 2021-06-15:
Welcome @thorschtn, now we are three in the boat.
Did you manage to get deConz running at all?
I too tried to follow Smanars instructions, but I ran into issues above my head and couldn't even get the compiler to start. But I was planning to try again this week (unless you beat me to it)  🙂

Smanar, 2021-06-15:
Lol.
And the worst, I not sure my code will change something.

So I have updated the branch with the last official from today
Be sure the OS is updated, almost the dev-package

Use the command "make clean" to clean the old files

```
cd deconz-rest-plugin
git pull
make clean
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

To be sure you are on good branch
`git branch`

you need to be on tuya_plug_1

oywino, 2021-06-16:
When you write "... updated the branch with the last official from today" do you mean v12.1  ?

Smanar, 2021-06-16:
Yep at least v2.12.1-beta or more if someone have updated some code.

oywino, 2021-06-16:
Sorry, of course I meant v2.12.01
I did install that just now, but I didn't see any changes to the Tuya Plug.
Should it change anything?
Do I need to unpair and repair it?
What should I expect?

Mimiix, 2021-06-16:
> Sorry, of course I meant v2.12.01
> I did install that just now, but I didn't see any changes to the Tuya Plug.
> Should it change anything?
> Do I need to unpair and repair it?
> What should I expect?

You should re pair :)

Smanar, 2021-06-17:
But the code is based on it, but you still need to compile the code to replace the lib file.

harowi, 2021-06-23:
Same issue here. Smart plug could be paired easily and is working fine using Home Assistant. But the metering data are not available there although they are shown in deCONZ.

oywino, 2021-06-23:
@harowi, do you have the possibility to try to compile and run the deConz version that @Smanar has modified?

harowi, 2021-06-23:
@oywino unfortunately not.

Smanar, 2021-07-05:
Manup have made a PR recently that can perhaps help for this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5092

RuneNyhuus, 2021-07-05:
I'm looking forward to a fix for this too :)
Just got 3... and they only work with metering on zigbee2mqtt...
Would be awesome to work in deconz

[87 more replies]