dresden-elektronik/deconz-rest-plugin #4839: Cannot open Deconz app via VNC

hesspoint, 2021-05-05:
## Describe the question or issue you are having

Hi,
I have the issue that I have a Pi with a Raspbee II and the Pi is headless.

I can connect to the Pi via SSH and also via VNC on my iPad Pro (also tried on my MacBook). I can start Deconz on the command line and I can access the Phoscon app, add devices etc. everything fine.

However I want to open the Deconz app to be able to see the signal strength of the devices.

So I connect to VNC  and then open Programming > Deconz. And nothing happens. I thought the app is broken but when I connect the Pi to a screen and keyboard then I can open the Deconz app without any problem. But not in VNC. That is super strange.

Also I cannot use that way because I need to move the Pi to a different room and I need to to see the signal strength from the location where it is normally located. So VNC is the only option.

## Screenshots

## Environment
Raspbee II on Raspberry Pi 4b
I think all latest firmware
## deCONZ Logs
Not sure where I can get logs

## Additional context

senilio, 2021-05-07:
This is likely due to your DISPLAY environment variable being different when connected via VNC and via keyboard+monitor.

You can check what DISPLAY the deconz-gui service tries to start up on (likely ":0")
```
sudo systemctl cat deconz-gui
```

Next check what this var is set to in your VNC session, by opening a terminal and go:
```
echo $DISPLAY
```

Your systemctl service file needs to correspond to this. 

As an example, let's say your DISPLAY var within your VNC session is ":1.0"...

1. Create an override file for the deconz-gui service by running `sudo systemctl edit deconz-gui`. Put the following content in the editor, then save the file.
```
[Service]
Environment="DISPLAY=:1"          # use the display number from the first step
```
2. Update the systemd service by running `sudo systemctl daemon-reload`
3. Now try starting the deCONZ via the desktop shortcut, or by running `sudo systemctl start deconz-gui`.

zedc92, 2021-05-10:
I had exactly the exact same issue. The solution works perfectly. Thansk @senilio

hesspoint, 2021-05-12:
Awesome thanks so much. That did the trick! You made my day!

closed 2021-05-12 by hesspoint