dresden-elektronik/deconz-rest-plugin #4840: Wrong direction for motorised blinds _TZE200_xuzcvlku TS0601

Sian-Lee-SA, 2021-05-06:
## Describe the bug

The seems to be an issue with the api command for a cover state (_TZE200_xuzcvlku, TS0601) motorised blind chain. The positions seem to be correct and the state of close or open are also correct but when issuing a state `{'open': true}` or '{'open': false}` then the actions are inverted. So if the blind is already closed and I call `{'open': true}` then the motor won't run but I I do 
'{'open': false}` then it will actually open the blinds and vise versa.

These are the status for when the blind is closed:
`
{
    "etag": "bf042601379e4bfa12971a7bd45c459c",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-06T06:28Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
`

and these are the status when the blinds are open:
`{
    "etag": "dd89d684677943c3125191639d5ef16f",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-06T06:29Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 0,
        "lift": 0,
        "on": false,
        "open": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
`

Executing a position or lift to 100 or 0 will give the correct result though.

## Steps to reproduce the behavior

## Expected behavior

The open and close commands should be inverted but not the values

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-05-06:
I don't know what to do.
This device is not new, so if I reverse the command, it will reverse it for users that already use it, and it seem you are the first one for this device.

The Json look fine too, the logic is

```
shutter closed
on= true
bri = 255
open = false
lift = 100%
```

So the problem can be from the mounting. Can you try to reverse the side using the api request 

`/*! PUT, PATCH /api/<apikey>/lights/<id> with {"reverse":true}`

So with curl, something like 

`curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://IP:PORT/api/KEY/lights/ID`

Sian-Lee-SA, 2021-05-07:
I have to say that I found it strange that i couldn't find a related issue with this and that I seem to be the only one. I did the reverse prior to opening this issue and the open command opens the blinds and the closing will close the blinds but the vlues are messed up...

These are the values when open:
```
{
    "etag": "50b3139b3510f68a18f8de7c4681d328",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-07T01:49Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
```

and these are the values when closed:
```
{
    "etag": "82e9ad2b8efa58f09ad19f23bc505aea",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-07T01:49Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 12,
        "lift": 5,
        "on": true,
        "open": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
```

I mean...it's not a huge issue, when I call google assistant to close and open the blinds through home assistants integration then it opens and closes (but not if I reverse it like mentioned above) as expected (due to the source code using the set_position service and not the open and close services). I could do a band-aid fix by overriding the deconz cover.py for home assistant and inverting the open and close commands which would just mask the issue. fortunately this issue isn't breaking anything. But i thought maybe there may had been something overlooked when implementing this device on deconz...

ebaauw, 2021-05-07:
Please don't.  `state.open` should be computed from `state.lift`.  If `state.lift` equals 0, `state.open` should be `false`.

It looks like the blinds aren't calibrated?  `state.lift` reports 5 when closed, instead of 0.  On some roller blinds (like the IKEA, I found out the hard way), the blind might get closed the wrong way around, when not calibrated and and not stopping when closed.  After fully opening the blind, the rod continues turning, now closing the blind, while the electronics think they're still opening the blind.

Smanar, 2021-05-07:
Yep, it s a calibration sympthom, but the 2 second JSON have bri and open reversed, so better to use the 2 first (without the reverse command). On thoses one all is perfect, except the direction is not the good one.

But manup want to try something
`Have to try a few things out but would vote to keep /lights out of this and add something generic to /devices so you can do a PUT /devices/<uniqueid>/[config|state]/<attribute>`

Will be a good solution for that.

Sian-Lee-SA, 2021-05-08:
> Please don't. `state.open` should be computed from `state.lift`. If `state.lift` equals 0, `state.open` should be `false`.
>
> It looks like the blinds aren't calibrated? `state.lift` reports 5 when [...]

Actually if `state.lift` equals 0-99 then `state.open` is `true` according to [api-doc](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/#window-covering-parameters). Generally my driver does go to 0 and 100 respectively but not all the time which may leave a 0-5 gap but is no real issue (probably a firmware issue in that regards).

Sian-Lee-SA, 2021-05-08:
Also if i was to control the blind via lift state then I get the results expected... So doing
```
{
    "lift": 100
}
```
Will close the blinds as expected... I would've assumed the lift state and open state would've used the same or similar logic?

Smanar, 2021-05-08:
Ha ? (on the first situation, without using the "reverse" command"

So if you done

{'open': false} it open the cover
but
{"lift": 100} close it ?

If yes, the problem is on my side.

Sian-Lee-SA, 2021-05-08:
Yes that is correct

Sian-Lee-SA, 2021-05-08:
Just did a quck test to make sure I'm not losing my mind :)...

So I set the blinds to 50% and did a test on lift and open states and these are the results (without reverse command)

```
{
    "lift": 100
}
```
Closes the blinds

```
{
    "open": false
}
```
Opens the blinds

```
{
    "open": true
}
```
Closes the blinds

```
{
    "lift": 0
}
```
Opens the blinds

Again these were all tested with the blinds being 50% opened/closed

Smanar, 2021-05-08:
Ok so have started a modification.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch covering_9 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

It will reverse the command open/close for 2 devices
- _TZE200_xuzcvlku
- _TZE200_nueqqe6k

But I have not yet reversed the returned value (not sure if needed, for me, it is)

So can you check values with "open" and "lift" command after the cycle is finished.

Sian-Lee-SA, 2021-05-09:
> Ok so have started a modification.
> 
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> So for you, after having installed deconz :
> 
> ```
> sudo apt install deconz-dev
> git clone --branch covering_9 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```
> 
> It will reverse the command open/close for 2 devices
> 
> * _TZE200_xuzcvlku
> * _TZE200_nueqqe6k
> 
> But I have not yet reversed the returned value (not sure if needed, for me, it is)
> 
> So can you check values with "open" and "lift" command after the cycle is finished.
> 
> In fact the command can work if you use "open" and "lift" in the same request.

Just tested and works as expected now! Thanks for that! I had to compile and test on a raspberry pi so will have to wait till the next release to make full use of the fix as my dongle is on a NAS Bridge Server.

Sian-Lee-SA, 2021-05-09:
Also the values seemed accurate aswell

Sian-Lee-SA, 2021-05-09:
But the returned value for
```
{
    "open": true
}
```
gives:
```
[
    {
        "success": {
            "/lights/17/state/open": false
        }
    }
]
```
So I think the return value is reversed

Smanar, 2021-05-09:
Have updated the code, you can retry
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But BTW, if you make
```
{
    "open": true
}
```

xhen the move is finished the json is good ? the "open" value don't need to be reversed too ?

Sian-Lee-SA, 2021-05-09:
ok everthing looks good....

Reesults:
```
{
    "open": true
}
[
    {
        "success": {
            "/lights/17/state/open": true
        }
    }
]
```
GIves:
```
{
    "etag": "42086c7c7eebb31e7ff83d2d6ce08784",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-09T10:05Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 0,
        "lift": 0,
        "on": false,
        "open": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
```

Then
```
{
    "open": false
}
[
    {
        "success": {
            "/lights/17/state/open": false
        }
    }
]
```
Gives
```
{
    "etag": "8847fb424d4a4fd7b2b3e0974c457f9d",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-09T10:07Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Living Room Blinds",
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:8c:72:3d-01"
}
```

To me that all looks good. (Also did the lift commands 0-100 and they work as expected) 😄

Smanar, 2021-05-09:
Nice, thx, will make the PR

closed 2021-05-09 by manup