dresden-elektronik/deconz-rest-plugin #4841: Hama Smart Radiator Thermostat (TS0601 _TZE200_yw7cahqs)

franciscogouveia, 2021-05-06:
## Device

- Product name: Hama Smart Radiator Thermostat
- Manufacturer: _TZE200_yw7cahqs
- Model identifier: TS0601
- Device type:
  - Thermostat

## Screenshots

[image]

### Basic

[image]

### Identify

### Alarms

### Device Temperature

### Groups

[image]

### Scenes

[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

[image]

[image]

[image]

Searched in code and issues, couldn't find this specific one. This seems to be relevant: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4784

Smanar, 2021-05-06:
Hello have just added the device, can you make a try on your side, to test it ?
https://github.com/Smanar/deconz-rest-plugin/tree/tuya_trv_6

I can explain how to do, you just need a linux machine with full OS.

franciscogouveia, 2021-05-06:
Hi @Smanar, thank you!

I use Ubuntu 20.04 and run deconz from a docker container (using marthoc/deconz).

Here is what I did in your branch:

```
qmake && make -j2
```

It ran successfully and the file `libde_rest_plugin.so` was created. I mounted this file in the docker container, specifically:

```
./libde_rest_plugin.so:/usr/share/deCONZ/plugins/libde_rest_plugin.so
```

I fixed the ownership of the file to match the other plugins' files:

```
:/usr/share/deCONZ/plugins# ls -la
total 3436
drwxr-xr-x  2 root root    4096 May  1 06:04 .
drwxr-xr-x 10 root root    4096 May  1 06:04 ..
-rwxr-xr-x  1 root root 3215912 May  6 15:42 libde_rest_plugin.so
-rwxr-xr-x  1 root root   52464 Apr 30 17:01 libde_signal_plugin.so
-rwxr-xr-x  1 root root  233496 Apr 30 17:01 libstd_otau_plugin.so
```

Unfortunately, the rest plugin is not loading anymore:

[image]

Tried to check the container logs, logs in /var/log files inside the container and also the debug view of deConz but couldn't find anything relevant.

Docker output:

```
Starting deconz_deconz_1 ... done
Attaching to deconz_deconz_1
deconz_1  | [marthoc/deconz] Starting deCONZ...
deconz_1  | [marthoc/deconz] Current deCONZ version: 2.11.02
deconz_1  | [marthoc/deconz] Web UI port: 9080
deconz_1  | [marthoc/deconz] Websockets port: 9443
deconz_1  | [marthoc/deconz] VNC port: 5900
deconz_1  | Killing Xtigervnc process ID 23... which was already dead
deconz_1  | Cleaning stale pidfile '/root/.vnc/ghost:0.pid'!
deconz_1  | Cleaning stale x11 lock '/tmp/.X0-lock'!
deconz_1  | Cleaning stale x11 lock '/tmp/.X11-unix/X0'!
deconz_1  | 
deconz_1  | New 'ghost:0 (root)' desktop at :0 on machine ghost
deconz_1  | 
deconz_1  | Starting applications specified in /etc/X11/Xvnc-session
deconz_1  | Log file is /root/.vnc/ghost:0.log
deconz_1  | 
deconz_1  | Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd ghost:0 to connect to the VNC server.
deconz_1  | 
deconz_1  | WebSocket server settings:
deconz_1  |   - Listen on :6080
deconz_1  |   - Flash security policy server
deconz_1  |   - Web server. Web root: /usr/share/novnc
deconz_1  |   - SSL/TLS support
deconz_1  |   - Backgrounding (daemon)
deconz_1  | [marthoc/deconz] NOVNC port: 6080
deconz_1  | QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
deconz_1  | libpng warning: iCCP: known incorrect sRGB profile
deconz_1  | libpng warning: iCCP: known incorrect sRGB profile
deconz_1  | 16:51:48:102 HTTP Server listen on address 0.0.0.0, port: 9080, root: /usr/share/deCONZ/webapp/
deconz_1  | 16:51:48:245 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
deconz_1  | 16:51:48:289 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
deconz_1  | 16:51:48:292 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
deconz_1  | 16:51:49:337 COM check bootloader
deconz_1  | 16:51:49:347 COM detected application
deconz_1  | 16:51:49:359 Device firmware version 0x26660700 ConBee II
deconz_1  | 16:51:49:363 unlocked max nodes: 200
deconz_1  | 16:51:49:453 Device protocol version: 0x010C
deconz_1  | 16:51:49:458 new node - ext: 0x00212effff0423db, nwk: 0x0000
deconz_1  | 16:51:49:501 Current channel 15
deconz_1  | 16:51:49:513 CTRL got nwk update id 0
deconz_1  | 16:51:49:517 CTRL ANT_CTRL 0x03
deconz_1  | 16:51:49:541 Device protocol version: 0x010C
deconz_1  | 16:51:49:589 Current channel 15
deconz_1  | 16:51:49:601 CTRL got nwk update id 0
deconz_1  | 16:51:49:605 CTRL ANT_CTRL 0x03
deconz_1  | 16:51:51:103 ZDP discovery done in 381 ms
```

It seems to not detect the rest plugin. Any hints on how to debug it?

Smanar, 2021-05-06:
No sorry it s for that I m asking if you have a full OS.
The compilation is host dependent, so you can't compile it on Ubuntu and move the file on the docker. I know it s possible, but I m not linux engineer (and I hate docker for that).

The code change is small, I think it will be be enought, can make the PR in blind mode if you can't test it.

franciscogouveia, 2021-05-06:
I figured it out and it works! :) I can now successfully control and read the thermostat state from HA.

[image]

For those using docker, there is a way to test it. Just extend the image that you are using and compile it there.

Example of a Dockerfile (in my case, I was using marthoc/deconz:amd64-2.11.02):

```
FROM marthoc/deconz:amd64-2.11.02

ARG REPOSITORY=https://github.com/Smanar/deconz-rest-plugin.git
ARG GIT_COMMIT=283f0b2c6dda53a39d6f0c7ff483572b20bba299

ENV DEBIAN_FRONTEND="noninteractive"

RUN apt-get update && apt-get install -y git qt5-default libqt5websockets5-dev libqt5serialport5-dev sqlite3 libcap2-bin lsof curl libsqlite3-dev libssl-dev g++ make

COPY ./deconz-dev-2.11.02.deb /tmp/deconz-dev-2.11.02.deb

RUN dpkg -i /tmp/deconz-dev-2.11.02.deb

RUN git clone $REPOSITORY && cd deconz-rest-plugin && git checkout $GIT_COMMIT && qmake && make -j2

RUN cp ./libde_rest_plugin.so /usr/share/deCONZ/plugins/libde_rest_plugin.so
```
You will need to change the GIT_COMMIT to match your needs. You can also change the repository if the code is somewhere else.

Then, you need to download and place the deconz-dev-X.XX.XX.deb (pick your version) next to the Dockerfile.

To build it (inside the Dockerfile folder):

```
docker build -t marthoc/deconz:amd64-2.11.02-test .
```

Once done, use the new docker image (marthoc/deconz:amd64-2.11.02-test) instead of the original (marthoc/deconz:amd64-2.11.02).

closed 2021-05-06 by franciscogouveia

franciscogouveia, 2021-05-06:
Sorry, I rushed into closing the issue. Feel free to re-open if you prefer to leave it open until merged.

Thanks!

franciscogouveia, 2021-05-26:
@Smanar just an update on the docker build: I moved it to a repository with instructions on how to use it: https://github.com/franciscogouveia/docker-deconz-rest-plugin-test

I hope it is useful.

Works well up to commit 8c87fbb012b6c71543a04b3d80982e66e62ed19d with deconz 2.11.05.

Unfortunately, it doesn't seem to build with the current state of master with deconz 2.12.00:

```
In file included from /usr/include/deconz.h:20,
                 from de_web_plugin.h:16,
                 from air_quality.cpp:1:
/usr/include/deconz/aps.h:68:10: fatal error: deconz/timeref.h: No such file or directory
 #include "deconz/timeref.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:571: release/air_quality.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/deconz.h:20,
                 from de_web_plugin_private.h:24,
                 from authorisation.cpp:13:
/usr/include/deconz/aps.h:68:10: fatal error: deconz/timeref.h: No such file or directory
 #include "deconz/timeref.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:590: release/authorisation.o] Error 1
make[1]: Leaving directory '/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2
```

But this is offtopic for this ticket. I'm probably just missing something :)

Smanar, 2021-05-26:
Yep probably somthing is not updated

`sudo apt install deconz-dev`

The code have moved a lot too, to to start with clean project

```
cd deconz-rest-plugin
make clean
```

ahue, 2021-05-30:
@franciscogouveia  @Smanar  Thanks for adding support for this! I just tested it using the HA deCONZ plug-in an can confirm it works partly. Setting values in HA via the thermostat widget works, but manual changes using the gear are not displayed in HA. Messages are received though.

https://i.imgur.com/WEYdRuQ.png

Also, the actual temperature and battery state is not displayed and no entities are available in HA.

Using deCONZ 2.11.5

Should I open an issue? How can I contribute?

Smanar, 2021-05-31:
You are right, have skipped some code part, are you able to test code modification ?

```
sudo apt install deconz-dev
git clone --branch tuya_trv_12 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

ahue, 2021-05-31:
Where would I do the above? I run HA in a Virtual Machine on Synology... not tinkered with the deCONZ add-on yet.

Smanar, 2021-05-31:
Arf, you need a linux machine to compile the code, with full OS (not with docker or HA OS), have you something like that ?

ahue, 2021-05-31:
Virtual Machine (VirtualBox) doing the job? Then I'll set up one.

Smanar, 2021-05-31:
I know it s possible, but never succed on my side (I can run deconz, but can't compile)
You can make a try https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

First install deconz https://phoscon.de/en/conbee2/install , them

```
sudo apt install deconz-dev
git clone --branch tuya_trv_12 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If it don't compile, you can give up direclty

franciscogouveia, 2021-06-01:
@ahue You can use this if you use docker containers: https://github.com/franciscogouveia/docker-deconz-rest-plugin-test

It's fixed now and works with latest version (v2.12.00).

What you need to do in this case is:

```
git clone https://github.com/franciscogouveia/docker-deconz-rest-plugin-test.git
cd docker-deconz-rest-plugin-test
docker pull marthoc/deconz
docker build -t my-deconz-image --build-arg PLUGIN_REPOSITORY=https://github.com/Smanar/deconz-rest-plugin.git --build-arg PLUGIN_GIT_COMMIT=tuya_trv_12 .
```

This will build your local docker image with the changes from @Smanar. To confirm that everything went well, check if the git commit of rest plugin API matches the one from @Smanar branch:

[image]

Smanar, 2021-06-01:
@franciscogouveia Woaaaa, terrific, can you add the complete procedure from your previous post, to compile branch from random fork (with random branch) on your github page ?

Smanar, 2021-06-02:
@ahue tell me if you are able to test the code, else I will make the Pr in blind mode (realy few change)

ahue, 2021-06-03:
Building on the Ubuntu VM I set up did not work unfortunately (see below). Docker I did not try yet - but still I would not know how to integrate the modified docker image with my HA instance that runs as VM on my Synology.

```
firmware_update.cpp: In member function ‘void DeRestPluginPrivate::checkFirmwareDevices()’:
firmware_update.cpp:545:73: error: ‘instance’ is not a member of ‘deCONZ::DeviceEnumerator’
  545 | deCONZ::DeviceEnumerator *devEnumerator = deCONZ::DeviceEnumerator::instance();
      |                                                                     ^~~~~~~~
```

franciscogouveia, 2021-06-03:
@ahue The error that you get on your Ubuntu VM is because you have an older version of deconz/deconz-dev installed. You need to make sure you have version 2.12.00 (still in beta).

On Ubuntu (steps taken from: https://phoscon.de/en/conbee/install#ubuntu):

```
wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -

# Here, use the beta repository (v2.12.00 is still in beta)
sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz \
            $(lsb_release -cs)-beta main' > \
            /etc/apt/sources.list.d/deconz.list"

sudo apt update

# Install both deconz and deconz-dev
sudo apt install deconz deconz-dev
```

ahue, 2021-06-03:
Thanks @franciscogouveia ! Compile worked, I now have the `libde_rest_plugin.so` file. Please assist me where to put this within the HA setup. I have file system access to the HA `usr/share` directory, but there is no directory `/usr/share/deCONZ/plugins`. Thanks!

franciscogouveia, 2021-06-03:
As far as I know, this file is not used by HA. This file is used by deCONZ.

HA has a deCONZ plugin which connects to the deCONZ.

You should find the folder `/usr/share/deCONZ/plugins` in the machine that runs deCONZ (this should be the same machine where the gateway USB stick is plugged).

Smanar, 2021-06-06:
The change is small, have made the PR before forget it https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4980