dresden-elektronik/deconz-rest-plugin #4842: Aqara Wireless Remote Switch H1 (Double Rocker)

tomillr, 2021-05-06:
## Device

- Product name: Aqara Wireless Remote Switch H1 (Double Rocker)
- Manufacturer: not listed in deCONZ GUI Basic cluster.
- Model identifier: lumi.remote.b28ac1
- Device type : 
  - Switch

## Screenshots

[image]

[image]

[image]

tomillr, 2021-05-09:
Anything I can do to help getting this devices integrated? Can I send you one of those switches perhaps?

Aqara is about to release a bunch of new Zigbee 3.0-version of their popular switches and sensors btw.

netnetaip, 2021-05-16:
I'm also intrested to see H1 switch integrated in ConBee II or RaspBee II devices

SwoopX, 2021-05-20:
You guys might want to give the latest firmware a try (http://deconz.dresden-elektronik.de/deconz-firmware/beta), since it has some new builtin specialties for Xiaomi. We recently discovered that the latest Xiaomi's require the coodinator to have a Xiaomi manufacturer code to eventually reveal their full set of clusters, which I feel is the case here. This is too less for a dual rocker.

With the new FW, please also make sure to reset and re-pair the device and push all buttons a couple of times while device search is running.

tomillr, 2021-05-21:
Ok, so I've updated my Conbee 2 stick to the latest beta firmware and I've setup a completely new and clean test system:

[image]

[image]

However, I'm not able to pair the Aqara H1 Wireless Remote Switch anymore. It only shows up, when I first connect at least one additional Zigbee endpoint. 

Could it be that the new Aqara switch tries not to connect to the Conbee stick but instead acts like one of those light link remotes from Ikea that work without the hub?

SwoopX, 2021-06-02:
Have you also pushed both buttons simultaneously? I'd expect to also have an endpoint 02.

tomillr, 2021-06-02:
> Have you also pushed both buttons simultaneously? I'd expect to also have an endpoint 02.

Yes, I've pushed both buttons in different combinations and length. Where should I look for endpoint 02 in the Devons GUI btw?

SwoopX, 2021-06-02:
ok, 2nd screenshot of your last post. Would you be able to compile the plugin for testing?

tomillr, 2021-06-02:
> ok, 2nd screenshot of your last post. 
So basically every possible button (left, right, left+right) should show up as its own section like "01" and "03" shown here?

[image]

> Would you be able to compile the plugin for testing?
Never done this before, but I'd gladly give it a try. Do you have some kind of guide I can work with?

SwoopX, 2021-06-04:
Sure, can do. I just need 1-2 days more as I want to pass you a nice experimental gimmick to sniff the communication :wink:

tomillr, 2021-06-04:
Played around with those switches a bit more and tried to connect them to a SmartThings Hub as well.

I've noticed, that the switch seems to automatically connect itself to a random Zigbee light or smart plug. This also happens when there's no Zigbee Hub connect at all and I've manually reset every device beforehand. 

Could it be, that the new Aqara H1 switches are by default configured to establish a direct connection to the light source. Similar to how those TRÅDFRI remotes from Ikea come pre-connected to the Zigbee light bulb and work without the need for a Zigbee hub?

SwoopX, 2021-06-05:
Alright, so the first code to test is ready. Please find the instructions how to compile below

```
git clone --branch=h1 https://github.com/swoopx/deconz-rest-plugin.git h1       # Clone repository
cd ./h1
qmake && make -j2                                                               # Compile
sudo systemctl stop deconz                                                      # Stop deCONZ headless (in case it is running)
sudo systemctl stop deconz-gui                                                  # Stop deCONZ GUI (in case it is running)
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ/plugins/bak_libde_rest_plugin.so  # Make plugin backup
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins/                      # Copy newly compiled plugin to deCONZ plugin folder
/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > h1_debug                           # Run deCONZ with GUI and debug output
```

**Please make sure to collect around 15 mins of debug data and reset and re-pair the device as very first action.** Also, be sure to try out all thinkable combination of presses (single, double, tripple, hold) for each button. When done, please share the created file `h1_debug` from the h1 folder.

If required, you can restore the original plugin after deCONZ has been closed:

```
sudo cp /usr/share/deCONZ/plugins/bak_libde_rest_plugin.so /usr/share/deCONZ/plugins/libde_rest_plugin.so
```
and start deCONZ again in ONE preferred mode
```
sudo systemctl start deconz              # Start deCONZ headless
sudo systemctl start deconz-gui          # Start deCONZ GUI
```

> Could it be, that the new Aqara H1 switches are by default configured to establish a direct connection to the light source. Similar to how those TRÅDFRI remotes from Ikea come pre-connected to the Zigbee light bulb and work without the need for a Zigbee hub?

Yes, kind of. If appropriately configured, switches can control lights, etc. directly without the need of a coordinator. Switches typically get predefined groups they send their commands to. If the to be controlled devices subscribes to that group, the coordinator is not required any more.

It could be that the switch has also an alternate mode of operation. Maybe the generated debug output gives some hints about that.

tomillr, 2021-06-12:
Compiling the plugin has proved to be a bit difficult. Had to first install the missing components (QTCreator, libqt5websockets5-dev, libqt5serialport5-dev) on my system before I was able to run the compile command. But the compilation stops with an error message:

```
root@DietPi:~/h1# qmake && make -j2
make -f Makefile.Release
make[1]: Entering directory '/root/h1'
/usr/lib/qt5/bin/uic -no-stringliteral de_web_widget.ui -o ui_de_web_widget.h
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.12.00\" -DGW_SW_DATE=1622991386 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"716fbf13eb6f060f0a1e770f931bf84c321ce3c6\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/air_quality.o air_quality.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.12.00\" -DGW_SW_DATE=1622991386 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"716fbf13eb6f060f0a1e770f931bf84c321ce3c6\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/authorisation.o authorisation.cpp
In file included from air_quality.cpp:1:
de_web_plugin.h:16:10: fatal error: deconz.h: No such file or directory
 #include <deconz.h>
          ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:581: release/air_quality.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from authorisation.cpp:13:
de_web_plugin_private.h:23:10: fatal error: sqlite3.h: No such file or directory
 #include <sqlite3.h>
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:601: release/authorisation.o] Error 1
make[1]: Leaving directory '/root/h1'
make: *** [Makefile:40: release] Error 2
```

SwoopX, 2021-06-12:
Ah, you're missing the deconz dev package.

`sudo apt-get install deconz-dev` should do the trick.

Qtcreator is a bit too much, but ok 😄

tomillr, 2021-06-12:
So, I've finally managed to setup a new Debian environment and try compiling you plugin. Provided I followed your instruction correctly, I wasn't able to have the switch join the the newly setup network. Maybe the log can provide some details:

https://pastebin.com/KJMZzGdV

First, I manually reset the H1 switch a couple of times and tried to pair it using the Phoscon webapp. After that didn't work, I tried to pair a INNR smart plug to check the connection with the Conbee 2 stick, which didn't show up also.

Maybe I can just have you send you one of the actually switches? Might that help?
https://www.amazon.de/WRS-R02-WIRELESS-REMOTE-SWITCH-version/dp/B094QL17GY/ref=sr_1_1?__mk_de_DE=ÅMÅŽÕÑ&dchild=1&keywords=Aqara+h1&qid=1623496859&sr=8-1

SwoopX, 2021-06-12:
Lemme check what you shared. I recall we recently had quite some strange reports that devices "refuse" to join. Interestingly, a restart of deconz seems to let the device appear automagically.

tomillr, 2021-06-14:
@SwoopX Thanks, let me know if you need any further data.

tomillr, 2021-06-18:
I've done some more testing and even got out my old Smart Thing v3 hub. But no matter what I try, the switch always seems to only establish a connection to a Zigbee 3.0 light source and not the hub itself.

For example, when using the h1 switch with the SmartThings hub, the hub doesn't detect the presence of the switch at all. Instead it automatically pairs itself to all the Zigbee 3.0 lights. So basically same behavior as when using a setup with the Conbee stick.

SwoopX, 2021-06-18:
Ping me on Discord if you quickly wanna send one over 😉

closed 2021-07-06 by manup

MiAutomations, 2021-07-06:
Hello Team 

Any news about the support of this device in the deconz plugin available on the HA Home Assistant once the device is not yet supported deCONZ Current version: 6.9.0

[image]

Thank you and best regards

SwoopX, 2021-07-06:
This question was more a joke, right? 😄

MiAutomations, 2021-07-06:
> This question was more a joke, right? 😄

No, it's only to check if you have a ideia about when this is available on the deconz plugin on the HA ...

SwoopX, 2021-07-06:
Support is now included in the code since 1 hour, so its generally usable if you compile the plugin yourself. I'd expect the next beta to be officially release within the next 7 days. I don't know when the next stable is planned. I guess HA just prefers to use stable versions in the addon (where we have no stakes in).

MiAutomations, 2021-07-06:
> Support is now included in the code since 1 hour, so its generally usable if you compile the plugin yourself. I'd expect the next beta to be officially release within the next 7 days. I don't know [...]

Ok thank you for the information and for your support !! But once all must the people is using the deconz conbee II over the HA and once the people is using the HassOS and it's not possible to "compile" over this OS ... I think it's the best decrease this time or create a alternative in order  to make this available for more users ... 

Thank you and best regards

SwoopX, 2021-07-06:
Well, I agree that it will probably make ppl happier, but's that's absolutely not in our hands. You need to suggest that to HA folks maintaining the addon, I'm afraid.

oudoulj, 2023-01-14:
I run Hassio with a **Conbee 2** stick and **Deconz**.
My Conbee gateway version : 2.19.03 / 19/09/2022
and its firmware : 26780700

[image]

Pairing of the Aqara wireless double-rocker switch WRS-R02 aka H1 went smoothly.

I can catch the various `deconz_event` in the Developer Tools / Events of HA (single, double, long clicks on both buttons) but **the automation screen does not expose those events** for me to choose from.
Only the "Battery level changes" trigger 😢 
Am I missing something obvious here ?

[image]

More info and screenshots [on the HA community](https://community.home-assistant.io/t/aqara-h1-button-press-not-triggering-automation/389498/14?u=jeromeo).
Thanks in advance for your help ! 🙏

Mimiix, 2023-01-14:
> I run Hassio with a **Conbee 2** stick and **Deconz**. My Conbee gateway version : 2.19.03 / 19/09/2022 and its firmware : 26780700
>
> [image]
>
> Pairing of the Aqara wireless double-rocker switch [...]

Please check the deCONZ integration documentation . It's explained there.

oudoulj, 2023-01-15:
> Please check the deCONZ integration documentation . It's explained there.

Can you please point me to a specific link ?
Because I've been googling for a while and couldn't find anything relevant, thank you very much.
_(I was able to find a workaround using the deconz_event as a trigger but it's not super straightforward)_

oudoulj, 2023-01-15:
> deCONZ integration documentation

Ok, thank you.
It's just that for such a popular and 2-year old switch, I thought it would have been fully supported already so I thought I was the one doing something wrong :wink:

Mimiix, 2023-01-15:
As far as i know, not a lot of devices are supported that way in home assistant. Nevertheless: it's really something home assistant needs to do

yungSven, 2023-01-17:
Same Issue here

oudoulj, 2023-01-27:
> Same Issue here

I used this method. Not super straightforward but works fine.
https://community.home-assistant.io/t/aqara-h1-button-press-not-triggering-automation/389498/15?u=jeromeo