dresden-elektronik/deconz-rest-plugin #4845: IRIS 3405-L Keypad

hansaya, 2021-05-07:
## Device

- Product name: IRIS 3405-L
- Manufacturer: CentraLite
- Model identifier: 3405-L
- Device type :
  - Other: Keypad

## Screenshots
[image]
[image]
[image]

### Basic
[image]

### Diagnostics
[image]

### Power
[image]

### Poll Control
[image]

### Temperature measurement
[image]

### IAS zone
[image]

a1ad, 2021-05-07:
@Smanar lets go ;-)

a1ad, 2021-05-07:
@hansaya You cant use docker. You need a linux OS like Debian or Raspbian to compile @Smanar his keypad branch..

Smanar, 2021-05-07:
Yep it s possible with docker https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-833875719
But not realy easy, and I m not skilled enought to help you for that.

You haven't another linux machine (a VM or a raspberry) ?

hansaya, 2021-05-08:
It doesn't matter as long as I can get some instruction to compile.

Smanar, 2021-05-08:
To compile it s easy on classic OS, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch keypad_test https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Have just mimic the other keypad code to test, will check for difference after.

But I can't explain you how to replace the file on docker ...

hansaya, 2021-05-10:
that fine, you want me to test the keypad branch?

Smanar, 2021-05-10:
Yep, pls, have added your device.
You have somme explanation to use it here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4758 (take care, it s not the same branch)

If you can tell me what is working or not ?
- the inclusion
- if your device use the presence sensor
- if you can see websocket return when using it
- if you can change the armed mode
- if you can change the panel mode
- ect ...

hansaya, 2021-05-13:
I compiled it and injected it to my container. When I started to pair again it recognize as a temperature device and seems to not complete pairing process. It keeps blinking.
[image]
[image]

Smanar, 2021-05-13:
Ok So I have a look on your device, and it seem it realy work than the one aleady added.
But to test it, I realy don't know HA, perhaps @Kane610 can explain how to install his code ?

Do you know how to use the api ? to send request to the device ?

You can too, just looking logs in deconz (with "IAS" and "info") , and use it, you will see somereturn in logs.

Kane610, 2021-05-13:
Hass support was released last week so it's already out in the open with HASS 2021.5 :)

Smanar, 2021-05-13:
Ha but it s already released in official version ...

So what is missing to test the device ? A problem in deconz json ? Just need to install your blueprint ?

a1ad, 2021-05-13:
Hey can watch for the deconz events first

Kane610, 2021-05-13:
Yes the blueprint and to try the different states because it's in an unknown state right now

Smanar, 2021-05-16:
Do you need more explanation @hansaya ?

hansaya, 2021-05-17:
After playing around with it,
* Arming and disarming seems to work changes the color of the keypad.
* Temperature readout seems to work.
* **Battery level indication is wrong**.
* Presence sensor works.
* Tempered indicator seems to work.
* I **cannot arm or disarm** through the keypad.
* **Cannot read the keypad code or set it**.
[image]

Smanar, 2021-05-17:
>Battery level indication is wrong.

Yep, have corrected something, not yet online.

>I cannot arm or disarm through the keypad.

This is not possible, the device is a dummy one, when you use the keypad, it need to send a notification to third app, and it s it that arm or disarm the device.

>Cannot read the keypad code or set it.

When you type a code, you need to have a notification through websocket.

If you are using HA (It s the only one application that can manage the device ATM) I think you need this blueprint to manage code https://community.home-assistant.io/t/deconz-simple-alarm-control-panel-blueprint/305397

a1ad, 2021-05-17:
@Smanar like i said: he need to watch for decoz events in the HA dev tools.

hansaya, 2021-05-17:
Thanks for link on the blueprint. It worked!
Could someone explain how to get deconz event? What I should be doing right before getting it etc..

a1ad, 2021-05-17:
> Thanks for link on the blueprint. It worked!
> Could someone explain how to get deconz event? What I should be doing right before getting it etc..

You dont need the events if the BP is working :-)

I am using the keypad with the Alarmo plugin and have 2 automations made to tie them together:

Alarmo to keypad:

```
alias: alarmo-to-keypad
description: ''
trigger:
  - platform: state
    entity_id: alarm_control_panel.alarmo
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: armed_away
        sequence:
          - service: alarm_control_panel.alarm_arm_away
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: armed_home
        sequence:
          - service: alarm_control_panel.alarm_arm_home
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: armed_night
        sequence:
          - service: alarm_control_panel.alarm_arm_night
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: disarmed
        sequence:
          - service: alarm_control_panel.alarm_disarm
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: triggered
        sequence:
          - service: alarm_control_panel.alarm_trigger
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: arming
        sequence:
          - service: deconz.alarm_panel_state
            data:
              panel_state: exit_delay
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
      - conditions:
          - condition: state
            entity_id: alarm_control_panel.alarmo
            state: pending
        sequence:
          - service: deconz.alarm_panel_state
            target:
              device_id: 8f995b9f01fae89eef5a5e3107d2c719
            data:
              panel_state: entry_delay
    default: []
mode: single
``` 

Keypad to alarmo:

```
alias: keypad-to-alarmo
description: ''
trigger:
  - platform: event
    event_type: deconz_alarm_event
condition: []
action:
  - choose:
      - conditions:
          - condition: template
            value_template: '{{ trigger.event.data.event == "armed_away" }}'
        sequence:
          - service: alarmo.arm
            data:
              entity_id: alarm_control_panel.alarmo
              code: '{{ trigger.event.data.code }}'
      - conditions:
          - condition: template
            value_template: '{{ trigger.event.data.event == "armed_night" }}'
        sequence:
          - service: alarmo.arm
            data:
              entity_id: alarm_control_panel.alarmo
              code: '{{ trigger.event.data.code }}'
              mode: night
      - conditions:
          - condition: template
            value_template: '{{ trigger.event.data.event == "armed_home" }}'
        sequence:
          - service: alarmo.arm
            data:
              entity_id: alarm_control_panel.alarmo
              code: '{{ trigger.event.data.code }}'
              mode: home
      - conditions:
          - condition: template
            value_template: '{{ trigger.event.data.event == "disarmed" }}'
        sequence:
          - service: alarmo.disarm
            data:
              entity_id: alarm_control_panel.alarmo
              code: '{{ trigger.event.data.code }}'
    default: []
mode: single

```

Smanar, 2021-05-17:
Tell me is all is working, to know what I still need to change.
Have pushed the battery corrective.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

hansaya, 2021-05-18:
battery still shows 1%

hansaya, 2021-05-18:
it changed to 100% now, which I don't believe either

Smanar, 2021-05-18:
>it changed to 100% now, which I don't believe either

Lol, good question ^^, I have used the value from your capture on power cluster.

Battery voltage = 30
Battery quantity = 2

So was for me 2 * 1.5 V ?

hansaya, 2021-05-18:
There are two batteries but both in parallel for capacity. Unit takes in two CR123A batteries and configured in parallel. Battery rated at 3V so your math should be right. 

According to the multi-meter, while used battery is plugged in, it shows: 3.02v and through deconz shows: 30
New battery shows without any load: 3.27V
New battery in the unit shows: 3.13V and through deconz shows: 30

So my guess is 30 is max considering 3v battery and lowest according to the unit configured at 22. Having 100% seems correct even with a used battery because its was over 3v. 

I could test it with bench power supply down to 2.2v if you want to validate.

Smanar, 2021-05-18:
> lowest according to the unit configured at 22

Yeah I think too, but have used deconz defaut value for the algorytm.

>I could test it with a bench power supply down to 2.2v if you want to validate.

No thx, the hardware can do "magic" too with values, if realy the value don't move we take a look more closer, but for me it 's ok.

I will add your device to the previous PR, now need to wait for review, and BTW thx a lot for yours tests ^^

hansaya, 2021-05-18:
Thank you for doing the actual work :)

hansaya, 2021-05-18:
One last thing, as you can see there is a panic button on they keyboard. Anyways to get that information out?

Smanar, 2021-05-18:
Ha yes right, it s something not present on the other one.

Can you check for log, when using this button ? with "ias", if I m right you will have as log
>"[IAS ACE] - Address 0x...., Payload ...., Command 0x04"

But idk how to send it to third app, perhaps using "state/action" too ?

Kane610, 2021-05-18:
Action is expected to be a comma separated list right now. At least in hass. Maybe use button press or something?

hansaya, 2021-05-19:
I think this is what you looking for
`00:00:00:225 [INFO] - No button map for: 3405-L, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: 0x04, payload: None, zclSeq: 58`

Is there is anyway to link that button to trigger the alarm? Either by mapping it to different output or to a panic event?

hansaya, 2021-05-19:
I created two blueprints based on your guys work to simplify integrating with keypad and home assistant alarm panel.
https://gist.github.com/hansaya/4d79a1ac4ce748297087b8ef24cea3af
https://gist.github.com/hansaya/6b0b3ad4d86dab38711c8313bc1792b2

I saw @Smanar using [alarmo](https://github.com/nielsfaber/alarmo) but that seems to be different from what [Home Assistant comes with](https://www.home-assistant.io/integrations/manual/).

Let me know this is helpful and where I can improve.

Smanar, 2021-05-19:
>Is there is anyway to link that button to trigger the alarm? Either by mapping it to different output or to a panic event?

I can do, all is possible to imagine, but there is more work on @Kane610 side ^^, it s him who implement the device. This device is a "dummy" device, all need to be done on third app.

Need to check first what is better for third app.

hansaya, 2021-05-19:
Did more testing with home assistant and keypad. Found more issues. I made a video so its easy see what's going on https://www.youtube.com/watch?v=FiAMqK6-Rpw

tldr is
* Keypad needs to beep when arming so you know that you have time to get out of the house. This seems to work but it quickly stops after few seconds.
* Keypad needs to beep when something trips so that you have time unlock before alarm actually get triggered. This partial work as well only if alarm was triggered by keypad presence sensor.

When I used this keypad with smarthings, it did all of this so I believe we should be able to get it working. 

Thanks again for all your help.

Smanar, 2021-05-19:
Hu, @a1ad your advices about that ? ^^

a1ad, 2021-05-19:
Strange, i need to test his blueprints first because i am using Alarmo. I let you know after the tests.

But i don't use the presence sensor to trip the alarm, it is way too sensitive.

Kane610, 2021-05-19:
> Did more testing with home assistant and keypad. Found more issues. I made a video so its easy see what's going on https://www.youtube.com/watch?v=FiAMqK6-Rpw
> 
> tldr is
> 
> * Keypad needs to beep when arming so you know that you have time to get out of the house. This seems to work but it quickly stops after few seconds.
> * Keypad needs to beep when something trips so that you have time unlock before alarm actually get triggered. This partially work as well only if alarm was triggered by keypad presence sensor.
> 
> When I used this keypad with smarthings, it did all of this so I believe we should be able to get it working.
> 
> Thanks again for all your help.

Yes, the blueprint only validates the input code and reflects back the arm state that was sent with the code. I'm planning on doing something that also does additional beeping but haven't gotten to it yet.

hansaya, 2021-05-20:
Blue print I linked above does notify the deconz code about the [pending status](https://gist.github.com/hansaya/4d79a1ac4ce748297087b8ef24cea3af#file-alarm_to_deconz_blueprint-yaml-L73) and the [alarm status](https://gist.github.com/hansaya/4d79a1ac4ce748297087b8ef24cea3af#file-alarm_to_deconz_blueprint-yaml-L65)

Kane610, 2021-05-20:
You need to control how long you want it to beep prior to setting it to armed_xxxx

hansaya, 2021-05-20:
is there any documentation how I can go about setting that?

hansaya, 2021-05-26:
@Smanar If you dont need anything else from you can close this out.

Kane610, 2021-05-26:
> is there any documentation how I can go about setting that?

Probably, but that's not really my speciality. Look at script documentation or action documentation

Smanar, 2021-05-26:
It will be closed automatically when the PR will be validated
Manup is buying a keypad too, to make test before validation.
And we can add too panic event, but I need to see with @Kane610 what is the better way for third app integration.

Kane610, 2021-06-17:
Go away bot your services are not needed here

closed 2021-07-17 by github-actions[bot]

hansaya, 2021-11-03:
I'm on 2.13.01 beta and I see that this is got merged but I cannot seems to get the panel working

Smanar, 2021-11-03:
it's the "config/reachable" that is false ?
The rest is working ?
Can you show the device sjon ?

hansaya, 2021-11-04:
everything else seems to work. Perhaps I need to repair?
I do see this in my HA logs as well
`2021-11-03 08:58:43 WARNING (MainThread) [homeassistant.helpers.service] Unable to find referenced entities alarm_control_panel.garage_keypad`

Smanar, 2021-11-05:
IDK what HA is looking to set the device as unreachable.
Can you show the device JSON ? You can find it http://phoscon.de/pwabeta in help / API information.

Kane610, 2021-11-05:
> IDK what HA is looking to set the device as unreachable.
> Can you show the device JSON ? You can find it http://phoscon.de/pwabeta in help / API information.

The integration bases availability on the websocket connection between hass and deconz and the reachable state of the ancillary control sensor
https://github.com/home-assistant/core/blob/8a1d80c60933f07c8325bcc37ac51a3c6dd37d5b/homeassistant/components/deconz/deconz_device.py#L98

Smanar, 2021-11-05:
So it mean the config/recheable is false.
And if the device is working, this field need to be true. When you said "everything else seems to work" the device have already updated a field "action","panel" or other ?

Kane610, 2021-11-07:
Best thing would be to enable debug logs for hass and see what is sent from Deconz

hansaya, 2021-11-09:
This is what I see on the API json output

Event when I pressed a button
```
    "11:17:57:290": {
        "attr": {
            "id": "42",
            "lastannounced": "2021-10-08T01:30:32Z",
            "lastseen": "2021-11-09T17:17Z",
            "manufacturername": "CentraLite",
            "modelid": "3405-L",
            "name": "AncillaryControl 42",
            "swversion": null,
            "type": "ZHAAncillaryControl",
            "uniqueid": "00:0d:6f:00:0e:2d:c6:e4-01-0501"
        },
        "e": "changed",
        "id": "42",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:0d:6f:00:0e:2d:c6:e4-01-0501"
    },
```
REST API:
```
    "alarmsystems": {
        "1": {
            "config": {
                "armed_away_entry_delay": 120,
                "armed_away_exit_delay": 120,
                "armed_away_trigger_duration": 120,
                "armed_night_entry_delay": 120,
                "armed_night_exit_delay": 120,
                "armed_night_trigger_duration": 120,
                "armed_stay_entry_delay": 120,
                "armed_stay_exit_delay": 120,
                "armed_stay_trigger_duration": 120,
                "armmode": "disarmed",
                "configured": false,
                "disarmed_entry_delay": 0,
                "disarmed_exit_delay": 0
            },
            "devices": {},
            "name": "default",
            "state": {
                "armstate": "disarmed",
                "seconds_remaining": 0
            }
        }
    },
"42": {
            "config": {
                "battery": 100,
                "enrolled": 0,
                "on": true,
                "pending": [],
                "reachable": true
            },
            "ep": 1,
            "etag": "3ba9c460fe3eb166d4989a0a4e2e8317",
            "lastannounced": "2021-10-08T01:30:32Z",
            "lastseen": "2021-11-09T17:19Z",
            "manufacturername": "CentraLite",
            "modelid": "3405-L",
            "name": "AncillaryControl 42",
            "state": {
                "action": "not_ready",
                "lastupdated": "2021-11-09T17:17:58.255",
                "lowbattery": false,
                "panel": null,
                "seconds_remaining": 0,
                "tampered": false
            },
            "type": "ZHAAncillaryControl",
            "uniqueid": "00:0d:6f:00:0e:2d:c6:e4-01-0501"
        },
"43": {
            "config": {
                "battery": 100,
                "duration": 60,
                "enrolled": 1,
                "on": true,
                "pending": [],
                "reachable": true
            },
            "ep": 1,
            "etag": "3ba9c460fe3eb166d4989a0a4e2e8317",
            "lastannounced": "2021-10-08T01:30:32Z",
            "lastseen": "2021-11-09T17:19Z",
            "manufacturername": "CentraLite",
            "modelid": "3405-L",
            "name": "Presence 43",
            "state": {
                "lastupdated": "2021-11-09T17:05:39.172",
                "lowbattery": false,
                "presence": false,
                "tampered": true
            },
            "type": "ZHAPresence",
            "uniqueid": "00:0d:6f:00:0e:2d:c6:e4-01-0500"
        },
        "44": {
            "config": {
                "battery": 100,
                "offset": 0,
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "74718a83a5c93176fb71153e3060a6e8",
            "lastannounced": "2021-10-08T01:30:32Z",
            "lastseen": "2021-11-09T17:19Z",
            "manufacturername": "CentraLite",
            "modelid": "3405-L",
            "name": "Temperature 44",
            "state": {
                "lastupdated": "2021-11-09T17:19:31.234",
                "temperature": 2331
            },
            "type": "ZHATemperature",
            "uniqueid": "00:0d:6f:00:0e:2d:c6:e4-01-0402"
        },
```

Smanar, 2021-11-09:
And you still have "Unavailable" ?
Because on the json you have  "reachable": true

We are searching how to use it on last version, It seem you need to set the "code0" but there is an issue on it https://forum.phoscon.de/t/with-deconz-v2-13-01-pin-code-cannot-be-set-anymore-nor-can-arming-be-changed/613/

So you can try to set a code0, but there is lot of chance you are blocked at same step.

hansaya, 2021-11-09:
Yes it shows as unavailable.
[image]

Kane610, 2021-11-09:
The keypad is not part of the alarmsystem `"devices": {},`

hansaya, 2021-11-09:
Anything I can do to get that in there?

Smanar, 2021-11-10:
> The keypad is not part of the alarmsystem `"devices": {},`

Ha nice to know, so the plugin check for the sensor ZHAAncillaryControl is reachable AND it s a part of the alarm system ?

@hansaya Try first to set a "code0" if it work we will continue, but there is lot of chance you was blocked at this step for the moment.

Kane610, 2021-11-11:
> > The keypad is not part of the alarmsystem `"devices": {},`
> 
> Ha nice to know, so the plugin check for the sensor ZHAAncillaryControl is reachable AND it s a part of the alarm system ?
> 
> @hansaya Try first to set a "code0" if it work we will continue, but there is lot of chance you was blocked at this step for the moment.
> 
> If you have the issue, are you able to compile the code ? It will solve the issue too.

Yes, you don't know what alarm system to control if they're not paired

Smanar, 2021-11-12:
I realy don't know how the alarm system, I was thinking alone device  were able to work in stand alone mode.

hansaya, 2021-12-09:
sorry for taking time to test this. I managed to change the alarmsystem to configured but HA shows keypad as unavailable. 
```
    "alarmsystems": {
        "1": {
            "config": {
                "armed_away_entry_delay": 120,
                "armed_away_exit_delay": 120,
                "armed_away_trigger_duration": 120,
                "armed_night_entry_delay": 120,
                "armed_night_exit_delay": 120,
                "armed_night_trigger_duration": 120,
                "armed_stay_entry_delay": 120,
                "armed_stay_exit_delay": 120,
                "armed_stay_trigger_duration": 120,
                "armmode": "disarmed",
                "configured": true,
                "disarmed_entry_delay": 0,
                "disarmed_exit_delay": 0
            },
            "devices": {},
            "name": "default",
            "state": {
                "armstate": "disarmed",
                "seconds_remaining": 0
            }
        }
    },
```

Kane610, 2021-12-09:
As you see in devices: {} there is no device listed, that's the reason, something is wrong on the deconz side

[12 more replies]