dresden-elektronik/deconz-rest-plugin #4847: _TZ3000_qzjcsmar  - events fire twice on pressed

thomasgermain, 2021-05-07:
## Describe the bug

I Recently bought a zigbee switch -> https://zigbee.blakadder.com/Smart9_S9TSZGB_3.html.
Sounds it should be supported from version 2.10.4, but I have issues with buttons press (single double and hold).

Multiple events are fired on button pressed (events 1002, 1003, 1004, 2002, 2003, 2004, 3002, 3003, 3004):
There is ~ 1 second delay between the two events (but I only do the singled pressed/double pressed/hold once).
```
12:41:53:467 [INFO] - Button 1002 - TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 1
12:41:54:461 [INFO] - Button 1002 - TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 1
``` 
I can also see the `deconz_event` in HA are coming twice.

Also, the double pressed doesn't seem to be detected each time I did it, sometimes it is recognized as a single pressed (but the event is still fired twice).

Bonus question, the switch is recognized as "Switch" and not as "Smart switch", meaning I can't link the switch to a light group, is there anything to do about that ? (Top one is aquara switch, below is the _TZ3000_qzjcsmar switch).
[image]

 

## Steps to reproduce the behavior

Binding the switch. Press one or twice or hold

## Expected behavior

Event should only be fired once.

## Screenshots

## Environment

 - Host system: RPI 4
 - Running method: HA
 - Firmware version: 26660700
 - deCONZ version: 2.10.04
 - Device: Conbee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? External SSD on USB3

## deCONZ Logs

## Additional context

Smanar, 2021-05-07:
There is a PR for that https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4716
But it seem it is not on your version yet, v2.11.0-beta minimum (if I m right)

thomasgermain, 2021-05-07:
Ah yeah, just checked deeper the release notes. I'll wait for the release then :)

dpozimski, 2021-05-15:
when do you plan to release v2.11? there is a lot of pending fixes and new supported devices that are not published yet :)

Smanar, 2021-05-15:
Depend of OS, but some of them are already available http://deconz.dresden-elektronik.de/raspbian/beta/

closed 2021-06-14 by github-actions[bot]