dresden-elektronik/deconz-rest-plugin #4870: SINOPE RM3250ZB SMART CONTROLLER

joelwener, 2021-05-13:
## Device

- Product name: SINOPE smart controller
- Manufacturer: Sinope.
- Model identifier: RM3250ZB
- Device type :
  - Other: Smart Plug-Controller
[image]

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

joelwener, 2021-05-13:
Missing on/off
[image]

Mimiix, 2021-05-13:
Is this a double post of #4871 ?

joelwener, 2021-05-13:
> Is this a double post of #4871 ?

No, two separate devices.  One is a smart plug, one is a smart controller (load)

Mimiix, 2021-05-14:
@joelwener  the one in #4871 has the same model ID as this one (according to the screenshots)... 

Can you make them both complete with all cluster screenshots?

joelwener, 2021-05-14:
Here are other cluster screen shots which I left out as I didn't consider them relevant at the time.
[image]

joelwener, 2021-05-14:
Sorry about that, I chose all the photos when I made the request, but I just realized it only posted one.  The post has been edited to include missing screenshots.

closed 2021-06-13 by github-actions[bot]