dresden-elektronik/deconz-rest-plugin #4871: SINOPE Smart Plug

joelwener, 2021-05-13:
## Device

- Product name: Sinope smart plug
- Manufacturer: Sinope
- Model identifier: SP2600ZB
- Device type :
  - Other: Smart Plug
[image]

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

joelwener, 2021-05-13:
Missing on/off
[image]

joelwener, 2021-05-14:
Smart plug / controller may qualify as sensor as well.

joelwener, 2021-05-14:
To be clear, I would like the Phoscon I/f to recognize the following cluster points (as either sensors or readable points of the smart plug/controller)

Current Temperature

Electrical Measurement Cluster
RMS Current
RMS Voltage
Active Power
AC Frequency

Simple Metering Cluster :
Current Summation Delivered (kW)

On/Off (is already recognized)

Currently recognized as a light in Phoscon

Mimiix, 2021-05-14:
@joelwener Can you add all screenshots in this issue?

joelwener, 2021-05-14:
Corrected issue, included all missing screenshots.  Sorry about that.

Mimiix, 2021-05-14:
@joelwener No problem at all! :)

closed 2021-06-13 by github-actions[bot]