dresden-elektronik/deconz-rest-plugin #4874: Devices disappear / will be removed from network, if switched off power

acid115, 2021-05-13:
## Describe the bug

Hi, Since a couple of weeks i have the issue, that when a device got powerloss (plug it off, push the wall switch or sth.) deconz forget the device after 5-10min. After rescan the device appears again, but the name is removed AND the device has a new ID.
The the ID is really annoying because i use a lot of http put requests in scripts to control lights, switches, sensors and rules.
so if devices are out of reach for 10-15 minutes (no problem if only 1-4 minutes) and they are removed from network, i always have to restore my last backup to get back the Devices and their IDs - i hate it and hope, that all devices have power all the time :)

## Steps to reproduce the behavior

1. add a new device to your zigbee network (or use an existing)
2. Give it a name and a group
3. plug off the device from power or switch it off - make it powerloss
4. wait about 15 minutes
5. the device disappears (in phoscon an in .../lights/) and is not only greyed out - it is gone
6. replug the device to power - switch it on.
7. device connects to network. If you control the group, to which the device was before, the device works (Zigbee switches can control the device)
But you will not see the device in phoscon or ../lights/
8. start rescan for new devices - the device appears
9. the device get the default name and a new ID
10. the device will automatically connect to the groups, to which the device was assigned before (maybe the group is stored on the device ?)

## Expected behavior

the device is only greyed out on power lost...
after replug the device has the same ID and Name as before 

## Screenshots

## Environment

Firmware: 

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26330500 ( i also tried the new 26390500)
 - deCONZ version: 2.11.3
 - Device:  ConBee I 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No
## deCONZ Logs

## Additional context

Mimiix, 2021-05-14:
Can you add some logs :)?

manup, 2021-05-14:
Which device device is this?

Can you please check the file permissions of the related files, where device information is stored:

```
$ ls -l ~/.local/share/dresden-elektronik/deCONZ/

-rw-r--r-- 1 pi pi   2533 Mai 10 18:44 config.ini
-rw-r--r-- 1 pi pi   1348 Jan 15 19:50 session.default
-rw-r--r-- 1 pi pi     35 Sep 25  2020 zcldb.txt
-rw-r--r-- 1 pi pi 135168 Mai 10 18:44 zll.db
```

acid115, 2021-05-14:
the permissions are ok. i need some time to provide the logs
```
total 884
drwxr-xr-x 2 pi pi   4096 May 14 14:29 .
drwxr-xr-x 3 pi pi   4096 Dec 16  2018 ..
-rw-r--r-- 1 pi pi   2441 May 13 08:35 config.ini
-rw-r--r-- 1 pi pi  10082 May 14 02:45 session.default
-rw-r--r-- 1 pi pi     35 Dec 16  2018 zcldb.txt
-rw-r--r-- 1 pi pi 438272 Feb  2  2019 zll-bal.db
-rw-r--r-- 1 pi pi 438272 May 14 14:29 zll.db
```

acid115, 2021-05-14:
and the devices i tested with:
lidl silvercrest HG06337 (smart plug)
lidl silvercrest HG06104A (led stripe)
blitzwolf  _TZ3000_g5xawfcq (smart plug)

Mimiix, 2021-05-14:
Are you on docker?

acid115, 2021-05-14:
no docker

Mimiix, 2021-05-14:
hmm okay. Can you provide some logs on how this happens?

acid115, 2021-05-14:
yes yes, i am in the log now and try to find some relevant things..
Basic information:
Devicename: Heiman4
id: 31
time unplugged: 14:55:00
time disappeared: 15:05:30

at 14:55 i see no relevant log entries...
but at 15:05 is see:
```
15:05:28:559 discard sensor state push for 16: state/lastupdated (already pushed)
15:05:28:559 rule event /sensors/16/state/lastupdated: 0 -> 0
15:05:28:752 schedule 4 diff 21272, Fri May 14 21:00:00 2021
15:05:28:753 enqueue event config/localtime for /config/
15:05:28:754 rule event /config/localtime: 15:05:27.752 -> 15:05:28.753 (1001ms)
15:05:28:827 poll node 7c:b0:3e:aa:0a:02:14:5d-03
15:05:28:827 Poll light node Sterne
15:05:28:880 Poll APS request to 0x7CB03EAA0A02145D cluster: 0x0006 dropped, values are fresh enough
15:05:28:989 Node 0xBC33ACFFFEFE417E is known by 1 neighbors, last seen 68 s
15:05:29:052 Set sensor check interval to 1000 milliseconds
15:05:29:468 Node 0x60A423FFFEF5E1D1 is known by 1 neighbors, last seen 45 s
15:05:29:752 schedule 4 diff 21271, Fri May 14 21:00:00 2021
15:05:29:753 enqueue event config/localtime for /config/
15:05:29:754 poll node f0:d1:b8:00:00:13:59:29-01
15:05:29:754 Poll light node Ledvance1_TV-Schlafzimmer
15:05:29:755 rule event /config/localtime: 15:05:28.753 -> 15:05:29.753 (1000ms)
15:05:29:803 Poll APS request to 0xF0D1B80000135929 cluster: 0x0006 dropped, values are fresh enough
15:05:29:949 Node 0x680AE2FFFE3E5398 is known by 12 neighbors, last seen 700 s
15:05:29:949 0x74A1 seems to be a zombie recv errors 11
15:05:29:950 enqueue event event/deleted for /lights/31
15:05:29:995 Node zombie state changed 0x680ae2fffe3e5398
15:05:30:429 Node 0xBC33ACFFFE54DC64 is known by 18 neighbors, last seen 205 s
15:05:30:749 poll node f0:d1:b8:00:00:13:66:6a-01
15:05:30:749 Poll light node Ledvance2
15:05:30:752 schedule 4 diff 21270, Fri May 14 21:00:00 2021
15:05:30:753 enqueue event config/localtime for /config/
15:05:30:754 rule event /config/localtime: 15:05:29.753 -> 15:05:30.753 (1000ms)
15:05:30:802 Poll APS request to 0xF0D1B8000013666A cluster: 0x0006 dropped, values are fresh enough
15:05:30:909 Node 0x842E14FFFE2C44F1 is known by 15 neighbors, last seen 100 s
15:05:31:389 Node 0x001788010403FAF0 is known by 19 neighbors, last seen 10 s
15:05:31:629 Mgmt_Lqi_req zdpSeq: 133 to 0x842E14FFFE058432 start index 3
15:05:31:727 ZDP status = 0x00 -> SUCCESS
15:05:31:727 ZDP Mgmt_Lqi_rsp zdpSeq: 133 from 0x842E14FFFE058432 total: 23, startIndex: 3, listCount: 3
15:05:31:728     * neighbor: 0x842E14FFFE32CDA1 (0x4676), LQI: 105, relation: 0x02 rxOnWHenIdle: 1
15:05:31:728     * neighbor: 0x000B57FFFE14950B (0x53D6), LQI: 134, relation: 0x02 rxOnWHenIdle: 1
15:05:31:728     * neighbor: 0x00158D00038BA766 (0x5A85), LQI: 219, relation: 0x02 rxOnWHenIdle: 1
15:05:31:750 poll node d0:cf:5e:ff:fe:ba:66:5f-01
15:05:31:750 Poll light node Höhle
15:05:31:752 schedule 4 diff 21269, Fri May 14 21:00:00 2021
15:05:31:753 enqueue event config/localtime for /config/
15:05:31:754 rule event /config/localtime: 15:05:30.753 -> 15:05:31.752 (999ms)
15:05:31:783 Master: read param with arg 0x19
15:05:31:803 Poll APS request to 0xD0CF5EFFFEBA665F cluster: 0x0006 dropped, values are fresh enough
15:05:31:815 Current channel 15
...
15:06:15:263 ZDP status = 0x00 -> SUCCESS
15:06:15:263 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x4676
15:06:15:263 ZDP Mgmt_Lqi_rsp zdpSeq: 162 from 0x842E14FFFE32CDA1 total: 19, startIndex: 18, listCount: 1
15:06:15:263     * neighbor: 0x001788010403F93E (0xF90F), LQI: 99, relation: 0x02 rxOnWHenIdle: 1
15:06:15:269 API error 3, /lights/31, resource, /lights/31, not available
15:06:15:301 Poll APS request to 0xD0CF5EFFFEBA665F cluster: 0x0006 dropped, values are fresh enough
15:06:15:357 Node 0xF0D1B80000136464 is known by 13 neighbors, last seen 51 s
15:06:15:587 Poll APS request to 0xD0CF5EFFFEBA665F cluster: 0x0008 dropped, values are fresh enough
15:06:15:684 read attributes of 0xD0CF5EFFFEBA665F cluster: 0x0300: [ 15:06:15:685 0x0003 15:06:15:685 0x0004 15:06:15:685 0x0008 15:06:15:685 0x4001 15:06:15:685 ]
```
Look at: 15:05:29:950
and 15:06:15:269 

Do you need more ?

Mimiix, 2021-05-14:
I'd like to have a full log file with:

INFO, ERROR,ERROR_L2,APS,APS_L2

And for a few minutes. Can you provide that?

Mimiix, 2021-05-14:
15:06:15:269 API error 3, /lights/31, resource, /lights/31, not available

That is because the light is not reachable.

acid115, 2021-05-14:
yes i know,
but interesting is:
15:05:29:950 enqueue event event/deleted for /lights/31
There was a Delete-Event ?
But why ?
i can provide the log. give me a little bit more time :)

acid115, 2021-05-14:
so, sorry. 
my kids need a lot of time :)

New basic informations. After rescan the device 31 got  the  new ID: 27. 
Devicename: Heiman4
id: 27
time unplugged: 16:01:40
time disappeared: 16:22

Now, it took 20 minutes ...

Started deconz with: --dbg-info=2 --dbg-error=2 --dbg-aps=2

the deleted-event:
16:21:37:350 enqueue event event/deleted for /lights/27 

[logstd.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6479591/logstd.log)

SwoopX, 2021-05-14:
What is its mac address and nwk? And what kind of Heiman device is that? Anf why do you unplug it?

acid115, 2021-05-14:
> What is its mac address and nwk? 
nwk???
mac address of the light? do you mean unique id?
uniqueid": "68:0a:e2:ff:fe:3e:53:98-0b" 
> And what kind of Heiman device is that? 
Sorry, it is a lidl silvercrest HG06337. i call it heiman, because a half year ago deconz recognized it as Heiman-Device. this was fixed later. But for me these plugs are Heimans... :)
> Anf why do you unplug it?
Ehm, Why not? a have several plugs and bulbs, i do not need at the moment. so they are in a box ... 
or sometimes bulbs are not reachable ...

Smanar, 2021-05-14:
The NWK, is the network adress something like 0x1234

>16:01:53:436 node 0680AE2FFFE3E5398 leave wait state

This log is from deconz core too ?

And you have it this issue for all your devices that are power off ?

acid115, 2021-05-14:
Ok, do you need the nwk?
If yes, then where can i find ?

I have this issue for all devices...
Ikea bulbs, blitzwolf plug, osram plug...
If "important" devices are deleted, i have to restore a backup.

But i have no problems with sensors and switches

SwoopX, 2021-05-14:
Yeah, it helps since log data is a bit jumpy...  Check here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support , 2nd screenshot (the device must be selected in deconz GUI).

acid115, 2021-05-14:
Oh, the GUI :(
Then i have to find a way to access the gui...just work per ssh and webinterfaces...

SwoopX, 2021-05-14:
Regardless, your experience is quite remarkable. I'm running the same version and I also have mains devices which are cut from power, but the always come back as they should. Same in my dev environment where I got quite some orphan devices, they all persist...

I know there have been some changes around the capabilities to delete a node, but that must trigerred explicitly to my knowledge.

acid115, 2021-05-17:
So, sorry.
Had a lot of problems to get the deconz ui... No success with rdp :(
No success with win10
But i have an old win7 system, the stick is recognized.
I had repaired the plug to see the node infos. 
But i dont know, if this information is really needed, because with every device i have this issue...
[image]

acid115, 2021-05-18:
so, now i can access the gui from my raspberry...
So if you need more infos from deconz gui, i can provide it.

there is a new log from deconz gui.
At 12:48 i unplugged the "Heiman4"-LIDL PLUG-Unit
At 12:53 the device disappeared in Phoscon.

But the node in gui is visible and stays visible.
It lost the connection stripes.
After replug the node starts blinking and connects to again. 
any ideas ?

[logFileDeconzD210518_1255.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6503824/logFileDeconzD210518_1255.log)

After replug:
[image]

acid115, 2021-05-26:
Upgraded firmware to 26390500...
No change.
Any ideas?

acid115, 2021-06-16:
Hi, 
the issue can be closed.
for months i have generated the libde_rest_plugin.so by myself, because of small changes in rest_lights.cpp
Periodically i merged the current upstream/master into my fork and compiled the plugin.
I had no problems so far.
But the only module i have updated for months was this plugin. The other modules for deconz like the main binary /usr/bin/deCONZ and the other plugins  i have disabled in apt... :(
sorry for that!
now it works again !

closed 2021-06-16 by acid115