dresden-elektronik/deconz-rest-plugin #4877: SonOff Zigbee issue : Not connecting to the nearest router (Legrand in my case)

samsam-rolon, 2021-05-14:
## Describe the bug
Hello all !

I'm trying to add some SonOff Temp devices (SNZB-02) and I've notice that a can't paire them, execpt if i'm near of the deconz gateway because pairing them to their futur place is not working.

All my devices are in Zigbee 2.0, most of them are router, (Legrand Router) and the SNZB-02 looks like directly connected to deCONZ or prefere my 2 Ikea bulbs.

Any idea ?

## Steps to reproduce the behavior
Pairing SNZB-02 with a Legrand router

## Expected behavior
SNZB-02 should accept to be paired with the nearest router

## Screenshots
NC

## Environment
Full Legrand Environment (router and wireless switch),, 2 ikea bulbs, 1 Xiami button, 1 Xiaomi Temp sensors with 1 CC2530+CC2591 extender for garage for SonOff Sensors and Orvibo Actioner

 - Host system: (NAS Synology with Docker)
 - Running method: (Marthoc Docker container )
 - Firmware version: (26680700)
 - deCONZ version: (2.10.04 )
 - Device: (ConBee II)
 - Do you use an USB extension cable: yes, USB 3.0 Hub with external power supply
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
NC

## Additional context
NC

SwoopX, 2021-05-14:
Where is the deconz bug then?

Smanar, 2021-05-14:
Hello, So if you make the inclusion close an ikea bulb it works, but not if you make the same thing close the Legrand device ?

samsam-rolon, 2021-05-14:
> 
> 
> Where is the deconz bug then?

I'm sorry, deCONZ is not really linked to this issues, as far as a Zigate or Zigbee2MQTT. I'm just tracing a zigbee issue there, I think is alway a good things because deCONZ, Zigbee2MQTT and Zigate Issues are a good DB of Zigbee Issue, not only about gateway but also about devices. If there is a forum or git repo to issue only relevant of devices comportment on zigbee stack, please tell me :-)

> 
> 
> Hello, So if you make the inclusion close an ikea bulb it works, but not if you make the same thing close the Legrand device ?
> There is a similar issue here, a device that can pair direclty, but not using a router. #4829
> 
> But I don't see how a router can prevent an inclusion.

Close to Ikea Bulb or deCONZ : yes
Close to Legrand, no. 
Worst, I see the device in the GUI like this : (pairing not completed so ?) and all legrand router are leaving the inclusion mode just after "pairing'
[image]

Mimiix, 2021-05-14:
@SwoopX Can i proceed to close this issue as it isn't a deCONZ bug or deCONZ related at all?

samsam-rolon, 2021-05-14:
> 
> 
> @SwoopX Can i proceed to close this issue as it isn't a deCONZ bug or deCONZ related at all?
I just discover the TAG "User question" . If this topic is requalified as User Question which is more relevant :-) please do not close it so quickly :) It would be intrested to know if other people get trouble like this

SwoopX, 2021-05-14:
Tbh, I'd prefer if you open a new dedicated question for this and make sure that you want other experiences (something like: Your experience? Router devices that prefer coordinator as neighbor than nearest router).

It's still nothing apparently really related to deconz but eventually worth gathering as common knowledge for other ppl.

samsam-rolon, 2021-05-14:
Okay, I undestand ! 
For the "side of" : I see the device in GUI but it's still not paring, is it can be related to deCONZ or device ? How can i see it in logs ? Which options ?

SwoopX, 2021-05-14:
@Smanar Haven't we had that Legrad pairing issue like ages ago?

From my very personal experience, device leave if the do not get a proper `match descriptor response` on zigbee level. But to confirm that, we'd require a zigbee traffic sniff. It can be mitigated by exposing the "wanted" cluster through the coordinator (if any is missing, good example is the IAS ACE cluster). However, if memory serves, it was something different for Legrand.

Smanar, 2021-05-14:
IDK, second time I m seing this kind of issue for router, the second one is for vimar router https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4829

Legrand device have a subtility that can make themself leave, but not the others devices.

But I m thinking like you, @samsam-rolon have you already tried to "finish" manualy the inclusion ?
You can send again request using the "first circle" to make the second one appear.

And I think the one with the "big number" are the one blocked. But It mean Legrand device are blocking thoses request is they are used as router ..... For me it s not in Legrand Netatmo mentality.
But if It s that, I think we can ask direclty to Legrand.

samsam-rolon, 2021-05-14:
As far as I have tested : nothing ! 

But the more strange thing was the previous point :
I see the device like a partial inclusion BUT a the moment that the device appear in deCONZ GUI, all my legrand stuff (which was led ON green notifying "Inclusion Mode") will switch their LED OFF !

samsam-rolon, 2021-05-14:
I have also notice this issue when I was trying to pair my device to the next to my ORVIBO Door Controler

Smanar, 2021-05-14:
>As far as I have tested : nothing !

It mean none command are working to finish he inclusion ?

>I see the device like a partial inclusion BUT a the moment that the device appear in deCONZ GUI, all my legrand stuff (which was led ON green notifying "Inclusion Mode") will switch their LED OFF !

If I m right, Legrand device make that when you close the network ? Nothing visible on log about that ?

>It's smell more "general issue" that a specific device issue. Or related to SonOff ?

Good question, no one else have issue to include sonoff routers ?

samsam-rolon, 2021-05-14:
> Edit:
> [Koenkk/zigbee2mqtt#6979](https://github.com/Koenkk/zigbee2mqtt/issues/6979)

OMG ...
Okay so I explain ! Usually, it's not the things to do ...  undestands why all of them that I have paire next to te gateway are working, and not the orhers ...

Smanar, 2021-05-15:
>Usually, it's not the things to do 

Right, not logic at all, but it seem the problem is for the inclusion only, and as this device is router, can move it after, I hope he is able to rebuild the mesh ....

samsam-rolon, 2021-05-15:
Okay ... @Mimiix I've just closed that issue !
Looks working ! @SwoopX @Smanar : the mesh is re-calculated well afte few transmition, cf screenshot : sensors are routed through legrand devices !
[image]

closed 2021-05-15 by samsam-rolon